mavlink/mavros #430: libmavconn reports "compiler /usr/bin/c++ has no C++11 support" with g++-4.8 and 4.9

saihv, 2015-11-16:
Tried it with both gcc/g++ 4.8 and 4.9, and the update-alternatives was set correctly as well; yet libmavconn complains about C++11 support. I am on Linux Mint 17.2 (based off of Ubuntu 14.04)

vooon, 2015-11-16:
Check your environment variables, perhaps you have setted `CXX=g++4.5` or so.

Anyway mavros compile with GCC 4.6 on Travis without a problem. So check what compiller used.
Also check `g++` alternative.

But you may simply do that:
```sh
export CC="gcc-4.9" CXX="g++-4.9"
catkin clean --all
catkin build
```

saihv, 2015-11-16:
Still does not work after setting the variable CXX and CC to 4.9. It's quite confusing.

vooon, 2015-11-16:
That command should print cmake coutput:
```sh
catkin build -v
```

Look for this line:
```
-- The C compiler identification is GNU 5.2.1
-- The CXX compiler identification is GNU 5.2.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
```

vooon, 2015-11-22:
Updates?

saihv, 2015-11-23:
So sorry about the delay, this completely slipped my mind. Here's what the verbose output says.

    -- The C compiler identification is GNU 4.9.3
    -- The CXX compiler identification is GNU 4.9.3
    -- Check for working C compiler: /usr/bin/cc
    -- Check for working C compiler: /usr/bin/cc -- works
    -- Detecting C compiler ABI info
    -- Detecting C compiler ABI info - done
    -- Check for working CXX compiler: /usr/bin/c++
    -- Check for working CXX compiler: /usr/bin/c++ -- works
    -- Detecting CXX compiler ABI info
    -- Detecting CXX compiler ABI info - done

    Failed to build package 'libmavconn' because the following command:

    # Command to reproduce:
    cd /home/sai/catkin_ws/build/libmavconn && /home/sai/catkin_ws/build/libmavconn/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j; cd -

    # Path to log:
    cat /home/sai/catkin_ws/build/build_logs/libmavconn.log

    Exited with return code: 2 

The log file is full of errors about keywords not being recognized.

saihv, 2015-11-23:
Update: I just cloned the latest commit and now it builds fine!

closed 2015-11-23 by saihv