mavlink/mavros #432: Assertion failed

AndreasAntener, 2015-11-17:
@vooon Since the update to 0.16.xx I sometimes get this error on launch:
```
[ INFO] [1447755664.993827991]: udp0: Remote address: 192.168.170.1:14545
[ INFO] [1447755665.016338185]: VER: 1.1: Capabilities 0x00000000000004eb
[ INFO] [1447755665.016468145]: VER: 1.1: Flight software:     00000000 (c314ea3eafcfcdcd)
[ INFO] [1447755665.016508326]: VER: 1.1: Middleware software: 00000000 (c314ea3eafcfcdcd)
[ INFO] [1447755665.016535656]: VER: 1.1: OS software:         00000000 (0000000000000000)
[ INFO] [1447755665.016557951]: VER: 1.1: Board hardware:      00000000
[ INFO] [1447755665.016582857]: VER: 1.1: VID/PID: 26ac:0011
[ INFO] [1447755665.016698187]: VER: 1.1: UID: 3432470831323533
[ INFO] [1447755665.016985892]: CON: Got HEARTBEAT, connected.
[ INFO] [1447755665.100706283]: IMU: High resolution IMU detected!
mavros_node: /usr/include/boost/smart_ptr/shared_ptr.hpp:653: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = sensor_msgs::Imu_<std::allocator<void> >; typename boost::detail::sp_member_access<T>::type = sensor_msgs::Imu_<std::allocator<void> >*]: Assertion `px != 0' failed.
================================================================================REQUIRED process [mavros-2] has died!
process has died [pid 28113, exit code -6, cmd /sitl/catkin_ws/devel/lib/mavros/mavros_node __name:=mavros __log:=/root/.ros/log/d26785da-8d14-11e5-b305-0242ac110001/mavros-2.log].
log file: /root/.ros/log/d26785da-8d14-11e5-b305-0242ac110001/mavros-2*.log
Initiating shutdown!
================================================================================
```

Maybe every 1 out of 4 launches it happens. Do you have an idea what that could be?

vooon, 2015-11-17:
Looks like assert in boost internals. That bad sign because boost is a ton of complex code...

Interesting that error happen in imu_pub. You could blacklist it for verification (but some other plugins will be broken).
Other possible source - you running out of memory.

AndreasAntener, 2015-11-19:
It's hard to reproduce, but I blacklisted most of the plugins, including imu_pub and just got the error again:
```
[ INFO] [1447927217.524614960]: FCU URL: /dev/ttySAC0:921600
[ INFO] [1447927217.526272126]: serial0: device: /dev/ttySAC0 @ 921600 bps
[ INFO] [1447927217.528877960]: GCS URL: udp://:14555@:14550
[ INFO] [1447927217.530297460]: udp1: Bind address: 0.0.0.0:14555
[ INFO] [1447927217.550894251]: Plugin 3dr_radio blacklisted
[ INFO] [1447927217.551038501]: Plugin actuator_control blacklisted
[ INFO] [1447927217.551127710]: Plugin cam_imu_sync blacklisted
[ INFO] [1447927217.755198793]: Plugin command loaded and initialized
[ INFO] [1447927217.971945960]: udp1: Remote address: 192.168.43.175:14550
[ INFO] [1447927218.231698877]: Plugin distance_sensor loaded and initialized
[ INFO] [1447927218.232249710]: Plugin ftp blacklisted
[ INFO] [1447927218.290797877]: Plugin global_position loaded and initialized
[ INFO] [1447927218.291276793]: Plugin image_pub blacklisted
[ INFO] [1447927218.291403793]: Plugin imu_pub blacklisted
[ WARN] [1447927218.310095043]: GP: No GPS fix
[ INFO] [1447927218.321514085]: Plugin local_position loaded and initialized
[ INFO] [1447927218.321951627]: Plugin manual_control blacklisted
[ INFO] [1447927218.322087377]: Plugin mocap_pose_estimate blacklisted
[ INFO] [1447927218.338171793]: Plugin param loaded and initialized
[ INFO] [1447927218.338508668]: Plugin px4flow blacklisted
mavros_node: /usr/include/boost/smart_ptr/shared_ptr.hpp:653: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = sensor_msgs::Imu_<std::allocator<void> >; typename boost::detail::sp_member_access<T>::type = sensor_msgs::Imu_<std::allocator<void> >*]: Assertion `px != 0' failed.
================================================================================REQUIRED process [mavros-2] has died!
process has died [pid 4054, exit code -6, cmd /home/px4/catkin_ws/devel/lib/mavros/mavros_node __name:=mavros __log:=/home/px4/.ros/log/54ecbe9a-8ea4-11e5-98f9-7062b89dbe3d/mavros-2.log].
log file: /home/px4/.ros/log/54ecbe9a-8ea4-11e5-98f9-7062b89dbe3d/mavros-2*.log
Initiating shutdown!
================================================================================
```

AndreasAntener, 2015-11-19:
Any way I can enable more output to find the actual location of the error?

AndreasAntener, 2015-11-19:
It actually happens every time now (since imu_pub is blacklisted), so it could be somewhere where imu data is used

AndreasAntener, 2015-11-19:
Ok, the only other use I found is in `local_position`, and if I blacklist that I don't get the error anymore. So we have a race condition here.

PR #434 (2015-11-19): Use safe methods to get IMU data in local_position plugin
Fixes #432
@scott-eddy : your latest change in the `local_position` plugin introduced a race condition that was previously handled by using "safe" methods to get the IMU data from `uas`.
@vooon: This PR adds another fallback method for velocity if that's your preferred solution. @scott-eddy it would probably be good if you give this a test.
closed 2015-11-19 by vooon