mavlink/mavros #433: Sending attitude commands to the Pixhawk using mavros

hridaybavle, 2015-11-17:
Hello all, 
I am trying to control the pixhawk with px4 native stack using an external stack. The external stack is 
installed on an odroid and communicates serially with the pixhawk.
I am not connecting the px4flow to the pixhawk, but connecting it using a USB to the odroid, so that I can send the px4flow data serially to the external stack (I know I can send px4flow data using mavros by connecting it to the i2c port on the pixhawk)

The external stack needs rotation angles (imu data) and the px4flow data and it can then send the roll, pitch, yaw and throttle commands to a UAV. I want to implement this external stack as a high level controller on the pixhawk. 

I am successfully able to get the imu data from the pixhawk, I can also successfully send throttle from the external stack by using the topic "mavros/setpoint_attitude/att_throttle". 

The problem I am facing now is controlling the roll, pitch and yaw. I want to send the roll, pitch and yaw values to the pixhawk in the same way as one can be send it by using an RC.
Since I am not connecting the px4flow or the GPS to the pixhawk so the local position estimate being published wont be accurate (I dont know if I am correct with this one) and hence if I send the commands using the topic "mavros/setpoint_velocity/cmd_vel" or using "mavros/setpoint_attitude/cmd_vel" topics I don't see any commands being accepted by the pixhawk properly. 
I also figured that the topic "mavros/actuator_control" is not the correct one to send the roll, pitch, yaw commands in my case because I want to send commands to the pixhawk as they are sent by an RC but by using the external stack and not directly to the actuators.  

So my questions are
1. Do I always have to connect a px4flow or a GPS to the pixhawk when using offboard mode or I can use the setup mentioned above?
2. Is there anyway I can send the attitude commands to the pixhawk with mavros using the some method like manual control (except in this case the external stack would be sending this commands to the pixhawk instead of an RC) ? 

I know there is topic called rc_override but since I am using px4 native stack I cannot use this topic as well.

Thanks.

closed 2015-11-19 by hridaybavle

vinhk, 2017-02-15:
@hridaybavle Could you share your code? I am working on it as well. I do not have GPS. My issue is that attitude control did not make the quad fly. Although, the topics display the throttle value.

LArayane, 2017-07-13:
@vinhk  how could you solve the problem!! which topic you are publishing in because I don't have GPS either

mandelyoni, 2018-01-26:
Hi Hriday,

How are you using 'actuator_control' topic to command the desired Roll, Pitch, Yaw and Throttle values, directly? Can you upload a short publisher example?

Thanks, Yoni.

vinhk, 2018-01-26:
@LArayane excuse me for the late reply. I don't it working at this point. Hoping the other user may help.

mandelyoni, 2018-01-27:
Ok, thanks.