mavlink/mavros #437: Sending Position and Yaw Commands to Px4

austinsteamboat, 2015-11-24:
Hey,
I'm trying to write position set-point commands as well as a yaw set-point command to a quadcopter running px4+pixhawk. I'm flying indoors using a Vicon system, and I've worked out all the coordinate transforms so I've got px4's local position estimator lining up with the Vicon's coordinates. I'm publishing the Vicon translated local x,y,z coordinates to the px4 and it's accepting those. I can reliably command the quad to fly say 1 m above the origin with 0 yaw. Now I'm working on getting the orientation information plugged in and I'm having some difficulty:

1. I want to feed Vicon's orientation information into px4 using the vision_pose topic's quaternion field. The controller seems to accept the x,y,z position fine, but it doesn't seem to be taking the quaternion orientation from Vicon, are there additional parameters/fields I have to set for px4 to accept vision based orientation?

2. I want to command a local x,y,z location as well as yaw. I'm publishing to the setpoint_postion/local topic and I can command an x,y,z position and I can close the loop using the Vicon system to input local x,y,z coordinates. So the question now is how do I command non-zero yaw? In the px4 firmware, it's clear it's looking for a yaw set point as part of the x,y,z position set point. Do I just need to translate my desired yaw command to quaternions and input it that way in the setpoint_position/local's orientation fields? That said, it explicitly states in the Mavros ROS wiki page that orientation is not used, though. I've been reading through the code, forums, and documentation, but things seem to be changing pretty rapidly and the documentation doesn't seem to be consistent with what's actually going on. So any help would be greatly appreciated and I'd be happy to elaborate on all of this. 

Thanks,
Austin

vooon, 2015-11-24:
Yaw is extracted from orientation, but other axis is unused. Need to fix documentation. You may see it in `~setpoint_raw/target_local` loopback topic.

About firmware side please ask at `px4-users` google group. I never used any mocap systems, so i know not much.

LorenzMeier, 2015-11-24:
http://dev.px4.io/external-position.html

austinsteamboat, 2015-11-24:
Thanks Lorenz that answers that question exactly!

Regarding the yaw set-points, though, just to clarify, that orientation is in quaternion format, but the px4 firmware is taking in a yaw set point in it's multi-copter position controller:

https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp

line 797

So that seems like it's an Euler angle not a quaternion. So when I want to command a yaw setpoint as an Euler angle in radians, do I need to convert to quaternions to write to the mavros topic which will then convert back to an Euler angle? Then when you say the other axes are unused, I would still need to write to q.z and q.w to represent my yaw (according to a stack overflow post I saw), but I just wouldn't need to send q.x and q.y? 

Also to clarify,  as some of the documentation is outdated, I saw this on the px4 website:

https://pixhawk.org/dev/ros/mavros_offboard

It seems to suggest that we need to blacklist the mavros control topics we aren't using, is that still the case?

Thanks,
Austin

AlexisTM, 2016-03-02:
Dear,

It depends of the language you use, but in Python, with ROS (tf.transformation library) it will looks like : 

```
Input_quaternion = ( .... ) # From the Pixhawk
My_measured_yaw = ( ... ) # From Vicon or other source
roll, pitch, yaw = euler_from_quaternion(Input_quaternion , axes="sxyz") # Roll pitch and yaw in radians extracted from the PixHawk 
q = quaternion_from_euler(roll, pitch, My_measured_yaw , axes="sxyz") # Replace the yaw
```

So, for your message, send your position with the whole new quaternion computed.

closed 2016-06-23 by vooon