mavlink/mavros #438: Coordinate Frame Mismatches

austinsteamboat, 2015-11-24:
So I still have an issue with all of this ENU NED business. I took the pixhawk out with a laptop and a GPS device, monitored the /mavros/local_position/local topic, and walked around. Taking the arrow on the pixhawk to be forward flight direction I walked North and saw the x-field of the local position estimate in Mavros increase. I walked East and saw the y-field decrease. I raised the laptop and saw the z-field increase. This indicates to me that XYZ maps to NWU, not NED, and not ENU. I've jumped through a lot of hoops to correct this throughout the code-base I'm building for quadrotor control. I guess I just think someone should indicate this better in the documentation or correct the documentation or correct it in the code.

vooon, 2015-11-24:
Coordinate frame transforms is much painful topic. Perhaps current one rule for all transforms is wrong, but i want detailed investigation and full list of message - frame used.

That much tied to firmware side, and because we support both PX4 and APM (Copter, Plane and Rover) that little difficult to support (seek & workaround) differences.

For most messages current rule (rotate X + 180°) is right,
while proper ENU-NED require rotate X +180° & Y +90°.

austinsteamboat, 2015-11-24:
I guess to be clear, it would be ENU if the pixhawk was mounted sideways. I'll be commanding set-points in a local xyz coordinate frame so from a standpoint of pure translation I'm not so concerned, but this could have an effect when I'm giving yaw setpoints. So maybe a different question would be:

1. Is 0 yaw aligned with true North? And is that North aligned with the arrow on the pixhawk's body or is it actually assuming we're mounted sideways? 

Thanks,
Austin

LorenzMeier, 2015-11-24:
@austinsteamboat Do you have a robotics / academic background? The reason I ask is that I'd like to drill down on this one and get it properly evaluated. I have followed the conversion conversations loosely but I feel I need to spend some quality time on this and help to get it the rigour it requires.

@tfoote If I walk in the earth-fixed ROS coordinate frame into the zero degree orientation, in which axis (X, Y, Z) direction am I walking? I just want to be sure there is no fundamental confusion on our end.

How do you put your ENU frame into GPS coordinates? Where is zero degrees (north or east?)?

austinsteamboat, 2015-11-24:
Hi Lorenz,
Yes to both. We're trying to fly indoors without GPS using either a Vicon motion capture system and eventually a visual SLAM system, so establishing what 0 yaw is relative to is quite important. To reiterate, I tested this out using a GPS initially by monitoring the /mavros/local_position/local topic. I moved North, West, East, and South and recorded how the local x,y,z estimates changed, and again found x,y,z maps to N,W,U. I took the arrow on the pixhawk to be forward flight direction. I did not look very closely into the orientation data when I ran that test. I can check that out, though, and let you know what I find. 
Austin

tfoote, 2015-11-25:
In a globally referenced ENU frame, I'd expect the X to point east, Y north, and Z up. So zero degrees yaw would be about the z axis pointing towards the east. Wrapping counter clockwise following right hand rule. 

Converting ENU units into a globally referenced position such as lat/long you need to use pretty complicated systems to get from the euclidian space to the spherical coordinates. There are many different representations of lat/long (different datums etc) as well as many different representations. I highly recommend using something like a utm conversion library. 

@austinsteamboat In the topic name I see 'local' did you start your vehicle pointing north? Local would suggest to me that it's not a globally referenced and typically it will be relative to the startup position. 

Other resources: [geodesy metapackage](http://wiki.ros.org/geodesy) And there's the in process PR here connected to this: https://github.com/ros-infrastructure/rep/pull/103/files

austinsteamboat, 2015-11-25:
@tfoote the topic was a local estimate, the px4 generates a local position estimate by either converting from GPS, motion capture, or a visual SLAM or visual odometry system. I used GPS as the motion capture system has it's own coordinate frame issues we were debugging. The vehicle was me with a laptop outside. I don't recall if I had the pixhawk pointed North when I started it up. I'll repeat the test with it pointing East to see if that changes anything. So you're saying the expected behavior is to set the local frame relative to a starting location and orientation? Is there a way to set the local frame's origin and 0-degree yaw in software?

LorenzMeier, 2015-11-25:
Thanks Tully!

@austinsteamboat This suggests that the conversion of mavros is not correct with respect to the ROS standards.

ilia-sheremet, 2015-11-30:
Hey! 

I had the similar issue today with Mavros 0.16.3. When I publish to mavros/setpoint_position/local 
"x=1;y=0;z=current_z" copter flies towards the West, while giving "x=0;y=1;z=current_z" as an input the direction of flight is directly to the North. So it seems that current implementation utilize WNU, but not ENU.

austinsteamboat, 2015-11-30:
So I did some experiments and determined a couple things:

1. Using a pixhawk with a GPS, I verified that I'm seeing the local x,y,z estimates map to N,W,U (not WNU). 
2. Using a GPS, 0-yaw orients to true North with again, the local x,y,z estimates orienting to N,W,U. 
3. Without a GPS, 0-yaw seems to orient itself to magnetic North. This was tricky to test indoors with sources of magnetic interference around, but watching the raw magnetometer data and the local orientation data seems to confirm this. 
4. I ran tests with and without a GPS powering up the pixhawk in different orientations to confirm that in fact, 0-yaw is not set relative to a starting location/orientation, but is in fact set based on again magnetic or true North. 

So a question:

1. Is there a way to set a GPS home location through Mavros so the pixhawk can correct for true North even without a GPS module attached? Will the command home service do that?  

Thanks,
-Austin

ilia-sheremet, 2015-12-01:
Sorry, my bad, I got the same reference frame, namely N,W,U. Flying without GPS referencing only to magnetometer.

rosn, 2015-12-21:
I had some issue with wrong ENU to NED conversion (resulting in a real bad crash) so I wanted to see if some correction were to come. If you want has well, I did these corrections, so we could continue our project safely. If you want I can fork the solution. We have been flying with it for a month and no more problems appeared. The thing is vision orientation configuration still need to be tested. Thanks for your attention.

mhkabir, 2016-01-16:
@rosn Could you open a PR with your changes?

scott-eddy, 2016-01-19:
Everyone, testing on #473 would be appreciated.

scott-eddy, 2016-03-07:
@vooon this should be fixed in #473, okay to close

austinsteamboat, 2016-03-07:
I can confirm that the coordinate frames did change with the latest mavros updates. I haven't checked it to confirm the new coordinate frames. I'm good with closing it.

closed 2016-03-07 by vooon