mavlink/mavros #444: Issue with Eigen::Vector3d UAS::quaternion_to_rpy(const Eigen::Quaterniond &q)

jespestana, 2015-11-27:
I suggest an alternative implementation to the function
`Eigen::Vector3d UAS::quaternion_to_rpy(const Eigen::Quaterniond &q)`
From the source file (I provide the link to the current commit/implementation for future reference):
[mavros/mavros/src/lib/uas_quaternion_utils.cpp](https://github.com/mavlink/mavros/blob/8cfdd89aa67e21a4a6758a5225bc3a8c184608ed/mavros/src/lib/uas_quaternion_utils.cpp#L41).

As a small background: the calculated yaw, pitch and roll are intrinsic rotations in the ZYX axes: (First) rotation of yaw around the Z axis, (Second) rotation of pitch around the Y axis of the moving frame and (Third) rotation of roll around the X axis of the moving frame.

The problem is that the EulerAngles implementation from Eigen would give the following output range for the {yaw, pitch, roll} angles: yaw in [0,pi], pitch in [-pi,+pi] and roll in [-pi,+pi].

I think that the most desirable output would be in these ranges for the {yaw, pitch, roll} angles: yaw in [-pi,pi], pitch in [-pi/2,+pi/2] and roll in [-pi,+pi]. This is even more the case given that the YPR-ZYX euler angles present the gimbal-lock at `pitch=-pi/2` and `pitch=+pi/2`. The calculation suggested in this wikipedia article [Conversion between quaternions and Euler angles](https://en.wikipedia.org/wiki/Conversion_between_quaternions_and_Euler_angles#Conversion) gives an output on this desirable ranges. Its use was already suggested in the code for the function
`double UAS::quaternion_get_yaw(const Eigen::Quaterniond &q)`, but somehow not used for `quaternion_to_rpy`.

The resulting code, which I already tested and only fails at the gimbal-lock (unavoidable problem), would be the following:
```cpp
Eigen::Vector3d drone_space_utils::quaternion_to_rpy(const Eigen::Quaterniond &q)
{
    // Following the example of Vladimir Ermakov, I use the equations from this wikipedia article
    // https://en.wikipedia.org/wiki/Conversion_between_quaternions_and_Euler_angles
    // Note that the Eigen implementation is not good for YPR, because of the output range, which would be:
    //    [Eigen EulerAngles implementation] yaw, pitch, roll in the ranges [0:pi]x[-pi:pi]x[-pi:pi]

    const double &qw = q.w();
    const double &qx = q.x();
    const double &qy = q.y();
    const double &qz = q.z();

    Eigen::Vector3d rpy;
    /* roll  */ rpy.x() = std::atan2(2. * (qw*qx + qy*qz), 1. - 2. * (qx*qx + qy*qy));
    double sin_pitch = 2. * (qw*qy - qz*qx);
    sin_pitch = sin_pitch >  1.0 ?  1.0 : sin_pitch;
    sin_pitch = sin_pitch < -1.0 ? -1.0 : sin_pitch;
    /* pitch */ rpy.y() = std::asin( sin_pitch  );
    /* yaw   */ rpy.z() = std::atan2(2. * (qw*qz + qx*qy), 1. - 2. * (qy*qy + qz*qz));
    return rpy;
}
```

Let me know what do you think about this :-),
Am I missing something?

TSC21, 2017-04-04:
@jespestana have you tried that implementation of yours with acceptable results? If yes, please issue a PR. Closing for now.

closed 2017-04-04 by TSC21

jespestana, 2017-07-24:
@TSC21 I just issued a pull request:
PR https://github.com/mavlink/mavros/pull/761

I have tested the "new implementation" with the test cases (only with minor modifications):
```
mavros/mavros/test/test_quaternion_utils.cpp
  TEST(UAS, quaternion_to_rpy__123)
  TEST(UAS, quaternion_to_rpy__full_yaw_range)
  TEST(UAS, quaternion_to_rpy__pm_pi)
```