mavlink/mavros #446: mavros+ardupliot SITL: takeoff failed

CrazyPopLin, 2015-12-03:
I am using Ubuntu 14.04 to test ardupliot SITL to takeoff through mavros.
I run `/mavros/global_positn/global`, I got 
**latitude: 0.0**
**longitude: 0.0**
**altitude: -0.17**
Then
`rosrun mavros mavsafety arm`
`rosrun mavros mavsys mode -c GUIDED`
`rosrun mavros mavcmd takeoff 0 0 0 0 2`
The performance is takeoff first and then randomly fly in this [video](https://www.youtube.com/watch?v=grrwJ41Kicw).
Then I add `rosrun mavros mavsetp local -p 0 0 3 0`, it seems has same result.
I am confused:
1: **takeoff** command will lead to quad hover at the altitude?
2: if I set **latitude** and **longitude** both to zero, is that supported to takeoff from initial location?
3: **setpoint_position** can be used for leading quad to move based on local position of quad, maybe it has error, but it should be works, right?
4:when I run `rostopic echo /mavros/local_position/local`, I did't get anything. Someone said the new version of mavros replace that with **/mavros/local_position/pose**, should I install mavros from source again?
Thx for your help. :)

vooon, 2015-12-03:
1. what you expect? Copter should takeoff to some altitude and position, hovering is logical state at command ending.
2. I don't think that there any limitation (other than wrapping to ±π)
3. If you in right flight mode. You may emulate takeoff in local coordinates.
4. In release 0.16 name is changed, but you should be warned about unknown topic. I'm always recommend to use latest release…

CrazyPopLin, 2015-12-04:
For the `cmd/takeoff` command, I mean the copter takeoff to that altitude and keep there? Because I test that and just fly away. Please see that link. Could you give me some suggestion for that performances?

For the `setpoint_position`command, it can implement copter to takeoff and keep in that position?  

Thx.

vooon, 2015-12-04:
Check state of `~global_position/raw/fix` states of lat/long. Looks like copter go away to 0/0.
Also please try `setpoint_position_demo` from test_mavros (but check that it uses right local_position topic.).

CrazyPopLin, 2015-12-08:
I check the gps, the copter location is 0/0. Any suggestions? Thanks.
**/mavros/global_positi/raw/fix**
header: 
  seq: 855
  stamp: 
    secs: 489
    nsecs: 465000002
  frame_id: fcu
status: 
  status: 0
  service: 1
latitude: 0.0
longitude: 0.0
altitude: 0.05
position_covariance: [1.4641000923156753, 0.0, 0.0, 0.0, 1.4641000923156753, 0.0, 0.0, 0.0, 5.856400369262701]
position_covariance_type: 1

CrazyPopLin, 2015-12-16:
@alexbuyval, could you see this issue? The gps is publish to 0/0. But the simulation should be`Starting up at CMAC : -35.363261,149.165230,584,353`. Is that problem make the copter takeoff and fly away?
Thanks.

alejodosr, 2015-12-17:
I have the same problem with that SITL, when taking off it starts to fly randomly.. I have tried to change to AUTO mode to load a mission but no success. Hope it is solved soon

alexbuyval, 2015-12-21:
@CrazyPopLin Hi, sorry for delay. In fact, I don't use the GPS sensor in the gazebo model, so the model send zero coordinates always. Also for some reasons the model has significant drift, so you need the control node. 
Please, pay attention on this [post](http://diydrones.com/profiles/blogs/705844:BlogPost:2151758?id=705844%3ABlogPost%3A2151758&page=1). I think it is more powerful.

LorenzMeier, 2016-01-02:
This is how you connect MAVROS: http://dev.px4.io/simulation-ros-interface.html

closed 2016-02-05 by vooon

cs2r, 2019-02-05:
I solve this issue by getting the latitude and the longitude from `/mavros/global_position/global` then pass it in the service request.