mavlink/mavros #447: Publishing to mavros/mocap/pose no longer updates /mavors/local_position/pose

BrownKyle, 2015-12-04:
I was previously able to publish to the topic /mavros/mocap/pose and the topic /mavros/local_position/pose would then be updated with accurate position data from the motion capture system.

Recently this hasn't been working. Previously I would check the topic /mavros/local_position/local to see the position. It appears there is a change in topic names, how would I adapt to this. The topic /mavros/mocap/pose hasn't changed except that it appears to be loaded by default.

Additionally I started getting these warnings, are they related?
```
[ INFO] [1449239158.600693616]: FCU URL: /dev/ttyAMA0:921600
[ INFO] [1449239158.603954657]: serial0: device: /dev/ttyAMA0 @ 921600 bps
[ INFO] [1449239158.607210647]: GCS bridge disabled
[ INFO] [1449239158.989941480]: Plugin 3dr_radio loaded and initialized
[ INFO] [1449239159.028172209]: Plugin actuator_control loaded and initialized
[ INFO] [1449239159.041161428]: Plugin altitude loaded and initialized
[ INFO] [1449239159.552757730]: Plugin cam_imu_sync loaded and initialized
[ INFO] [1449239159.616813407]: Plugin command loaded and initialized
[ INFO] [1449239159.617621897]: Plugin distance_sensor blacklisted
[ INFO] [1449239159.705131636]: Plugin ftp loaded and initialized
[ INFO] [1449239159.799604397]: Plugin global_position loaded and initialized
[ INFO] [1449239159.801138980]: Plugin image_pub blacklisted
[ INFO] [1449239159.887386792]: Plugin imu_pub loaded and initialized
[ INFO] [1449239159.902212001]: IMU: High resolution IMU detected!
[ INFO] [1449239159.958247417]: Plugin local_position loaded and initialized
[ INFO] [1449239159.969667886]: Plugin manual_control loaded and initialized
[ INFO] [1449239160.043462834]: Plugin mocap_pose_estimate loaded and initialized
[ INFO] [1449239160.085020594]: Plugin param loaded and initialized
[ INFO] [1449239160.170286271]: Plugin px4flow loaded and initialized
[ INFO] [1449239160.238307990]: Plugin rc_io loaded and initialized
[ INFO] [1449239160.239867365]: Plugin safety_area blacklisted
[ INFO] [1449239160.300703198]: Plugin setpoint_accel loaded and initialized
[ INFO] [1449239160.497982053]: Plugin setpoint_attitude loaded and initialized
[ INFO] [1449239160.595726063]: Plugin setpoint_position loaded and initialized
[ INFO] [1449239160.761701792]: Plugin setpoint_raw loaded and initialized
[ INFO] [1449239160.809501792]: Plugin setpoint_velocity loaded and initialized
[ INFO] [1449239160.908234396]: Plugin sys_status loaded and initialized
[ INFO] [1449239160.960347313]: Plugin sys_time loaded and initialized
[ INFO] [1449239160.981905073]: Plugin vfr_hud loaded and initialized
[ INFO] [1449239160.983911115]: Plugin vibration blacklisted
[ INFO] [1449239161.144646115]: Plugin vision_pose_estimate loaded and initialized
[ INFO] [1449239161.205342365]: Plugin vision_speed_estimate loaded and initialized
[ INFO] [1449239161.271763042]: Plugin waypoint loaded and initialized
[ INFO] [1449239161.273166583]: Autostarting mavlink via USB on PX4
[ INFO] [1449239161.273677313]: Built-in SIMD instructions: None
[ INFO] [1449239161.274002729]: Built-in MAVLink dialect: ardupilotmega
[ INFO] [1449239161.274299396]: MAVROS started. MY ID [1, 240], TARGET ID [1, 1]
[ INFO] [1449239161.807084240]: CON: Got HEARTBEAT, connected.
[ INFO] [1449239161.817739031]: IMU: High resolution IMU detected!
[ WARN] [1449239163.822450385]: VER: broadcast request timeout, retries left 4
[ WARN] [1449239164.822005853]: VER: broadcast request timeout, retries left 3
[ WARN] [1449239165.821071790]: VER: unicast request timeout, retries left 2
[ WARN] [1449239166.821488404]: VER: unicast request timeout, retries left 1
[ WARN] [1449239167.820813404]: VER: unicast request timeout, retries left 0
[ WARN] [1449239168.821118299]: VER: your FCU don't support AUTOPILOT_VERSION, switched to default capabilities
[ WARN] [1449239172.810141892]: PR: request list timeout, retries left 2
[ WARN] [1449239173.811674860]: PR: request list timeout, retries left 1
[ WARN] [1449239174.812907360]: PR: request list timeout, retries left 0
[ WARN] [1449239177.811937775]: WP: timeout, retries left 2
[ WARN] [1449239178.813818816]: WP: timeout, retries left 1
[ WARN] [1449239179.815454285]: WP: timeout, retries left 0
[ERROR] [1449239180.817174961]: WP: timed out.
```

Any help is appreciated.

vooon, 2015-12-04:
1. mavros version?
2. fcu/firmware version?
3. what changed recently?

BrownKyle, 2015-12-04:
1. Tried git pull in macros directory and it stated Already up-to-date. So the latest version. (Is there a better way to get version?)

2. 3DR Pixhawk, with PX4 firmware master build installed a 4 days ago.

3. Biggest change was updating PX4 firmware. I was able to get the position data to improve the estimate in /mavros/local_position/local, by publishing the motion capture data to /mavros/mocap/pose. But the quad rotor was very unstable in manual flight so I never tried the autonomous tests even though I have the script and am able to put the quad into off board mode. The latest master build resolved the control issues. I also updated macros, I saw a change in topics and the VER, PR and WP warnings were new.

vooon, 2015-12-04:
PR & WP — check IDs.

BrownKyle, 2015-12-04:
What about VER?

vooon, 2015-12-04:
Two possibilities:

1. You have firmware older than 3-5 month
2. PX4 still do not support broadcast version request, so ID mismatch possible reason.

BrownKyle, 2015-12-04:
The firmware is the master fr four days ago, so it must ID issues. I'll check tomorrow and post the results.

vooon, 2015-12-04:
Check new script from #382.

BrownKyle, 2015-12-04:
Do I just rosrun that before connecting?

vooon, 2015-12-04:
It require started mavros.

BrownKyle, 2015-12-04:
Ok, thanks.

BrownKyle, 2015-12-05:
@vooon, I ran your new script. I got:
```
OK. I got your message from 1:1.
```

From your script those numbers are the target and system numbers, as they both match does this mean something else is the issue?

vooon, 2015-12-05:
Hmm, interesting. Did you tried `--verbose` mode?

BrownKyle, 2015-12-05:
Nope, I'll try it on Monday, I'm no longer get in the lab.

BrownKyle, 2015-12-07:
I used a different raspberry pi and a pixhawk clone. I installed the latest macros and px4 firmware and get the same errors in the command line when I start mavros. When I run your checked script with --verbose I get:

    OK. I got messages from 1:1
    ___
    Received 815 messages, from 1 dresses
    sys:comp    list of messages
        1:1         32, 0, 2, 36, 1, 105, 74, 140, 111, 76, 83, 147, 87, 30, 245

Although its a different setup I am getting the same errors, so its likely the same issue. But I'll stick check with the quad I had the issues on tomorrow.

vooon, 2015-12-07:
Try another firmware. Also ask at px4-users google group. I think that it is firmware problem.
Other possible error source - usage of different mavlink definitions, but *MISSION_xyz* messages not changed while ago (and support politic prohibit changing stable & deployed messages).

BrownKyle, 2015-12-07:
I tired both then px4 firmware beta and stable and both result in similar sets of numbers.

BrownKyle, 2015-12-07:
@vooon I tried to connect via usb and and don't get the same errors, the VER waring doesn't reach zero (assuming thats means it eventually works) no PR or WP warnings. Only error is FCU: WP CMD:partner id mismatch and warning about clock skew.

This was on the stable build from sudo apt-get (debugging, trying older versions of mavros).

It appears the raspberry pi 2 (ubuntu 14) connecting with GPIO to telem2 is the issue, it previously worked. I can't use USB in flight, do you know of any ways to fix the link?

LorenzMeier, 2015-12-07:
@BrownKyle Have you set SYS_COMPANION on Pixhawk to the same baud rate as used by MAVROS? 

https://pixhawk.org/dev/companion_link

BrownKyle, 2015-12-07:
@LorenzMeier yes

BrownKyle, 2015-12-07:
I tested my setup in the lab, connecting with USB works great. I only get a clock skew warning, but am able to publish data to /mavros/mocap/pose and getting 
    [inav] MOCAP data valid 
I can also see /mavros/local_position/pose get updated with accurate position readings from the motion capture system. 

The issue seems to be the link between the raspberry pi 2 GPIO and the telem2 port.

I was previously following the DroneKit guide to the companion computer setup with ground, power, TX and RX and it worked. But the guide @LorenzMeier posted suggest using the RTS and CTS pins, as the Raspberry Pi does not seem to have them shall I try to use serial port 4 instead of telem2. Or use an USB to FTDI converter?

vooon, 2015-12-07:
@BrownKyle did you fixed your rpi2 config.txt file?

https://github.com/vooon/rarog/blob/master/rarog_configs/rpi2/config.txt#L45
```
# Higher UART Speed (PX4)
init_uart_baud=921600
init_uart_clock=14745600
```

Without that lines UART do not work at speeds higher than 115200.

BrownKyle, 2015-12-07:
@voon Yeah, and I double checked just there.

BrownKyle, 2015-12-07:
@voon I don't have the Max power USB or the tty AMA rules you have, is it worth implementing those.

vooon, 2015-12-07:
Max USB power not required, just allow to draw up to 2 Amps from power source (i used dedicated SBEC for OBC). And be sure not connect to pixhawk's 5V pin. It can't power OBC.

BrownKyle, 2015-12-07:
I made the USB-FTDI-TELEM2 cable described in https://pixhawk.org/dev/companion_link  and it worked great, I only have a warning about clock stew (that appears to correct itself). I haven't tested with Mocap, but I tested USB-micorUSB and it removed the warnings and the Mocap data worked. So its safe to assume that it should now work.

My issue seems to be with using the Raspberry Pi 2 GPIO pins when it runs Ubuntu. With rasbian, the exact same setup is able to connect using mavproxy.py. 

I am unsure why it no longer works, as it previously did. But I have repeated the issue on two separate raspberry pi's and two separate pixhawks and haven't found out the fault.

closed 2015-12-07 by BrownKyle

BrownKyle, 2015-12-08:
```
[ INFO] [1449592934.384396801]: FCU URL: /dev/ttyUSB0:921600
[ INFO] [1449592934.387841540]: serial0: device: /dev/ttyUSB0 @ 921600 bps
[ INFO] [1449592934.395005342]: GCS bridge disabled
[ INFO] [1449592934.848557738]: Plugin 3dr_radio loaded and initialized
[ INFO] [1449592934.885238571]: Plugin actuator_control loaded and initialized
[ INFO] [1449592934.899198519]: Plugin altitude loaded and initialized
[ INFO] [1449592935.453258883]: Plugin cam_imu_sync loaded and initialized
[ INFO] [1449592935.523122061]: Plugin command loaded and initialized
[ INFO] [1449592935.525167373]: Plugin distance_sensor blacklisted
[ INFO] [1449592935.621108363]: Plugin ftp loaded and initialized
[ INFO] [1449592935.711636592]: Plugin global_position loaded and initialized
[ INFO] [1449592935.712714977]: Plugin image_pub blacklisted
[ INFO] [1449592935.796840967]: Plugin imu_pub loaded and initialized
[ INFO] [1449592935.813116644]: IMU: High resolution IMU detected!
[ INFO] [1449592935.855689040]: Plugin local_position loaded and initialized
[ INFO] [1449592935.865316696]: Plugin manual_control loaded and initialized
[ INFO] [1449592935.924371748]: Plugin mocap_pose_estimate loaded and initialized
[ INFO] [1449592935.959137373]: Plugin param loaded and initialized
[ INFO] [1449592936.033653571]: Plugin px4flow loaded and initialized
[ INFO] [1449592936.087089352]: Plugin rc_io loaded and initialized
[ INFO] [1449592936.088421539]: Plugin safety_area blacklisted
[ INFO] [1449592936.136407217]: Plugin setpoint_accel loaded and initialized
[ INFO] [1449592936.286887529]: Plugin setpoint_attitude loaded and initialized
[ INFO] [1449592936.366042008]: Plugin setpoint_position loaded and initialized
[ INFO] [1449592936.487624612]: Plugin setpoint_raw loaded and initialized
[ INFO] [1449592936.523502164]: Plugin setpoint_velocity loaded and initialized
[ INFO] [1449592936.606396279]: Plugin sys_status loaded and initialized
[ INFO] [1449592936.648465289]: Plugin sys_time loaded and initialized
[ INFO] [1449592936.657666487]: Plugin vfr_hud loaded and initialized
[ INFO] [1449592936.658719925]: Plugin vibration blacklisted
[ INFO] [1449592936.774137425]: Plugin vision_pose_estimate loaded and initialized
[ INFO] [1449592936.822295237]: Plugin vision_speed_estimate loaded and initialized
[ INFO] [1449592936.873215810]: Plugin waypoint loaded and initialized
[ INFO] [1449592936.874582893]: Autostarting mavlink via USB on PX4
[ INFO] [1449592936.875259612]: Built-in SIMD instructions: None
[ INFO] [1449592936.875677841]: Built-in MAVLink dialect: pixhawk
[ INFO] [1449592936.876071331]: MAVROS started. MY ID [1, 240], TARGET ID [1, 1]
[ INFO] [1449592936.990455237]: CON: Got HEARTBEAT, connected.
[ INFO] [1449592937.006237945]: IMU: High resolution IMU detected!
[ WARN] [1449592939.006680809]: VER: broadcast request timeout, retries left 4
[ WARN] [1449592940.006947528]: VER: broadcast request timeout, retries left 3
[ WARN] [1449592941.006628204]: VER: unicast request timeout, retries left 2
[ WARN] [1449592942.005943100]: VER: unicast request timeout, retries left 1
[ WARN] [1449592943.010262214]: VER: unicast request timeout, retries left 0
[ WARN] [1449592944.005909974]: VER: your FCU don't support AUTOPILOT_VERSION, switched to default capabilities
[ WARN] [1449592947.994636431]: PR: request list timeout, retries left 2
[ WARN] [1449592948.996063305]: PR: request list timeout, retries left 1
[ WARN] [1449592949.996947263]: PR: request list timeout, retries left 0
[ WARN] [1449592950.252050336]: TM: Clock skew detected (-1449592734.986129999 s). Hard syncing clocks.
[ INFO] [1449592951.998555804]: WP: mission received

[ INFO] [1449593081.421691901]: FCU: [inav] MOCAP data valid

```