mavlink/mavros #448: mavros build error

CrazyPopLin, 2015-12-05:
Hi, I am using odroid xu4 board with ubuntu 14.04 in arm. The previous environment that were installed 
**ros-indigo-base** and **ros-indigo-mavlink** which is 2015.11.11 version. And I have a ros packaged created.
I am trying to install mavros from source follow the **README.md**.
First four steps were fine(first step just run `wstool init src`), when I run `catkin build`, it said try it replace with catkin_make_isolated (build isolation required for mavlink as you said in #34 ).
After that have issues below:
>g++-4.8.real: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/mavros.dir/src/lib/uas_stringify.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
virtual memory exhausted: Cannot allocate memory
make[2]: *** [CMakeFiles/mavros.dir/src/lib/mavros.cpp.o] Error 1
g++-4.8.real: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/mavros.dir/src/lib/uas_quaternion_utils.cpp.o] Error 4
g++-4.8.real: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/mavros.dir/src/lib/uas_data.cpp.o] Error 4
make[1]: *** [CMakeFiles/mavros.dir/all] Error 2
make: *** [all] Error 2
<== Failed to process package 'mavros': 
  Command '['/home/odroid/skywalker/devel_isolated/mavros_msgs/env.sh', 'make', '-j8', '-l8']' returned non-zero exit status 2
>
>Reproduce this error by running:
==> cd /home/odroid/skywalker/build_isolated/mavros && >/home/odroid/skywalker/devel_isolated/mavros_msgs/env.sh make -j8 -l8

When I try `roscd mavros &mavlink`, it seems works and the path was in my ros package. I just wondering if it ok.
Thanks for your help. :)

andre-nguyen, 2015-12-05:
> virtual memory exhausted: Cannot allocate memory

Sounds like you need to decrease the number of cores/threads used to compile.

vooon, 2015-12-05:
You out of memory.
a) Reduce jobs count
b) Setup *swap* file
c) Setup *distcc*

Also you should get error with master and mavlink 2015.11.11 - you need >= 2015.11.25.

closed 2015-12-05 by vooon

CrazyPopLin, 2015-12-10:
Hi @vooon , I reinstall all the packages required by ardupliot SITL and the newest mavlink and mavros from source. using `catkin_build` to build them successfully. And I can roscd all the rospackage but mavlink. I know mavlink need plain cmake, so using cakin_make_isolated again, which make it works in my odroid. And I still can't use roscd to located new mavlink. Could you give me some suggestions. Thanks.

vooon, 2015-12-10:
I recommend to forgot all that `rosbuild` & `catkin_make` crap. Use [`catkin-tools`](https://catkin-tools.readthedocs.org/en/latest/) luke!

CrazyPopLin, 2015-12-11:
sorry, I am using `catkin build` like you said for installation and get the issues above.

vooon, 2015-12-11:
Do you update environment with this command?
```sh
source <path-to-ws>/devel/setup.bash
```

CrazyPopLin, 2015-12-14:
yes, because I can roscd to mavros packages but mavlink. I reinstall mavros and mavlink on the odroid and setup environments. I really get confused. Thanks for your help.
**printenv | grep ROS**:
>
ROS_ROOT=/opt/ros/indigo/share/ros
ROS_PACKAGE_PATH=/home/odroid/Skywalker/src/mavros/mavros_msgs:/home/odroid/Skywalker/src/mavros/libmavconn:/home/odroid/Skywalker/src/mavros/mavros:/home/odroid/Skywalker/src/mavros/mavros_extras:/home/odroid/Skywalker/src/mavros/test_mavros:/opt/ros/indigo/share:/opt/ros/indigo/stacks
ROS_MASTER_URI=http://localhost:11311
ROSLISP_PACKAGE_DIRECTORIES=/home/odroid/Skywalker/devel/share/common-lisp
ROS_DISTRO=indigo
ROS_ETC_DIR=/opt/ros/indigo/etc/ros
>

**catkin config**:
>
Profile:                     default
Extending:          [cached] /opt/ros/indigo
Workspace:                   /home/odroid/Skywalker
Source Space:       [exists] /home/odroid/Skywalker/src
Build Space:        [exists] /home/odroid/Skywalker/build
Devel Space:        [exists] /home/odroid/Skywalker/devel
Install Space:     [missing] /home/odroid/Skywalker/install
DESTDIR:                     None
>
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
>
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
>
Whitelisted Packages:        None
Blacklisted Packages:        None
>
Workspace configuration appears valid.
>

**catkin_find mavlink**:
>
/home/odroid/Skywalker/devel/include/mavlink
/home/odroid/Skywalker/devel/share/mavlink
/opt/ros/indigo/include/mavlink
/opt/ros/indigo/share/mavlink
>

vooon, 2015-12-14:
What error? Please describe, i do not see in that quotes.

CrazyPopLin, 2015-12-15:
it works. Last time when I `catkin_build_isolated` I can roscd to newest mavlink by accident. Maybe I confused the build tools. Thanks.

IngTIKNA, 2021-02-16:
I encountered the same problem. Your mavlink might require plain cmake. Therefore, make sure that you installed python packages : 

 **pip install future**
 pip3 install future

vooon, 2021-02-16:
@IngTIKNA future module is in mavlink dependency list, so rosdep should be able to install it if you follows "source installation" instructions.