mavlink/mavros #451: Rotate in yaw keeping the same alttitude

alquimist, 2015-12-09:
Hi all, 

I would like to get the support of the community, recently I am working with ros+mavros to get offboard control of a quadcopter. During some weeks I've been trying to get a simple behavior of rotation on the same place the uav, I men,  if the uav is flying at any geometric point, (0,0,10) linear part of a TwistStamped and with heading to the north, I want to keep the same position and rotate 45 degrees or any desired degrees. 

I tested the topics 

/mavros/local_position/local
/mavros/local_position/velocity
/mavros/setpoint_velocity/cmd_vel

Any o them and other tested let me rotate the uav in the same place even I make the transformation of the euler or the queaternion information.. 

somebody can help me and let me know what I'm doing grown?

Thanks by any help. 

best regars.

alquimist, 2015-12-09:
Additional comment, I'm using the package version 0.16.3 monitoring the results in mavproxy simulator

robnsngh, 2016-02-09:
Assuming you are using /mavros/setpoint_velocity/cmd_vel to command the mav, cmd_vel will send mav angular velocity for yaw. If you want to continue using cmd_vel for controlling the drone, you will have to poll the current heading and keep tracking the error until it hits 45 degrees and stop yawing. 

Alternatively, you can use /mavros/setpoint_position/local to directly command the yaw you desire in angles.

harewei, 2016-03-28:
@robnsngh could you give a more precise answer?  In mavros local_position/pose, there's a quaternion component [http://docs.ros.org/api/geometry_msgs/html/msg/Quaternion.html](url)

robnsngh, 2016-03-28:
@harewei You can use something like this cmd_pos.pose.orientation = tf::createQuaternionMsgFromYaw(pixhawkYaw);

where pixhawkYaw is yaw command in radians.

alquimist, 2016-03-28:
All, 

My apologies by slow reply, but I was working on other different things and I was now hable to response before. 

I tested the /mavros/setpoint_velocity/local, however saw (simulator SITL) UAV rotate when the altitude decrease or increase and give me other different behavior, also I remember use Quaternion to modify the UAV pose, but honesty I don't remember if the method used was createQuaternionMsgFromYaw.

I will double check with the info provided and I let know results I get, 

Thanks in advance..

alquimist, 2016-03-31:
@robnsngh 

Thanks in advance by you support, I have tested the /mavros/setpoint_position/local topic using Quaternion to modify the pose of the vehicle in the free space. Time before I had tested the topic and I double ckeck some minutes ago. After make a small code on c I have been sending a yaw to tf::createQuaternionMsgFromYaw(), Once I issue the quaternion value the copter look rotate, however it goes to 0 altitude (I mean, it goes to land) when the behavior I wish is keep the same altitude, for example, takeoff 40 over HOME and then rotate n degrees in order to change the heading direction. 

Thanks. 

Note, 
/mavros/local_position/local  & /mavros/local_position/velocity does not work because it change the heading according a tarjet point and I would like control the direction without any specific point from relative position.

closed 2016-06-23 by vooon

Reissss, 2023-04-21:
> @robnsngh
>
> Thanks in advance by you support, I have tested the /mavros/setpoint_position/local topic using Quaternion to modify the pose of the vehicle in the free space. Time before I had tested [...]

Guys if your copter goes down when you change the euler axis you have to assign the position x , y and z parameters to your copter's local positions