mavlink/mavros #452: Mavlink Dialect pixhawk does not exits anymore

devbharat, 2015-12-13:
Need to get rid of it from mavconn/mavlink_dialect.h

vooon, 2015-12-13:
What proffered: error or warning + fallback?

vooon, 2015-12-13:
Mavlink-gpb-release take care on providing known dialects. Cmake checks that correct setting set.

So `mavlink_dialect.h` not need to change.

devbharat, 2015-12-13:
If what was in pixhawk.XML was essentially the same as what we have in common.XML, the warning +fallback should be OK I guess.

vooon, 2015-12-13:
I rechecked - warning not possible now. Only build error.

vooon, 2015-12-13:
Fixed.

Note to users: if you has build error like that:
```
CMake Error at cmake/Modules/MavrosMavlink.cmake:27 (message):
  Unknown MAVLink dialect: pixhawk, known dialects:
  slugs;paparazzi;python_array_test;autoquad;ardupilotmega;test;ASLUAV;minimal;common;ualberta;matrixpilot
Call Stack (most recent call first):
  CMakeLists.txt:16 (include)
```

Then just use **common** dialect.

closed 2015-12-13 by vooon