mavlink/mavros #454: mavros catkin build or catkin_make problem

won13y, 2015-12-19:
1) First Question!

When I catkin_make about mavros,

The error is the following:

I don't know the reason?

wkyoun@wkyoun-XPS-13-9343:~/catkin_ws$ catkin_make

Base path: /home/wkyoun/catkin_ws
Source space: /home/wkyoun/catkin_ws/src
Build space: /home/wkyoun/catkin_ws/build
Devel space: /home/wkyoun/catkin_ws/devel
Install space: /home/wkyoun/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/wkyoun/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/wkyoun/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/wkyoun/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/wkyoun/catkin_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/wkyoun/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.16
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 48 packages in topological order:
-- ~~  - kobuki (metapackage)
-- ~~  - kobuki_capabilities
-- ~~  - kobuki_desktop (metapackage)
-- ~~  - kobuki_gazebo
-- ~~  - kobuki_rapps
-- ~~  - kobuki_rviz_launchers
-- ~~  - kobuki_soft (metapackage)
-- ~~  - kobuki_softapps
-- ~~  - libmavconn
-- ~~  - navigation (metapackage)
-- ~~  - mavros_msgs
-- ~~  - kobuki_keyop
-- ~~  - my_first_ros_pkg
-- ~~  - kobuki_bumper2pc
-- ~~  - oroca_ros_tutorials
-- ~~  - kobuki_auto_docking
-- ~~  - mavros
-- ~~  - fake_localization
-- ~~  - gazebo_ros
-- ~~  - kobuki_dashboard
-- ~~  - kobuki_softnode
-- ~~  - map_server
-- ~~  - amcl
-- ~~  - robot_pose_ekf
-- ~~  - gazebo_plugins
-- ~~  - kobuki_gazebo_plugins
-- ~~  - mavros_extras
-- ~~  - test_mavros
-- ~~  - voxel_grid
-- ~~  - costmap_2d
-- ~~  - nav_core
-- ~~  - base_local_planner
-- ~~  - carrot_planner
-- ~~  - clear_costmap_recovery
-- ~~  - dwa_local_planner
-- ~~  - move_slow_and_clear
-- ~~  - navfn
-- ~~  - global_planner
-- ~~  - rotate_recovery
-- ~~  - move_base
-- ~~  - kobuki_description
-- ~~  - kobuki_controller_tutorial
-- ~~  - kobuki_random_walker
-- ~~  - kobuki_safety_controller
-- ~~  - kobuki_node
-- ~~  - kobuki_testsuite
-- ~~  - kobuki_qtestsuite
-- ~~  - yujin_maps
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'kobuki'
-- ==> add_subdirectory(kobuki/kobuki)
-- +++ processing catkin package: 'kobuki_capabilities'
-- ==> add_subdirectory(kobuki/kobuki_capabilities)
-- +++ processing catkin metapackage: 'kobuki_desktop'
-- ==> add_subdirectory(kobuki_desktop/kobuki_desktop)
-- +++ processing catkin package: 'kobuki_gazebo'
-- ==> add_subdirectory(kobuki_desktop/kobuki_gazebo)
-- +++ processing catkin package: 'kobuki_rapps'
-- ==> add_subdirectory(kobuki/kobuki_rapps)
-- +++ processing catkin package: 'kobuki_rviz_launchers'
-- ==> add_subdirectory(kobuki_desktop/kobuki_rviz_launchers)
-- +++ processing catkin metapackage: 'kobuki_soft'
-- ==> add_subdirectory(kobuki_soft/kobuki_soft)
-- +++ processing catkin package: 'kobuki_softapps'
-- ==> add_subdirectory(kobuki_soft/kobuki_softapps)
-- +++ processing catkin package: 'libmavconn'
-- ==> add_subdirectory(mavros/libmavconn)
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
CMake Error at mavros/libmavconn/cmake/Modules/MavrosMavlink.cmake:4 (find_package):
  By not providing "Findmavlink.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "mavlink", but
  CMake did not find one.

  Could not find a package configuration file provided by "mavlink" with any
  of the following names:

    mavlinkConfig.cmake
    mavlink-config.cmake

  Add the installation prefix of "mavlink" to CMAKE_PREFIX_PATH or set
  "mavlink_DIR" to a directory containing one of the above files.  If
  "mavlink" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  mavros/libmavconn/CMakeLists.txt:16 (include)

-- Configuring incomplete, errors occurred!
See also "/home/wkyoun/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/wkyoun/catkin_ws/build/CMakeFiles/CMakeError.log".
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

2) Second Question

1) First Question!

When I catkin_build about mavros,

The error is the following:

I don't know the reason?

wkyoun@wkyoun-XPS-13-9343:~/catkin_ws$ catkin build

[build] Runtime: 0.0 seconds 
Error: Build space `/home/wkyoun/catkin_ws/build` exists but appears to have previously been created by the `catkin_make` or `catkin_make_isolated` tool. Please choose a different directory to use with `catkin build` or clean the build space

hridaybavle, 2015-12-19:
Your first error is because you have not installed mavlink. Please refer to issue #240 for installing a fresh copy of mavlink or mavros documentation. 
The second is error is because you are trying to use catkin_build command in a workspace where you have used catkin_make before. In order to use catkin_build command delete the build and devel folders in your workspace and then try catkin_build.

vooon, 2015-12-19:
1) https://github.com/mavlink/mavros/tree/master/mavros#source-installation

But because you used `catkin_make` do `catkin clean --all` first.
Perhaps init step unneeded too. Proceed to step 3.

won13y, 2015-12-20:
I tried to install mavlink as hridaybavie

But, another error is found

It states that control_toolbox package is the problem

please let me know the soultion

wkyoun@wkyoun-XPS-13-9343:~/catkin_ws$ catkin_make

Base path: /home/wkyoun/catkin_ws
Source space: /home/wkyoun/catkin_ws/src
Build space: /home/wkyoun/catkin_ws/build
Devel space: /home/wkyoun/catkin_ws/devel
Install space: /home/wkyoun/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/wkyoun/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/wkyoun/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/wkyoun/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/wkyoun/catkin_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/wkyoun/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.16
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 48 packages in topological order:
-- ~~  - kobuki (metapackage)
-- ~~  - kobuki_capabilities
-- ~~  - kobuki_desktop (metapackage)
-- ~~  - kobuki_gazebo
-- ~~  - kobuki_rapps
-- ~~  - kobuki_rviz_launchers
-- ~~  - kobuki_soft (metapackage)
-- ~~  - kobuki_softapps
-- ~~  - libmavconn
-- ~~  - navigation (metapackage)
-- ~~  - mavros_msgs
-- ~~  - kobuki_keyop
-- ~~  - my_first_ros_pkg
-- ~~  - kobuki_bumper2pc
-- ~~  - oroca_ros_tutorials
-- ~~  - kobuki_auto_docking
-- ~~  - mavros
-- ~~  - fake_localization
-- ~~  - gazebo_ros
-- ~~  - kobuki_dashboard
-- ~~  - kobuki_softnode
-- ~~  - map_server
-- ~~  - amcl
-- ~~  - robot_pose_ekf
-- ~~  - gazebo_plugins
-- ~~  - kobuki_gazebo_plugins
-- ~~  - mavros_extras
-- ~~  - test_mavros
-- ~~  - voxel_grid
-- ~~  - costmap_2d
-- ~~  - nav_core
-- ~~  - base_local_planner
-- ~~  - carrot_planner
-- ~~  - clear_costmap_recovery
-- ~~  - dwa_local_planner
-- ~~  - move_slow_and_clear
-- ~~  - navfn
[... 146 lines cut ...]
  installed.

-- Eigen found (include: /usr/include/eigen3)
-- Selected MAVLink dialect: ardupilotmega
-- +++ processing catkin package: 'test_mavros'
-- ==> add_subdirectory(mavros/test_mavros)
-- Using these message generators: gencpp;geneus;genlisp;genpy
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:75 (find_package):
  Could not find a package configuration file provided by "control_toolbox"
  with any of the following names:

    control_toolboxConfig.cmake
    control_toolbox-config.cmake

  Add the installation prefix of "control_toolbox" to CMAKE_PREFIX_PATH or
  set "control_toolbox_DIR" to a directory containing one of the above files.
  If "control_toolbox" provides a separate development package or SDK, be
  sure it has been installed.
Call Stack (most recent call first):
  mavros/test_mavros/CMakeLists.txt:7 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/wkyoun/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/wkyoun/catkin_ws/build/CMakeFiles/CMakeError.log".
make: *** [cmake_check_build_system] Error 1

hridaybavle, 2015-12-20:
sudo apt-get install ros-indigo-control-toolbox, if you are using ros-jade use jade instead of indigo

won13y, 2015-12-21:
It finally Works!

Thank you very much!

closed 2015-12-21 by won13y