mavlink/mavros #456: mavros catkin make problem : make -j8 -l8" failed

won13y, 2015-12-21:
I tried to catkin make mavros

but the error is as following:

" make -j8 -l8" failed"

Would any of you please let me know how to fix it?

odroid@odroid:~/catkin_ws$ catkin_make

Base path: /home/odroid/catkin_ws
Source space: /home/odroid/catkin_ws/src
Build space: /home/odroid/catkin_ws/build
Devel space: /home/odroid/catkin_ws/devel
Install space: /home/odroid/catkin_ws/install
####
#### Running command: "cmake /home/odroid/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/odroid/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/odroid/catkin_ws/install -G Unix Makefiles" in "/home/odroid/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/odroid/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/odroid/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/odroid/catkin_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/odroid/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.16
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 22 packages in topological order:
-- ~~  - geometry (metapackage)
-- ~~  - libmavconn
-- ~~  - rqt_moveit
-- ~~  - rqt_nav_view
-- ~~  - rqt_pose_view
-- ~~  - rqt_robot_dashboard
-- ~~  - rqt_robot_monitor
-- ~~  - rqt_robot_plugins (metapackage)
-- ~~  - rqt_robot_steering
-- ~~  - rqt_runtime_monitor
-- ~~  - rqt_tf_tree
-- ~~  - eigen_conversions
-- ~~  - kdl_conversions
-- ~~  - mavros_msgs
-- ~~  - realtime_tools
-- ~~  - control_toolbox
-- ~~  - mavros
-- ~~  - tf
-- ~~  - tf_conversions
-- ~~  - mavros_extras
-- ~~  - rqt_rviz
-- ~~  - test_mavros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'geometry'
-- ==> add_subdirectory(geometry/geometry)
-- +++ processing catkin package: 'libmavconn'
-- ==> add_subdirectory(mavros/libmavconn)
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
-- Selected MAVLink dialect: ardupilotmega
-- +++ processing catkin package: 'rqt_moveit'
-- ==> add_subdirectory(rqt_robot_plugins/rqt_moveit)
-- +++ processing catkin package: 'rqt_nav_view'
-- ==> add_subdirectory(rqt_robot_plugins/rqt_nav_view)
-- +++ processing catkin package: 'rqt_pose_view'
-- ==> add_subdirectory(rqt_robot_plugins/rqt_pose_view)
[... 261 lines cut ...]
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [mavros/libmavconn/CMakeFiles/mavconn.dir/src/udp.cpp.o] Error 4
make[1]: *** [mavros/libmavconn/CMakeFiles/mavconn.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/odroid/catkin_ws/src/geometry/tf/src/pytf.cpp:476:1: warning: converting to non-pointer type 'int' from NULL [-Wconversion-null]
 };
 ^
[ 72%] Building CXX object control_toolbox/CMakeFiles/control_toolbox.dir/src/dither.cpp.o
In file included from /home/odroid/catkin_ws/src/control_toolbox/src/dither.cpp:37:0:
/home/odroid/catkin_ws/src/control_toolbox/include/control_toolbox/dither.h: In member function 'bool control_toolbox::Dither::init(const double&, const double&)':
/home/odroid/catkin_ws/src/control_toolbox/include/control_toolbox/dither.h:96:22: warning: right shift count >= width of type [enabled by default]
     seed_[2] = (s >> 32) ^ 5598;
                      ^
[ 72%] Building CXX object control_toolbox/CMakeFiles/control_toolbox.dir/src/limited_proxy.cpp.o
[ 73%] Building CXX object control_toolbox/CMakeFiles/control_toolbox.dir/src/sinusoid.cpp.o
Linking CXX shared library /home/odroid/catkin_ws/devel/lib/libcontrol_toolbox.so
[ 73%] Built target control_toolbox
Linking CXX shared library /home/odroid/catkin_ws/devel/lib/libtf.so
[ 73%] Built target tf
[ 73%] Building CXX object geometry/tf/CMakeFiles/pytf_py.dir/src/cache.cpp.o
[ 74%] Building CXX object geometry/tf/CMakeFiles/pytf_py.dir/src/transform_listener.cpp.o
Linking CXX shared library /home/odroid/catkin_ws/devel/lib/python2.7/dist-packages/tf/_tf.so
[ 74%] Built target pytf_py
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

hridaybavle, 2015-12-21:
Run it with catkin_make -j2 -l2 if you are building it on odroid.

vooon, 2015-12-21:
Please follow instructions https://github.com/mavlink/mavros/tree/master/mavros#installation

won13y, 2015-12-22:
Thank Voon

I already follow the intstruction https://github.com/mavlink/mavros/tree/master/mavros#installation

as following:

But I get the error  as I described above: Invoking "make -j8 -l8" failed

Source installation

Use  wstool  utility for retriving sources and  catkin  tool for build.

sudo apt-get install python-catkin-tools

# 1. unneded if you already has workspace
mkdir -p ~/catkin_ws/src
cd ~/catkin_ws
catkin init
wstool init src

# 2. get source (upstream - released)
rosinstall_generator --upstream mavros | tee /tmp/mavros.rosinstall
# alternative: latest source
rosinstall_generator --upstream-development mavros | tee /tmp/mavros.rosinstall

# 3. latest released mavlink package
# you may run from this line to update ros-*-mavlink package
rosinstall_generator mavlink | tee -a /tmp/mavros.rosinstall

# 4. workspace & deps
wstool merge -t src /tmp/mavros.rosinstall
wstool update -t src
rosdep install --from-paths src --ignore-src --rosdistro jade -y

# 5. finally - build
catkin build

vooon, 2015-12-22:
Limit number of jobs, as @hridaybavle said.
But command is `catkin build -j2`

won13y, 2015-12-29:
@hridaybavle  

I did catkin_make -j2 -l2

but still error is as following:

odroid@odroid:~/catkin_ws$ catkin_make -j2 -l2

Base path: /home/odroid/catkin_ws
Source space: /home/odroid/catkin_ws/src
Build space: /home/odroid/catkin_ws/build
Devel space: /home/odroid/catkin_ws/devel
Install space: /home/odroid/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/odroid/catkin_ws/build"
####
####
#### Running command: "make -j2 -l2" in "/home/odroid/catkin_ws/build"
####
[  1%] Built target eigen_conversions
[  3%] Built target mavconn
[  3%] Built target kdl_conversions
[  3%] Built target geometry_msgs_generate_messages_py
[  3%] Built target std_msgs_generate_messages_py
[  3%] Built target _mavros_msgs_generate_messages_check_deps_FileRename
[  3%] Built target _mavros_msgs_generate_messages_check_deps_PositionTarget
[  3%] Built target _mavros_msgs_generate_messages_check_deps_FileList
[  3%] Built target _mavros_msgs_generate_messages_check_deps_WaypointClear
[  3%] Built target _mavros_msgs_generate_messages_check_deps_FileOpen
[  3%] Built target _mavros_msgs_generate_messages_check_deps_CamIMUStamp
[  3%] Built target _mavros_msgs_generate_messages_check_deps_State
[  3%] Built target _mavros_msgs_generate_messages_check_deps_HomePosition
[  3%] Built target _mavros_msgs_generate_messages_check_deps_VFR_HUD
[  3%] Built target _mavros_msgs_generate_messages_check_deps_CommandHome
[  3%] Built target _mavros_msgs_generate_messages_check_deps_FileRead
[  3%] Built target _mavros_msgs_generate_messages_check_deps_ParamGet
[  3%] Built target _mavros_msgs_generate_messages_check_deps_ActuatorControl
[  3%] Built target _mavros_msgs_generate_messages_check_deps_ParamPush
[  3%] Built target _mavros_msgs_generate_messages_check_deps_Vibration
[  3%] Built target _mavros_msgs_generate_messages_check_deps_GlobalPositionTarget
[  3%] Built target _mavros_msgs_generate_messages_check_deps_SetMode
[  3%] Built target _mavros_msgs_generate_messages_check_deps_BatteryStatus
[  3%] Built target _mavros_msgs_generate_messages_check_deps_ParamValue
[  3%] Built target _mavros_msgs_generate_messages_check_deps_Waypoint
[  3%] Built target _mavros_msgs_generate_messages_check_deps_ParamSet
[  3%] Built target _mavros_msgs_generate_messages_check_deps_RadioStatus
[  3%] Built target _mavros_msgs_generate_messages_check_deps_CommandTriggerControl
[  3%] Built target _mavros_msgs_generate_messages_check_deps_OverrideRCIn
[  3%] Built target _mavros_msgs_generate_messages_check_deps_CommandLong
[  3%] Built target _mavros_msgs_generate_messages_check_deps_WaypointPull
[  3%] Built target _mavros_msgs_generate_messages_check_deps_FileTruncate
[  3%] Built target _mavros_msgs_generate_messages_check_deps_FileRemoveDir
[  3%] Built target _mavros_msgs_generate_messages_check_deps_WaypointList
[  3%] Built target _mavros_msgs_generate_messages_check_deps_Altitude
[  3%] Built target _mavros_msgs_generate_messages_check_deps_OpticalFlowRad
[  3%] Built target _mavros_msgs_generate_messages_check_deps_StreamRate
[  3%] Built target _mavros_msgs_generate_messages_check_deps_Mavlink
[  3%] Built target _mavros_msgs_generate_messages_check_deps_FileRemove
[  3%] Built target _mavros_msgs_generate_messages_check_deps_WaypointPush
[  3%] Built target _mavros_msgs_generate_messages_check_deps_FileEntry
[  3%] Built target _mavros_msgs_generate_messages_check_deps_RCOut
[  3%] Built target _mavros_msgs_generate_messages_check_deps_ExtendedState
[  3%] Built target _mavros_msgs_generate_messages_check_deps_FileChecksum
[  3%] Built target _mavros_msgs_generate_messages_check_deps_CommandBool
[  3%] Built target _mavros_msgs_generate_messages_check_deps_AttitudeTarget
[  3%] Built target _mavros_msgs_generate_messages_check_deps_CommandCode
[  3%] Built target _mavros_msgs_generate_messages_check_deps_RCIn
[  3%] Built target _mavros_msgs_generate_messages_check_deps_CommandTOL
[  3%] Built target _mavros_msgs_generate_messages_check_deps_ManualControl
[  3%] Built target _mavros_msgs_generate_messages_check_deps_FileClose
[... 129 lines cut ...]
[ 96%] Built target transform_listener_unittest
[ 96%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/setpoint_attitude.cpp.o
Scanning dependencies of target tf_conversions
[ 97%] Building CXX object geometry/tf_conversions/CMakeFiles/tf_conversions.dir/src/tf_kdl.cpp.o
[ 97%] Building CXX object geometry/tf_conversions/CMakeFiles/tf_conversions.dir/src/tf_eigen.cpp.o
[ 98%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/setpoint_raw.cpp.o
Linking CXX shared library /home/odroid/catkin_ws/devel/lib/libtf_conversions.so
[ 98%] Built target tf_conversions
[ 98%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/vfr_hud.cpp.o
Scanning dependencies of target sitl_test_node
[ 98%] Building CXX object mavros/test_mavros/CMakeFiles/sitl_test_node.dir/sitl_test/sitl_test_node.cpp.o
[ 98%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/ftp.cpp.o
Linking CXX executable /home/odroid/catkin_ws/devel/lib/test_mavros/sitl_test_node
[ 98%] Built target sitl_test_node
[100%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/actuator_control.cpp.o
[100%] Building CXX object mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/manual_control.cpp.o
[... cut ...]
/home/odroid/catkin_ws/src/mavros/mavros/src/plugins/altitude.cpp:61:71: error: 'mavlink_altitude_t' has no member named 'time_usec'
         ros_msg->header = uas->synchronized_header(frame_id, altitude.time_usec);
                                                                       ^
make[2]: *** [mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/altitude.cpp.o] Error 1
make[1]: *** [mavros/mavros/CMakeFiles/mavros_plugins.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j2 -l2" failed

won13y, 2015-12-29:
@vooon 

Thanks voon,

I did catkin build -j2, but still error is following as:

odroid@odroid:~/catkin_ws$ catkin build -j2

[build] Runtime: 0.0 seconds 
Error: Build space `/home/odroid/catkin_ws/build` exists but appears to have previously been created by the `catkin_make` or `catkin_make_isolated` tool. Please choose a different directory to use with `catkin build` or clean the build space.
odroid@odroid:~/catkin_ws$ catkin clean --all
[clean] Removing buildspace: /home/odroid/catkin_ws/build
[clean] Removing develspace: /home/odroid/catkin_ws/devel
odroid@odroid:~/catkin_ws$ catkin build -j2
Creating build space directory, '/home/odroid/catkin_ws/build' 
-----------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/indigo
Workspace:                   /home/odroid/catkin_ws
Source Space:       [exists] /home/odroid/catkin_ws/src
Build Space:        [exists] /home/odroid/catkin_ws/build
Devel Space:       [missing] /home/odroid/catkin_ws/devel
Install Space:     [missing] /home/odroid/catkin_ws/install
DESTDIR:                     None
-----------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
-----------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
-----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------
Workspace configuration appears valid.
----------------------------------------------------------- 
Found '22' packages in 0.1 seconds. 
Starting ==> eigen_conversions                                                  
Starting ==> geometry                                                           
Starting ==> kdl_conversions                                                    
Starting ==> libmavconn                                                         
Starting ==> mavros_msgs                                                        
Starting ==> realtime_tools                                                     
Starting ==> rqt_moveit                                                         
Starting ==> rqt_nav_view                                                       
Finished <== geometry            [ 6.1 seconds ]                                
Starting ==> rqt_pose_view                                                      
Finished <== rqt_nav_view        [ 12.0 seconds ]                               
Starting ==> rqt_robot_dashboard                                                
Finished <== eigen_conversions   [ 14.7 seconds ]                               
Starting ==> rqt_robot_monitor                                                  
Finished <== rqt_robot_dashboard [ 8.0 seconds ]                                
Starting ==> rqt_robot_plugins                                                  
Finished <== kdl_conversions     [ 24.9 seconds ]                               
Starting ==> rqt_robot_steering                                                 
Finished <== rqt_robot_monitor   [ 11.7 seconds ]                               
Starting ==> rqt_runtime_monitor                                                
Finished <== rqt_robot_steering  [ 7.7 seconds ]                                
Starting ==> rqt_rviz                                                           
Finished <== rqt_rviz            [ 30.9 seconds ]                               
Starting ==> rqt_tf_tree                                                        
Finished <== rqt_runtime_monitor [ 56.7 seconds ]                               
Starting ==> tf                                                                 
Finished <== libmavconn          [ 1 minute and 23.3 seconds ]                  
Finished <== rqt_tf_tree         [ 25.8 seconds ]                               
Finished <== rqt_pose_view       [ 1 minute and 23.5 seconds ]                  
Finished <== rqt_moveit          [ 1 minute and 35.7 seconds ]                  
[... 54 lines cut ...]
[ 96%] Building CXX object CMakeFiles/mavros_plugins.dir/src/plugins/manual_control.cpp.o
[100%] Building CXX object CMakeFiles/mavros_plugins.dir/src/plugins/altitude.cpp.o
/home/odroid/catkin_ws/src/mavros/mavros/src/plugins/altitude.cpp: In member function 'void mavplugin::AltitudePlugin::handle_altitude(const mavlink_message_t*, uint8_t, uint8_t)':
/home/odroid/catkin_ws/src/mavros/mavros/src/plugins/altitude.cpp:61:71: error: 'mavlink_altitude_t' has no member named 'time_usec'
         ros_msg->header = uas->synchronized_header(frame_id, altitude.time_usec);
                                                                       ^
make[2]: *** [CMakeFiles/mavros_plugins.dir/src/plugins/altitude.cpp.o] Error 1
make[1]: *** [CMakeFiles/mavros_plugins.dir/all] Error 2
make: *** [all] Error 2
[mavros] <== '/home/odroid/catkin_ws/build/mavros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j' failed with return code '2'
                                                                                 
Failed   <== mavros              [ 8 minutes and 22.3 seconds ]                 
[build] There were '1' errors:                                                  

Failed to build package 'mavros' because the following command:

# Command to reproduce:
cd /home/odroid/catkin_ws/build/mavros && /home/odroid/catkin_ws/build/mavros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j; cd -

# Path to log:
cat /home/odroid/catkin_ws/build/build_logs/mavros.log

Exited with return code: 2 
                                                                                
[build] Runtime: 11 minutes and 3.4 seconds

vooon, 2016-01-02:
I do not see `mavlink` package in `catkin` report. Please follow instructions carefully.

won13y, 2016-01-05:
@vooon 

Thank you for advise

Still error is as following:

odroid@odroid:~/catkin_ws$ catkin build -j2
-----------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/indigo
Workspace:                   /home/odroid/catkin_ws
Source Space:       [exists] /home/odroid/catkin_ws/src
Build Space:        [exists] /home/odroid/catkin_ws/build
Devel Space:        [exists] /home/odroid/catkin_ws/devel
Install Space:     [missing] /home/odroid/catkin_ws/install
DESTDIR:                     None
-----------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
-----------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
-----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------
Workspace configuration appears valid.
----------------------------------------------------------- 
Found '6' packages in 0.0 seconds. 
Starting ==> mavlink                                                                                                                         
Starting ==> mavros_msgs                                                                                                                     
Finished <== mavros_msgs   [ 1 minute and 11.0 seconds ]                                                                                     
Finished <== mavlink       [ 3 minutes and 40.8 seconds ]                                                                                    
Starting ==> libmavconn                                                                                                                      
Finished <== libmavconn    [ 42.5 seconds ]                                                                                                  
Starting ==> mavros                                                                                                                          
Finished <== mavros        [ 6 minutes and 15.6 seconds ]                                                                                    
Starting ==> mavros_extras                                                                                                                   
Finished <== mavros_extras [ 2 minutes and 8.5 seconds ]                                                                                     
Starting ==> test_mavros                                                                                                                     
                                                                                                                                             
[test_mavros] ==> '/home/odroid/catkin_ws/build/test_mavros/build_env.sh /usr/bin/cmake /home/odroid/catkin_ws/src/mavros/test_mavros -DCATKIN_DEVEL_PREFIX=/home/odroid/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/odroid/catkin_ws/install' in '/home/odroid/catkin_ws/build/test_mavros'
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Using CATKIN_DEVEL_PREFIX: /home/odroid/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/odroid/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/odroid/catkin_ws/devel;/opt/ros/indigo
-- Found PythonInterp: /usr/bin/python (found version "2.7.6") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/odroid/catkin_ws/build/test_mavros/test_results
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
[... 16 lines cut ...]
  set "control_toolbox_DIR" to a directory containing one of the above files.
  If "control_toolbox" provides a separate development package or SDK, be
  sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:7 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/odroid/catkin_ws/build/test_mavros/CMakeFiles/CMakeOutput.log".
See also "/home/odroid/catkin_ws/build/test_mavros/CMakeFiles/CMakeError.log".
[test_mavros] <== '/home/odroid/catkin_ws/build/test_mavros/build_env.sh /usr/bin/cmake /home/odroid/catkin_ws/src/mavros/test_mavros -DCATKIN_DEVEL_PREFIX=/home/odroid/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/odroid/catkin_ws/install' failed with return code '1'
                                                                                                                                              
Failed   <== test_mavros   [ 7.5 seconds ]                                                                                                   
[... cut ...]
cd /home/odroid/catkin_ws/build/test_mavros && /home/odroid/catkin_ws/build/test_mavros/build_env.sh /usr/bin/cmake /home/odroid/catkin_ws/src/mavros/test_mavros -DCATKIN_DEVEL_PREFIX=/home/odroid/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/odroid/catkin_ws/install; cd -

# Path to log:
cat /home/odroid/catkin_ws/build/build_logs/test_mavros.log

Exited with return code: 1 
                                                                                                                                             
[build] Runtime: 12 minutes and 55.5 seconds

vooon, 2016-01-06:
`rosdep` had to install `control_toolbox` package. If not - install:
```
sudo apt-get install ros-$ROS_DISTRO-control-toolbox
```

closed 2016-01-12 by vooon

noise2, 2016-10-09:
@vooon still getting the error...! :|

Lhumd, 2018-11-01:
Hello everyone
➜  catkin_ws catkin_make
Base path: /home/fatemeh/Desktop/catkin_ws
Source space: /home/fatemeh/Desktop/catkin_ws/src
Build space: /home/fatemeh/Desktop/catkin_ws/build
Devel space: /home/fatemeh/Desktop/catkin_ws/devel
Install space: /home/fatemeh/Desktop/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/fatemeh/Desktop/catkin_ws/build"
####
####
#### Running command: "make -j2 -l2" in "/home/fatemeh/Desktop/catkin_ws/build"
####
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_RawImu
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_Compass
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_Supply
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_ControllerState
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_HeightCommand
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_VelocityXYCommand
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_RawMagnetic
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_MotorPWM
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_ThrustCommand
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_Altimeter
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_RuddersCommand
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_MotorCommand
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_YawrateCommand
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_RC
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_PositionXYCommand
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_ServoCommand
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_MotorStatus
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_AttitudeCommand
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_RawRC
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_VelocityZCommand
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target _hector_uav_msgs_generate_messages_check_deps_HeadingCommand
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target map_markers
[  1%] Built target hector_quadrotor_aerodynamics
[  2%] Built target p3at
[  2%] Built target sensor_msgs_generate_messages_lisp
[  2%] Built target rosgraph_msgs_generate_messages_eus
[  2%] Built target roscpp_generate_messages_py
[  2%] Built target roscpp_generate_messages_eus
[  2%] Built target geometry_msgs_generate_messages_py
[  2%] Built target roscpp_generate_messages_lisp
[  2%] Built target geometry_msgs_generate_messages_cpp
[  2%] Built target geometry_msgs_generate_messages_lisp
[  2%] Built target rosgraph_msgs_generate_messages_nodejs
[  2%] Built target geometry_msgs_generate_messages_nodejs
[  2%] Built target rosgraph_msgs_generate_messages_cpp
[  2%] Built target roscpp_generate_messages_nodejs
[  2%] Built target rosgraph_msgs_generate_messages_py
[  2%] Built target rosgraph_msgs_generate_messages_lisp
[  2%] Built target roscpp_generate_messages_cpp
[  2%] Built target sensor_msgs_generate_messages_py
[  2%] Built target geometry_msgs_generate_messages_eus
[  2%] Built target sensor_msgs_generate_messages_eus
[  2%] Built target sensor_msgs_generate_messages_nodejs
[  2%] Built target sensor_msgs_generate_messages_cpp
[  3%] Built target p2os_teleop
[  6%] Built target combine_grids
[  6%] Built target _aut_exploration_generate_messages_check_deps_Object
[  7%] Built target ltp
[  7%] Built target nav_msgs_generate_messages_cpp
[  7%] Built target _aut_exploration_generate_messages_check_deps_FarGoal
[  7%] Built target _aut_exploration_generate_messages_check_deps_Detections
[  7%] Built target nav_msgs_generate_messages_py
[... 102 lines cut ...]
                 from /opt/ros/kinetic/include/hector_gazebo_plugins/update_timer.h:33,
                 from /home/fatemeh/Desktop/catkin_ws/src/Invincible/src/hector_quadrotor/hector_quadrotor_gazebo_plugins/include/hector_quadrotor_gazebo_plugins/gazebo_quadrotor_simple_controller.h:42,
                 from /home/fatemeh/Desktop/catkin_ws/src/Invincible/src/hector_quadrotor/hector_quadrotor_gazebo_plugins/src/gazebo_quadrotor_simple_controller.cpp:29:
/usr/include/gazebo-7/gazebo/msgs/msgs.hh:24:37: fatal error: ignition/math/Inertial.hh: No such file or directory
compilation terminated.
Invincible/src/hector_quadrotor/hector_quadrotor_gazebo_plugins/CMakeFiles/hector_gazebo_quadrotor_simple_controller.dir/build.make:62: recipe for target 'Invincible/src/hector_quadrotor/hector_quadrotor_gazebo_plugins/CMakeFiles/hector_gazebo_quadrotor_simple_controller.dir/src/gazebo_quadrotor_simple_controller.cpp.o' failed
make[2]: *** [Invincible/src/hector_quadrotor/hector_quadrotor_gazebo_plugins/CMakeFiles/hector_gazebo_quadrotor_simple_controller.dir/src/gazebo_quadrotor_simple_controller.cpp.o] Error 1
CMakeFiles/Makefile2:4477: recipe for target 'Invincible/src/hector_quadrotor/hector_quadrotor_gazebo_plugins/CMakeFiles/hector_gazebo_quadrotor_simple_controller.dir/all' failed
make[1]: *** [Invincible/src/hector_quadrotor/hector_quadrotor_gazebo_plugins/CMakeFiles/hector_gazebo_quadrotor_simple_controller.dir/all] Error 2
[... cut ...]
make[2]: *** [Invincible/src/hector_quadrotor/hector_quadrotor_gazebo_plugins/CMakeFiles/hector_gazebo_quadrotor_propulsion.dir/src/gazebo_quadrotor_propulsion.cpp.o] Error 1
CMakeFiles/Makefile2:4548: recipe for target 'Invincible/src/hector_quadrotor/hector_quadrotor_gazebo_plugins/CMakeFiles/hector_gazebo_quadrotor_propulsion.dir/all' failed
make[1]: *** [Invincible/src/hector_quadrotor/hector_quadrotor_gazebo_plugins/CMakeFiles/hector_gazebo_quadrotor_propulsion.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j2 -l2" failed

Who can help me to fix the bug?

vooon, 2018-11-06:
@Lhumd what i can say definitely - you have problem with gazebo (likely with its [Ignition][1] dependency), not related to mavros.

[1]: https://ignitionrobotics.org/home