mavlink/mavros #457: mavros connection problem

harewei, 2015-12-22:
I've tried to connect from Raspberry Pi 2 running Xbuntu and using an USB to UART (Not FTDI) to pixhawk lateset firmware (updated through QGroundControl) through TELEM2.  The sys_companion parameter has been set to 57600 (didn't use 921600 since I have never gotten a connection this way).

Now, there are 3 possible outcomes when I use roslaunch mavros px4.launch (I've also edited the launch file to connect to USB0 and baud rate to 57600).
1.Never gets heartbeat
The program gets stuck at 
MAVROS started. MY ID [1, 240], TARGET ID [1, 1]

2.Still no heart beat, but got this IMU thing, then program gets stuck again
MAVROS started. MY ID [1, 240], TARGET ID [1, 1]
IMU: High resolution IMU detected!

3.Gets heartbeat, but has an error forcing an exit
symbol lookup error: /home/odroid/catkin_ws/devel/lib//libmavros_plugins.so: undefined symbol: _ZN3ros8DurationC1ERKNS_4RateE

Here is one of the diagnostics
level: 0
    name: mavros: System
    message: Normal
    hardware_id: /dev/ttyUSB0:57600
    values: 
      - 
        key: Sensor present
        value: 0x00000000
      - 
        key: Sensor enabled
        value: 0x00000000
      - 
        key: Sensor helth
        value: 0x00000000
      - 
        key: CPU Load (%)
        value: 0.0
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 2
    name: mavros: Battery
    message: No data
    hardware_id: /dev/ttyUSB0:57600
    values: 
      - 
        key: Voltage
        value: -1.00
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 0.0
  - 
    level: 2
    name: mavros: Time Sync
    message: No events recorded.
    hardware_id: /dev/ttyUSB0:57600
    values: 
      - 
        key: Timesyncs since startup
        value: 0
      - 
        key: Frequency (Hz)
        value: 0.000000
      - 
        key: Last dt (ms)
        value: 0.000000
      - 
        key: Mean dt (ms)
        value: 0.000000
      - 
        key: Last system time (s)
        value: 0.000000000
      - 
        key: Time offset (s)
        value: 0.000000000
---
header: 
  seq: 41
  stamp: 
    secs: 1450752094
    nsecs: 865901608
  frame_id: ''
status: 
  - 
    level: 1
    name: mavros: FCU connection
    message: not connected
    hardware_id: /dev/ttyUSB0:57600
    values: 
      - 
        key: Received packets:
        value: 0
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 0
      - 
        key: Tx sequence number:
        value: 46
      - 
        key: Rx total bytes:
        value: 35816
      - 
        key: Tx total bytes:
        value: 12915
      - 
        key: Rx speed:
        value: 722.000000
      - 
        key: Tx speed:
        value: 277.000000
  - 
    level: 2
    name: mavros: GPS
    message: No satellites
    hardware_id: /dev/ttyUSB0:57600
    values: 
      - 
        key: Satellites visible
        value: 0
      - 
        key: Fix type
        value: 0
      - 
        key: EPH (m)
        value: Unknown
      - 
        key: EPV (m)
        value: Unknown
  - 
    level: 2
    name: mavros: Heartbeat
    message: No events recorded.
    hardware_id: /dev/ttyUSB0:57600
    values: 
      - 
        key: Heartbeats since startup
        value: 0
      - 
        key: Frequency (Hz)
        value: 0.000000
      - 
        key: Vehicle type
        value: Generic
      - 
        key: Autopilot type
        value: Generic
      - 
        key: Mode
        value: ''
      - 
        key: System status
        value: Uninit
  - 
    level: 0
    name: mavros: System
    message: Normal
    hardware_id: /dev/ttyUSB0:57600
    values: 
      - 
        key: Sensor present
        value: 0x00000000
      - 
        key: Sensor enabled
        value: 0x00000000
      - 
        key: Sensor helth
        value: 0x00000000
      - 
        key: CPU Load (%)
        value: 0.0
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 2
    name: mavros: Battery
    message: No data
    hardware_id: /dev/ttyUSB0:57600
    values: 
      - 
        key: Voltage
        value: -1.00
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 0.0
  - 
    level: 2
    name: mavros: Time Sync
    message: No events recorded.
    hardware_id: /dev/ttyUSB0:57600
    values: 
      - 
        key: Timesyncs since startup
        value: 0
      - 
        key: Frequency (Hz)
        value: 0.000000
      - 
        key: Last dt (ms)
        value: 0.000000
      - 
        key: Mean dt (ms)
        value: 0.000000
      - 
        key: Last system time (s)
        value: 0.000000000
      - 
        key: Time offset (s)
        value: 0.000000000

vooon, 2015-12-22:
Check your hardware connection on TELEM2 (1) to USB bridge (2): Tx1-Rx2, Rx1-Tx2, GND1-GND2.

Then use terminal program (e.g. `miniterm.py` from `python-serial`) to verify that pix sending something (data is unreadable binary stream).
Then check connection URL and diagnostics.

But by your diagnostics i may say that seems pix sending in other protocol than mavlink.
Or baud rate mismatches.

Also you may use `strace` and `socat` for debugging.

rs1990, 2016-01-08:
Hello ,

I have been having the same issue as mentioned above 

1.Never gets heartbeat
The program gets stuck at 
MAVROS started. MY ID [1, 240], TARGET ID [1, 1].

i have checked my connections multiple times and its exactly as shown in the pictures.  Any solution ??

Thank you

Raghu.

rs1990, 2016-01-08:
Hello, It connected! . a reboot worked. weird!.

closed 2016-01-09 by vooon

KJNKJNKJN, 2017-03-13:
hello,i have a question.how to change the baud from 57600 to 115200.

alvikag, 2018-03-27:
Hi,
 I am facing a problem. I am using mavros 0.23.3 and ros kinetic setup on an odroid, I am connecting a pixracer via USB. Upon launching the mavros node, I am not able to get the hearbeat connection. I even tried setting the SYS_COMPANION parameter to 921600 as recommended and also did a reboot, but it doesnt seem to be working.

alvikag, 2018-03-27:
Hi, 
I am facing a problem. I am using mavros 0.23.3 and ros kinetic setup on an odroid, I am connecting a pixracer via USB. Upon launching the mavros node, I am not able to get the hearbeat connection. I even tried setting the SYS_COMPANION parameter to 921600 as recommended and also did a reboot, but it doesn't seem to be working.

freerunner-18, 2018-07-24:
it is definitely the baud rate, i had the exact same problem even after checking with the rqt_runtime_monitor , my port was correct, the only reason it was not working was due to a mismatch in baud rate, changed it to 115200 and it started working perfectly fine . Hope it helps someone .

fabrizioschiano, 2020-03-10:
We tried the baudrate change and that did not solve the issue. We have the same problem and we get actually stuck in 

```
[0m[ INFO] [1583860597.997290721]: serial0: device: /dev/ttyPixhawk @ 921600 bps[0m
[0m[ INFO] [1583860597.998859899]: GCS URL: udp://@192.168.1.128[0m
[0m[ INFO] [1583860597.999242777]: udp1: Bind address: 0.0.0.0:14555[0m
[0m[ INFO] [1583860597.999415445]: udp1: Remote address: 192.168.1.128:14550[0m
[0m[ INFO] [1583860598.061509040]: Plugin 3dr_radio loaded[0m
[0m[ INFO] [1583860598.078882005]: Plugin 3dr_radio initialized[0m
[0m[ INFO] [1583860598.079193215]: Plugin actuator_control loaded[0m
[0m[ INFO] [1583860598.115638741]: Plugin actuator_control initialized[0m
[0m[ INFO] [1583860598.125331689]: Plugin adsb loaded[0m
[0m[ INFO] [1583860598.168408598]: Plugin adsb initialized[0m
[0m[ INFO] [1583860598.168937810]: Plugin altitude loaded[0m
[0m[ INFO] [1583860598.180424606]: Plugin altitude initialized[0m
[0m[ INFO] [1583860598.180826567]: Plugin cam_imu_sync loaded[0m
[0m[ INFO] [1583860598.186991197]: Plugin cam_imu_sync initialized[0m
[0m[ INFO] [1583860598.187460867]: Plugin command loaded[0m
[0m[ INFO] [1583860598.238497920]: Plugin command initialized[0m
[0m[ INFO] [1583860598.238893131]: Plugin companion_process_status loaded[0m
[0m[ INFO] [1583860598.277732800]: Plugin companion_process_status initialized[0m
[0m[ INFO] [1583860598.278132053]: Plugin debug_value loaded[0m
[0m[ INFO] [1583860598.330105279]: Plugin debug_value initialized[0m
[0m[ INFO] [1583860598.330198447]: Plugin distance_sensor blacklisted[0m
[0m[ INFO] [1583860598.330624950]: Plugin fake_gps loaded[0m
[0m[ INFO] [1583860598.550009109]: Plugin fake_gps initialized[0m
[0m[ INFO] [1583860598.550495612]: Plugin ftp loaded[0m
[0m[ INFO] [1583860598.618376667]: Plugin ftp initialized[0m
[0m[ INFO] [1583860598.618849004]: Plugin global_position loaded[0m
[0m[ INFO] [1583860598.827608583]: Plugin global_position initialized[0m
[0m[ INFO] [1583860598.827995752]: Plugin gps_rtk loaded[0m
[0m[ INFO] [1583860598.864599654]: Plugin gps_rtk initialized[0m
[0m[ INFO] [1583860598.865153992]: Plugin hil loaded[0m
[0m[ INFO] [1583860599.045495521]: Plugin hil initialized[0m
[0m[ INFO] [1583860599.046006775]: Plugin home_position loaded[0m
[0m[ INFO] [1583860599.097160870]: Plugin home_position initialized[0m
[0m[ INFO] [1583860599.097647290]: Plugin imu loaded[0m
[0m[ INFO] [1583860599.196726496]: Plugin imu initialized[0m
[0m[ INFO] [1583860599.197135666]: Plugin landing_target loaded[0m
[0m[ INFO] [1583860599.420217808]: Plugin landing_target initialized[0m
[0m[ INFO] [1583860599.420773687]: Plugin local_position loaded[0m
[0m[ INFO] [1583860599.509902610]: Plugin local_position initialized[0m
[0m[ INFO] [1583860599.510330072]: Plugin log_transfer loaded[0m
[0m[ INFO] [1583860599.536502978]: Plugin log_transfer initialized[0m
[0m[ INFO] [1583860599.536985773]: Plugin manual_control loaded[0m
[0m[ INFO] [1583860599.576652864]: Plugin manual_control initialized[0m
[0m[ INFO] [1583860599.577043992]: Plugin mocap_pose_estimate loaded[0m
[0m[ INFO] [1583860599.630330478]: Plugin mocap_pose_estimate initialized[0m
[0m[ INFO] [1583860599.630707230]: Plugin mount_control loaded[0m
[0m[ INFO] [1583860599.664771904]: Plugin mount_control initialized[0m
[0m[ INFO] [1583860599.665147199]: Plugin obstacle_distance loaded[0m
[0m[ INFO] [1583860599.699540875]: Plugin obstacle_distance initialized[0m
[0m[ INFO] [1583860599.699928128]: Plugin odom loaded[0m
[0m[ INFO] [1583860599.805859136]: Plugin odom initialized[0m
[0m[ INFO] [1583860599.806359098]: Plugin param loaded[0m
[0m[ INFO] [1583860599.835116357]: Plugin param initialized[0m
[0m[ INFO] [1583860599.835507610]: Plugin px4flow loaded[0m
[0m[ INFO] [1583860599.931130331]: Plugin px4flow initialized[0m
[0m[ INFO] [1583860599.931212957]: Plugin rangefinder blacklisted[0m
[0m[ INFO] [1583860599.931721586]: Plugin rc_io loaded[0m
[0m[ INFO] [1583860599.977775434]: Plugin rc_io initialized[0m
[0m[ INFO] [1583860599.977864101]: Plugin safety_area blacklisted[0m
[0m[ INFO] [1583860599.978419022]: Plugin setpoint_accel loaded[0m
[0m[ INFO] [1583860600.023157109]: Plugin setpoint_accel initialized[0m
[0m[ INFO] [1583860600.023648321]: Plugin setpoint_attitude loaded[0m
[0m[ INFO] [1583860600.156542989]: Plugin setpoint_attitude initialized[0m
[0m[ INFO] [1583860600.157079702]: Plugin setpoint_position loaded[0m
[0m[ INFO] [1583860600.352911469]: Plugin setpoint_position initialized[0m
[0m[ INFO] [1583860600.353407973]: Plugin setpoint_raw loaded[0m
[0m[ INFO] [1583860600.470565856]: Plugin setpoint_raw initialized[0m
[... 6 lines cut ...]
[0m[ INFO] [1583860600.776632621]: Plugin sys_time initialized[0m
[0m[ INFO] [1583860600.777064999]: Plugin trajectory loaded[0m
[0m[ INFO] [1583860600.853008363]: Plugin trajectory initialized[0m
[0m[ INFO] [1583860600.853514950]: Plugin vfr_hud loaded[0m
[0m[ INFO] [1583860600.860007166]: Plugin vfr_hud initialized[0m
[0m[ INFO] [1583860600.860105208]: Plugin vibration blacklisted[0m
[0m[ INFO] [1583860600.860555461]: Plugin vision_pose_estimate loaded[0m
[0m[ INFO] [1583860600.970476457]: Plugin vision_pose_estimate initialized[0m
[0m[ INFO] [1583860600.970901293]: Plugin vision_speed_estimate loaded[0m
[0m[ INFO] [1583860601.032402339]: Plugin vision_speed_estimate initialized[0m
[... cut ...]
[0m[ INFO] [1583860601.081581918]: Built-in MAVLink package version: 2019.9.9[0m
[0m[ INFO] [1583860601.081700627]: Known MAVLink dialects: common ardupilotmega ASLUAV autoquad icarous matrixpilot paparazzi slugs standard uAvionix ualberta[0m
[0m[ INFO] [1583860601.081810795]: MAVROS started. MY ID 1.240, TARGET ID 1.0[0m
[0m[ INFO] [1583860601.083868518]: IMU: High resolution IMU detected![0m
[0m[ INFO] [1583860601.088877848]: IMU: Attitude quaternion IMU detected![0m
[0m[ INFO] [1583860601.127888349]: RC_CHANNELS message detected!

```