mavlink/mavros #459: catkin_make problem : invoking "make -j8 -l8" failed

won13y, 2015-12-29:
odroid@odroid:~/catkin_ws$ catkin_make

Base path: /home/odroid/catkin_ws
Source space: /home/odroid/catkin_ws/src
Build space: /home/odroid/catkin_ws/build
Devel space: /home/odroid/catkin_ws/devel
Install space: /home/odroid/catkin_ws/install
####
#### Running command: "cmake /home/odroid/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/odroid/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/odroid/catkin_ws/install -G Unix Makefiles" in "/home/odroid/catkin_ws/build"
####
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Using CATKIN_DEVEL_PREFIX: /home/odroid/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/odroid/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/odroid/catkin_ws/devel;/opt/ros/indigo
-- Found PythonInterp: /usr/bin/python (found version "2.7.6") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/odroid/catkin_ws/build/test_results
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.16
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 22 packages in topological order:
-- ~~  - geometry (metapackage)
-- ~~  - libmavconn
-- ~~  - rqt_moveit
-- ~~  - rqt_nav_view
-- ~~  - rqt_pose_view
-- ~~  - rqt_robot_dashboard
-- ~~  - rqt_robot_monitor
-- ~~  - rqt_robot_plugins (metapackage)
-- ~~  - rqt_robot_steering
-- ~~  - rqt_runtime_monitor
-- ~~  - rqt_tf_tree
-- ~~  - eigen_conversions
-- ~~  - kdl_conversions
-- ~~  - mavros_msgs
-- ~~  - realtime_tools
-- ~~  - control_toolbox
-- ~~  - mavros
-- ~~  - tf
-- ~~  - tf_conversions
-- ~~  - mavros_extras
-- ~~  - rqt_rviz
-- ~~  - test_mavros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'geometry'
-- ==> add_subdirectory(geometry/geometry)
[... 522 lines cut ...]
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [mavros/mavros/CMakeFiles/mavros.dir/src/lib/uas_timesync.cpp.o] Error 4
g++-4.8.real: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [mavros/mavros/CMakeFiles/mavros.dir/src/lib/uas_stringify.cpp.o] Error 4
g++-4.8.real: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [mavros/mavros/CMakeFiles/mavros.dir/src/lib/uas_data.cpp.o] Error 4
make[1]: *** [mavros/mavros/CMakeFiles/mavros.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Linking CXX executable /home/odroid/catkin_ws/devel/lib/tf/static_transform_publisher
[ 79%] Built target static_transform_publisher
Linking CXX executable /home/odroid/catkin_ws/devel/lib/tf/testBroadcaster
[ 79%] Built target testBroadcaster
The bug is not reproducible, so it is likely a hardware or OS problem.
make[2]: *** [geometry/tf/CMakeFiles/pytf_py.dir/src/transform_listener.cpp.o] Error 1
make[1]: *** [geometry/tf/CMakeFiles/pytf_py.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 9 jobserver tokens available; should be 8!
Invoking "make -j8 -l8" failed

won13y, 2015-12-29:
And I also did catkin build -j2

as @vooon described
 
but the error is as following:
 

odroid@odroid:~/catkin_ws$ catkin build -j2

[build] Runtime: 0.0 seconds 
Error: Build space `/home/odroid/catkin_ws/build` exists but appears to have previously been created by the `catkin_make` or `catkin_make_isolated` tool. Please choose a different directory to use with `catkin build` or clean the build space.
odroid@odroid:~/catkin_ws$ catkin clean --all
[clean] Removing buildspace: /home/odroid/catkin_ws/build
[clean] Removing develspace: /home/odroid/catkin_ws/devel
odroid@odroid:~/catkin_ws$ catkin build -j2
Creating build space directory, '/home/odroid/catkin_ws/build' 
-----------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/indigo
Workspace:                   /home/odroid/catkin_ws
Source Space:       [exists] /home/odroid/catkin_ws/src
Build Space:        [exists] /home/odroid/catkin_ws/build
Devel Space:       [missing] /home/odroid/catkin_ws/devel
Install Space:     [missing] /home/odroid/catkin_ws/install
DESTDIR:                     None
-----------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
-----------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
-----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------
Workspace configuration appears valid.
----------------------------------------------------------- 
Found '22' packages in 0.1 seconds. 
Starting ==> eigen_conversions                                                  
Starting ==> geometry                                                           
Starting ==> kdl_conversions                                                    
Starting ==> libmavconn                                                         
Starting ==> mavros_msgs                                                        
Starting ==> realtime_tools                                                     
Starting ==> rqt_moveit                                                         
Starting ==> rqt_nav_view                                                       
Finished <== geometry            [ 6.1 seconds ]                                
Starting ==> rqt_pose_view                                                      
Finished <== rqt_nav_view        [ 12.0 seconds ]                               
Starting ==> rqt_robot_dashboard                                                
Finished <== eigen_conversions   [ 14.7 seconds ]                               
Starting ==> rqt_robot_monitor                                                  
Finished <== rqt_robot_dashboard [ 8.0 seconds ]                                
Starting ==> rqt_robot_plugins                                                  
Finished <== kdl_conversions     [ 24.9 seconds ]                               
Starting ==> rqt_robot_steering                                                 
Finished <== rqt_robot_monitor   [ 11.7 seconds ]                               
Starting ==> rqt_runtime_monitor                                                
Finished <== rqt_robot_steering  [ 7.7 seconds ]                                
Starting ==> rqt_rviz                                                           
Finished <== rqt_rviz            [ 30.9 seconds ]                               
Starting ==> rqt_tf_tree                                                        
Finished <== rqt_runtime_monitor [ 56.7 seconds ]                               
Starting ==> tf                                                                 
Finished <== libmavconn          [ 1 minute and 23.3 seconds ]                  
Finished <== rqt_tf_tree         [ 25.8 seconds ]                               
Finished <== rqt_pose_view       [ 1 minute and 23.5 seconds ]                  
[... 55 lines cut ...]
[ 96%] Building CXX object CMakeFiles/mavros_plugins.dir/src/plugins/manual_control.cpp.o
[100%] Building CXX object CMakeFiles/mavros_plugins.dir/src/plugins/altitude.cpp.o
/home/odroid/catkin_ws/src/mavros/mavros/src/plugins/altitude.cpp: In member function 'void mavplugin::AltitudePlugin::handle_altitude(const mavlink_message_t*, uint8_t, uint8_t)':
/home/odroid/catkin_ws/src/mavros/mavros/src/plugins/altitude.cpp:61:71: error: 'mavlink_altitude_t' has no member named 'time_usec'
         ros_msg->header = uas->synchronized_header(frame_id, altitude.time_usec);
                                                                       ^
make[2]: *** [CMakeFiles/mavros_plugins.dir/src/plugins/altitude.cpp.o] Error 1
make[1]: *** [CMakeFiles/mavros_plugins.dir/all] Error 2
make: *** [all] Error 2
[mavros] <== '/home/odroid/catkin_ws/build/mavros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j' failed with return code '2'
                                                                                 
Failed   <== mavros              [ 8 minutes and 22.3 seconds ]                 
[build] There were '1' errors:                                                  

Failed to build package 'mavros' because the following command:

# Command to reproduce:
cd /home/odroid/catkin_ws/build/mavros && /home/odroid/catkin_ws/build/mavros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j; cd -

# Path to log:
cat /home/odroid/catkin_ws/build/build_logs/mavros.log

Exited with return code: 2 
                                                                                
[build] Runtime: 11 minutes and 3.4 seconds

vooon, 2016-01-02:
I do not see `mavlink` package in `catkin` report. Do all steps from manual!

vooon, 2016-01-02:
Also you duplicate #456. Why open new issue? Reopen existent.
And please use triple \` ("```") to mark code blocks!

closed 2016-01-02 by vooon