mavlink/mavros #461: Mavros topic not publishing on ArduCopter

PenguinCheng, 2015-12-30:
We are using the mavros library to connect with pixhawk(hardware). Our firmware is APM:Copter V3.4-dev and I launch the mavros code using:
----------------------------------------------------------------
roslaunch mavros apm2.launch
-----------------------------------------------------------------
It all seems good. And I can see the topic list as:
-----------------------------------------------------------------
/diagnostics
/mavlink/from
/mavlink/to
/mavros/battery
/mavros/fix
/mavros/gps_vel
/mavros/imu/atm_pressure
/mavros/imu/data
/mavros/imu/data_raw
/mavros/imu/mag
/mavros/imu/temperature
/mavros/mission/waypoints
/mavros/radio_status
/mavros/rc/in
/mavros/rc/out
/mavros/rc/override
/mavros/state
/mavros/time_reference
/rosout
/rosout_agg
...
--------------------------------------------------------------------------
However, I can just get topics like /mavros/state, /diagnostics, and I can't get other topics when I run the command like:
-------------------------------------------------------------------------------------------------------------
rostopic echo /mavros/imu/data rostopic echo /mavros/global_position/raw/fix
...
-------------------------------------------------------------------------------------------------------------
I Armed the motors and still can't get such topics. It seems that mavros publishs no data on those topics.
All these steps are correct in SITL with no issues. And I'm trying to do this on real Copter, but encountered this issues. Can anybody help?

killerlayup, 2015-12-30:
I have the same issue!!
can anyone help

LorenzMeier, 2015-12-30:
Use PX4 if you want full ROS support: http://dev.px4.io. The topics will publish once you arm.

PenguinCheng, 2015-12-31:
Yes, I can get data from some other topics using PX4.

vooon, 2016-01-02:
@PenguinCheng check SERIAL_PROTO and similar APM parameters. Then try to change stream group rates (`mavsys rate --help`).

Because in most cases mavros only translate mavlink to ros messages (and vice versa). Publishing rate depends on autopilot.

PenguinCheng, 2016-01-04:
@vooon I had a holiday last two days and sorry for late message checking. Thank you for you good advice! I can  get some data from topics like ‘/mavros/imu/data’, '/mavros/global_position/raw/fix'... after using command 'rosrun mavrosmavsys rate --all 100 '~~ But it warn me that "TM: Wrong FCU time", I don't know whether it matters?
And I can't get data from '/mavros/local_position/pose' even if I armed the motor, have you met this issue before?

vooon, 2016-01-04:
That topic totally depends in `LOCAL_POSITION_NED` message. Check APM sources (`GCS_Mavlink.cpp`).

alvikag, 2016-01-05:
I am facing a similar issue. I am using arducopter firmware version 3.3.2 on an iris+. I am launching mavros using : rosrun mavros mavros_node _fcu_url:=/dev/ttyACM0:115200. I am running this command on odroid which is connected to quad through a USB.
When I do a rostopic list I see all the mavros topics, but once I arm and takeoff and do a rosbag record
I dont see any topics like /mavros/global_position or copass heading. 

Everything was working fine upto AC 3.3 rc10.

Please Help!

PenguinCheng, 2016-01-05:
@liamstask ,try 'rosrun mavrosmavsys rate --all 10 '

closed 2016-01-28 by vooon

PenguinCheng, 2016-02-24:
@vooon , I want to use the altitude of the copter above the ground using mavros, but I can't get the message of topic /mavros/local_position/pose. I have checked the APM sources (GCS_Mavlink.cpp) and didn't find out the ansower. Is there any thing like the param _vehicle.location.alt_ (`api = local_connect()  vehicle = api.get_vehicles()[0]`) in DroneAPI.

vooon, 2016-02-24:
I'm do not know what DroneAPI uses for that parameter. Check what messages available by QGC.

PenguinCheng, 2016-02-25:
@vooon ,hi, I checked the messages available by QGC, and I list all of them as follows:
--  HEARTBEAT
- - SYS_STATUS
- - SYSTEM_TIME
- - GPS_RAW_INT
- - RAW_IMU
- - SCALED_PRESSURE
- - ATTITUDE
- - GLOBAL_POSITION_INT
- - RC_CHANNELS_RAW
- - SERVO_OUTPUT_RAW
- - MISSION_CURRENT
- - NAV)CONTROLLER_OUTPUT
- - VFR_HUD
- - SCALED_IMU2
- - POWER_STATUS
- - VIBRATION
- - RAW_IMU
- - SCALED_PRESSURE
- - ATTITUDE
- - GLOBAL_POSITION_INT
- - RC_CHANNELS_RAW
- - SERVO_OUTPUT_RAW
- - MISSION_CURRENT
- - NAV)CONTROLLER_OUTPUT
- - VFR_HUD
- - SCALED_IMU2
- - POWER_STATUS
- - VIBRATION
  I have checked the latest ardupilot source code(https://github.com/diydrones/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp), and the code has sent out LOCAL_POSITION_NED, but I don't get in QGC. sad!
  And I use the firmware ArduCopter-v2.px4(2016-02-25), I am not sure it is same with the latest code in github. Call for help!

vooon, 2016-02-25:
I'm do not see `Vehicle.location.alt` at [DroneKit documentation][1], can you please check?
Also try to request all message group streams by `rosrun mavros mavsys rate --all 10`.

[1]: http://python.dronekit.io/automodule.html?highlight=vehicle.location#dronekit.Vehicle.location

PenguinCheng, 2016-02-26:
@vooon , the `vehicle.location.alt` has been changed to `vehicle.location.global_releative_frame.alt`. 
After request all message group streams by 'rosrun mavros mavsys rate --all 10', the messages in QGC has no change.  So that means the firmare don't send out LOCAL_POSITION_NED datas?

vooon, 2016-02-26:
DroneKit docs says that `vehicle.location.local_frame` available after arming.
But global_relative_frame not related to LOCAL_POSITION_NED. I'm think that it extracted from GLOBAL_POSITION_INT. You can find that values in `global_position` plugin topics.

amirrezasadeqi, 2024-07-03:
I did have the same problem. running `rosrun mavros mavsys rate --all 10` solved my issue. thanks!