mavlink/mavros #462: Help!  mavros can't get heartbeat from jMAVsim ( ROS/Mavros <X> jMAVSim )

whyscience, 2015-12-31:
I have follow the instruction here https://pixhawk.org/dev/ros/hil 
as follows:
setpoint_demo.py
             /
QGC <-> ROS/Mavros <-> jMAVSim <-> Pixhawk

We can get it work : jMAVSim <-> Pixhawk  ,we can control the quadcopter via Tx controller .

we also change this 
	<arg name="fcu_url" default="udp://:14554@192.168.1.5:14555" />
	<arg name="gcs_url" default="udp://:14556@192.168.1.5:14550" />

But it seemed that ROS/Mavros <-> jMAVSim does not work .

It's like this 

[image]
 

Can anybody help?

whyscience, 2015-12-31:
I'm using QGC 2.8.0 stable( PX4 firmware 1.1.0) ,after I  add IP and add localhost:14556  (mine is 127.0.0.1), the QGC itself will change it back to 14555 .
Then the QGC will connect to jMAVsim, but not mavros.

Why is it like this ?

vooon, 2016-01-02:
Check please: http://dev.px4.io/offboard-control.html

Also there already similar issue in tracker: #453.

closed 2016-02-05 by vooon