mavlink/mavros #463: Mavros rc/override causes mode problems

harewei, 2016-01-07:
Hi
I've used mavros to send commands to pixhawk, such as changing mode and arming, which works fine.
I've also tried to send rc/override in an attempt to control vehicle movement.  Now, this works since I've subscribed to rc/in to make sure the rc values does change.

The problem is this:
1.When unarmed, if I set all 8 rc channels to 1300~1400, the mode jumped from alt_hold to stabilize
2.When unarmed, if I set all 8 rc channels to 1200 or below, the mode jumped from alt_hold/stabilize to loiter
3.When unarmed, if I set all 8 rc channels to 1500 or above, the mode jumped from stabilize/loiter to alt_hold
4.When armed and flying in alt_hold, if I attempt to change any RC value, it goes to stabilize and immediately changes to land

Is this due to some safety feature which I need to turn off?  Because it seems that my commands are issuing correctly.
here is one of my command for rc value changes:
rostopic pub -1 /mavros/rc/override mavros_msgs/OverrideRCIn '[1500, 1500, 1500, 1500, 1500, 1500, 1500, 1500]'

Help appreciated

khancyr, 2016-01-07:
Hi,

I don't know it is the same on PX4 firmware, but with APM firmware, rc channel 6 change flight mode : 

http://copter.ardupilot.com/wiki/common-rc-transmitter-flight-mode-configuration/

So you need to config your drone parameter first then use rc override

vooon, 2016-01-07:
And what youre expect if you override RC mode switches?

harewei, 2016-01-11:
RIght, as vooon said, I totally forgot channel 5~8 is used for mode switching.
As a side question, I saw on the mavros command page that giving a 0 is for channel release (giving control back to the remote controller) while 65535 means maintain whatever value is was previously on, is this correct?

vooon, 2016-01-11:
http://mavlink.org/messages/common#RC_CHANNELS_OVERRIDE

closed 2016-01-28 by vooon