mavlink/mavros #464: mavros px4.launch can not get CON: Got HEARTBEAT, connected.

won13y, 2016-01-07:
I use odroid XU4 connect to TELEM2, after I run roslaunch mavros px4.launch. It should get message:

CON: Got HEARTBEAT, connected.
however, I can not get it and can not switch to offboard switch . anyone can tell me the reason. thank in advance. following is the message in the terminal after running roslaunch mavros px4.launch:

But it used to work properly even yesterday!!

wkyoun@wkyoun-XPS-13-9343:~$ roslaunch mavros px4.launch fcu_url:=/dev/ttyUSB0:57600

... logging to /home/wkyoun/.ros/log/3a0577a4-b517-11e5-9f45-c48e8ff3cd6f/roslaunch-wkyoun-XPS-13-9343-30064.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.43.138:48825/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_url: /dev/ttyUSB0:57600
 * /mavros/gcs_url:
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: fcu
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: True
 * /mavros/mocap/use_tf: False
[... 239 lines cut ...]
 * /mavros/vision_pose/tf/rate_limit: 10.0
 * /mavros/vision_speed/listen_twist: False
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /
    mavros (mavros/mavros_node)

ROS_MASTER_URI=http://192.168.43.138:11311

core service [/rosout] found
process[mavros-1]: started with pid [31105]
[ INFO] [1452155674.346863504]: FCU URL: /dev/ttyACM0:57600
[ INFO] [1452155674.347236005]: serial0: device: /dev/ttyACM0 @ 57600 bps
[FATAL] [1452155674.347525799]: FCU: DeviceError:serial:open: No such file or directory
================================================================================REQUIRED process [mavros-1] has died!
process has finished cleanly
log file: /home/wkyoun/.ros/log/3a0577a4-b517-11e5-9f45-c48e8ff3cd6f/mavros-1*.log
Initiating shutdown!
================================================================================
[mavros-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

vooon, 2016-01-07:
Please learn github markdown firmatting. It is really hard to support when report is not properly formatted.

vooon, 2016-01-07:
Did you configured `SYS_COMPANION`?

won13y, 2016-01-07:
@vooon I will learn github markdown firmatting.

Int the following web:

It says:

I will configure SYS_COMPANION to 921600 in Qgroundcontrol,
and let you know whether it is working or not tommorow
**https://pixhawk.org/dev/companion_link**

TELEM2 is supported by default and can be activated by setting the SYS_COMPANION parameter to 921600. In software this port is located at /dev/ttyS2

won13y, 2016-01-08:
it was due to SYS_COMPANION
Thank you @vooon 
Solved!

closed 2016-01-08 by vooon

ChunJyeBehBeh, 2018-01-24:
may I know more details about the solution?
change the SYS_COMPANION to 921600 in Qgroundcontrol? Why it cannot work for me?