mavlink/mavros #466: OFFBOARD : Controlling the PixHawk with 3D informations

AlexisTM, 2016-01-12:
Dear,

In examples we always need a 6D information to send to the PixHawk.

Since I do not have the roll/pitch/yaw, but only the position (X,Y,Z), how should I do to take the control of the vehicle and let him use his own IMU for pitch yaw roll ?

Alexis

vooon, 2016-01-12:
http://wiki.ros.org/mavros#mavros.2BAC8-Plugins.setpoint_position

Here you need only Point (3D position). Orientation may be identity (1 0 0 0) => Yaw = 0.

AlexisTM, 2016-01-14:
Thanks, 

The setpoints are correctly received and understoob by the PixHawk when we send them to `mavros/setpoint_position/local`, but we would like to send positions, because we are indoor and there are not GPS.

We tried to send positions to `mavros/local_position/local` but mavros do not subscribe to this topic. Can't we send positions ? Do we have to simulate the GPS to use a 3D positionning system ?

vooon, 2016-01-14:
You may try `vision_position`.

kaustubha-jagtap, 2016-01-20:
@AlexisTM : You can send the positions to <code> mavros/setpoint_position/local </code>
If I understand correctly, all you want do is to send x,y and z position coordinates right? 
You can use <code> mavros/vision_pose/pose</code> or <code> mavros/mocap/pose </code>.
The pixhawk will move the to setpoint published on <code> mavros/setpoint_position/local</code> with reference to the FCU position published on <code> mavros/local_postion/pose </code> 
You do not need to simulate GPS for 3D positioning.
I hope I answered your question.

AlexisTM, 2016-01-20:
And what about the quaternion when publishing to `mavros/mocap/pose` ?  

Sending Q(0 0 0 1) won't make the PixHawk thinks he is wrong and correct his attitude ?

kaustubha-jagtap, 2016-01-20:
it depends on what mode you have selected on <code> ATT_EXT_HDG_M </code> . If the mode is 0 then pixhawk will try to align itself with magnetic north(be ready for sudden yaw change). If you have set mode to be 2 then it will be the same heading as given by Mocap. I suggest, if you are ONLY trying to  send the 3D ordinates then copy Q from topic <code> /mavros/local_position/pose </code> to send the PoseStamped(3D from you + Q from FCU) to <code> /mavros/setpoint_position/local </code>

closed 2016-02-05 by vooon