mavlink/mavros #468: Help with use of X8 with vicon system and mavros.

rs1990, 2016-01-14:
Hello,

I'm apologize if this isn't the right place for this question but I'd like to know if someone could help me out with how to get my x8 working with a VICON system. 
I have NY x8 with the pixhawk and an using an odroid u3 as the companion computer and have them working. Iv got mavros working and everything seems to be set up correct. I haven't flown it as yet but besides that its responding correctly.

Id like to know if there is any documentation for use with the vicon or if someone can help me out with what to do next. I can't fly my system outdoors yet.

I've seen a lot of discussions and groups but have no idea how to go about it. Eventually I'd like to be able to control my quad with a joystick and doing some position and velocity control.

I'd appreciate any help. Again apologies if its not the right place.

Thank you,

Raghavendra.

mhkabir, 2016-01-14:
http://dev.px4.io/external-position.html

rs1990, 2016-01-22:
so i set CBRK_NO_VISION to 0 and ATT_EXT_HDG_M to 1. is that correct ?

mhkabir, 2016-01-22:
That's what the documentation suggests. Maybe you could just try that?

closed 2016-01-28 by vooon