mavlink/mavros #469: Normalized throttle value for SITL on gazebo using mavros

gokoreas, 2016-01-15:
I am using a custom built quadrotor to fly offboard using MAVROS

I did the throttle mapping by just measuring the thrust compared to the normalized throttle value (/mavros/setpoint_attitude/att_throttle)

However, trying to use IRIS in the gazebo simulation with mavros, I have no idea what the throttle values should be compared to the thrust it generates

A guideline is needed for the thrust generated compared to the normalized thrust values at least for the standardized quadrotors such as IRIS or Solo

vooon, 2016-01-16:
http://mavlink.org/messages/common#SET_ATTITUDE_TARGET

gokoreas, 2016-01-16:
@voon 

What I mean by mapping is the thrust in N(force) to the value of throttle!

vooon, 2016-01-16:
I don't thing that FCU operate in newtons (because it depends on many factors like battery charge).
So i do not have quick answer.

gokoreas, 2016-01-16:
I understand. But we need the scales of normalized throttle value in order to fly in rate or acceleration control

Currently we are just measuring the thrust by experiments to match the normalized throttle values to N 

However for standardized quadrotors such as IRIS  which can run px4, if we have a guideline of on which battery voltage and thrust value creates which thrust(by real dimensions such as N), 

It can reduce the need for time and equipment to find the throttle value!

closed 2016-02-05 by vooon

IeiuniumLux, 2017-05-31:
@vooon , why did you close this?  I believe the main problem is that the on-board pid controller uses normalized thrust values @Jaeyoung-Lim is trying to use absolute thrust values instead.

Was guideline or map created to resolve this?