mavlink/mavros #472: Wrong mavros topic in ROS wiki

gokoreas, 2016-01-18:
In the documentation of mavros in ROS wiki, ( http://wiki.ros.org/mavros )

local_position/pose
local_position/velocity

topics should be changed into 

local_position/local
where the PoseStamped message includes orientation and position

For reference for the mavros package, rqt_graph shows the topics which are currently implemented in mavros
[image]

gokoreas, 2016-01-18:
I wonder if there is an alternative for the 

local_position/velocity

vooon, 2016-01-18:
https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/local_position.cpp#L50

closed 2016-01-18 by vooon