mavlink/mavros #474: how to find the topic providede by the package?

walkeryu123, 2016-01-19:
i want to learn the mavros,but i do not use ros before,now i have a problem,when I use the mavros package,how can i know the topic the package provided, so that i can published and subscribe messages?

andre-nguyen, 2016-01-19:
Easy peasy, read the documentation here: http://wiki.ros.org/mavros
All the topics are listed.

walkeryu123, 2016-01-19:
I saw that there is a topic named /mavros/rc/in at a forum, but I can not find it at http://wiki.ros.org/mavros

andre-nguyen, 2016-01-19:
It's right there http://wiki.ros.org/mavros#mavros.2BAC8-Plugins.rc_io

walkeryu123, 2016-01-19:
i see it,thank you very much.  I see the"~rc/in (mavros_msgs/RCIn)" , does it means the topc is /mavros/rc/in,and the message type I used should be mavros_msgs/RCIn?

andre-nguyen, 2016-01-19:
`~rc/in` will expand to `<node name>/rc/in` where `<node name>` is either the one you gave it in your launch file or the one specified to the `ros::init()` function https://github.com/mavlink/mavros/blob/master/mavros/src/mavros_node.cpp#L18

So yes by default `~rc/in` will expand to `/mavros/rc/in`, a topic of type `mavros_msgs::RCIn` unless you override the node name yourself.

walkeryu123, 2016-01-19:
OK,I get it.thank you for your patient answer.

closed 2016-01-19 by walkeryu123