mavlink/mavros #475: Is vision_pose related to local_position/local?

harewei, 2016-01-19:
Hello

Currently I'm using OFFBOARD mode in gazebo simulator for position control.  With mavros/setpoint_position/local, I can control the position of the vehicle.  Now, I'm publishing to the topic mavros/vision_pose/pose in http://wiki.ros.org/mavros_extras#vision_pose_estimate as follows
```
ros::Publisher vision_pos_pub = nh.advertise<geometry_msgs::PoseStamped>("mavros/vision_pose/pose", 10);
geometry_msgs::PoseStamped fake_pose;
fake_pose.pose.position.x = 0;
fake_pose.pose.position.y = 0;
fake_pose.pose.position.z = 2;
vision_pos_pub.publish(fake_pose);
```
The mavros node does not give any error or confirmation, so I tried to echo the topic mavros/local_position/local, but the positions do not match the values I publish through vision_pose.  Is this because the vehicle's local position is bound to the gazebo simulated world, or does vision_pose not suppose to change its local position?

I'm doing this simulation because later on, I wish to implement a vision system on the copter for indoor movements.

I've read on http://dev.px4.io/external-position.html that I need to set two parameters for enabling external pose input, which I have not done so because I don't know how to do it through mavros code.  If this is the problem, can someone tell me how to?

Thank you.

gokoreas, 2016-02-09:
I guess in the gazebo simulation, you are not giving any visual feedback(e.g. cameras) for the position estimator to have reference

You should publish the proper vision data so the vision estimate can have a proper estimate in position

LorenzMeier, 2016-02-09:
The gazebo simulation for PX4 is publishing optical flow.

gokoreas, 2016-02-10:
@LorenzMeier Thanks for correcting comment :)

Thomas00010111, 2016-03-24:
Hi @harewei ,
I am facing the same issue at the moment. I am running gazebo with SITL and would like to update the current position by publishing data to mavros/vision_pose/pose. The idea is to use some sort of slam algorithm later to estimate the position in the future. When echoing mavros/local_position/local the position does not change. The message I am sending should be correct I am sending the same message like you. How did you solve this issue?

Thomas

closed 2016-06-23 by vooon

weiheng209, 2016-09-19:
Hi   @Thomas00010111  
i have the same problem.    Have you already solved it? can you tell me how did you solve this issue?

NCharabaruk, 2016-10-20:
I am also having this issue using LPE

trkygt, 2016-10-28:
It seems we have some people who have the same problem including me.

khoinguyen258, 2017-07-07:
@NCharabaruk How can use solve by using LPE
this is my c code but local pose does not change anything