mavlink/mavros #476: gazebo and mavlink connection problem

JM-SNU, 2016-01-20:
I already install the gazebo6, pkg, mavros, ...

And I run 'roscore' and 'make posix_sitl_default gazebo', then I can see iris quadcopter

And then, I run roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557".

Then, I cannot receive imu data or local position data, and so on (all topics are unknown)

However, I already install all package about gazebo and mavros, and I integrate gazebo with ROS

Whaat is the problem?

( In gazebo terminal, it turns out that mavlink sim timeout for 10ms)

vooon, 2016-01-20:
My guess is no connection issue. Check diagnostics, but as i'm not simulater developer i think you should check this sites:

- http://dev.px4.io/simulation-gazebo.html - docs
- http://discuss.px4.io/ - support forum
- https://groups.google.com/forum/#!forum/px4users

harewei, 2016-01-22:
you don't actually need to run roscore
the mavlink sim timeout for 10ms won't affect the imu and position data

The steps to do:
1.make posix_sitl_default gazebo
2.roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"
3.rostopic echo mavros/local_position/local

See if this works.

nsd1992, 2016-01-27:
I am having a similar issue . Everything is installed without errors but no connection or data received on any topics.

nachi1992, 2016-01-27:
I tried  fcu_url:="udp://:14550@127.0.0.1:14557" and I can connect now  and receive data on topics. But this happens only when I am not connected to QGC.

chutsu, 2016-01-27:
@nachi1992 +1 Thank you so much, you have no idea how much time you have potentially saved us.

closed 2016-01-28 by JM-SNU