mavlink/mavros #479: Problems on official package for armhf?

Shokman, 2016-01-22:
I downloaded from the official repository for armhf (version 0.16.2) and i got this message. I am using Indigo on Ubuntu for armhf.

```
[ WARN] [1453410128.515456137]: TM: Clock skew detected (-1453410120.126361370 s). Hard syncing clocks.
[ WARN] [1453410128.813980236]: TM: Wrong FCU time.
[ INFO] [1453410129.130386787]: CON: Got HEARTBEAT, connected.
/opt/ros/indigo/lib/mavros/mavros_node: symbol lookup error: /opt/ros/indigo/lib//libmavros_plugins.so: undefined symbol: _ZN3ros5Timer9setPeriodERKNS_8DurationEb
[mavros-2] process has died [pid 2955, exit code 127, cmd /opt/ros/indigo/lib/mavros/mavros_node __name:=mavros __log:=/home/ubuntu/.ros/log/3a8fef94-c082-11e5-9b57-f4f951f0fb35/mavros-2.log].
log file: /home/ubuntu/.ros/log/3a8fef94-c082-11e5-9b57-f4f951f0fb35/mavros-2*.log
```

Installing from sources solved the issue, so you might want to check this.

vooon, 2016-01-22:
Oh, as usual armhf is outdated. Problem more related to `roscpp` (`ros::Timer::setPeriod(ros::Duration())` ABI mismatch).

Use source install.

Shokman, 2016-01-22:
No problem, I already did, just reporting

closed 2016-01-23 by vooon

davidrotor, 2016-04-13:
can you explain to me how to do the source install? and is there a way to use rosmake for it