mavlink/mavros #480: No `/local_position/local` Topic

weiweikong, 2016-01-24:
I installed mavros by 
```
sudo apt-get install ros-indigo-ros*
```
and connected with Pixhawk (Stable Firmware) by
```
roslauch mavros px4.launch
```
I could get the topic correctly such as `imu/data`, `local_position/pose` and `local_position/velocity`, however, there was no topic named `local_position/local`.
Also, I checked the ROS wiki (http://wiki.ros.org/mavros#mavros.2BAC8-Plugins.local_position),
where did not introduce such topic.
So I am confused about this topic?

LorenzMeier, 2016-01-24:
@vooon Do you still need to arm to get those? It would seem like this should not depend on arming.

vooon, 2016-01-24:
@LorenzMeier i'm not sure about publishing rates, but i think question not about "not publishing" issue.

@weiweikong Use `rostopic list`! And i'm pretty sure you missed *~* symbol, which means private node namespace (`/mavros` by default).

weiweikong, 2016-01-25:
@LorenzMeier @vooon I tried some further test.
## mavros without arming the system
- Use `roslaunch mavros px4.launch`
[image]

- Then run `rostopic list`
[image]

- It seems that there was no `mavros/local_position/local` publishing.

## mavros with arming the system
- Arm the system by RC
[image]

- Then run `rostopic list`, still no `/mavros/local_position/local`. But other nodes receive data smoothly, such as `/mavros/local_position/pose`, by running 
```
rostopic ehco /mavros/local_position/pose
```
[image]

weiweikong, 2016-01-25:
I checked the Pull https://github.com/mavlink/mavros/pull/420,
and in `master` branch, the `local_position.cpp` has, 
https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/local_position.cpp#L50-L51,
renamed the publish nodes from `local_position/local` to
```
local_position = lp_nh.advertise<geometry_msgs::PoseStamped>("pose", 10);
local_velocity = lp_nh.advertise<geometry_msgs::TwistStamped>("velocity", 10);
```
So, it could be no `/mavros/local_position/local` topic in current version, no matter installing from `master` branch from github or using `sudo apt-get install ros-indigo-mavros*`.

vooon, 2016-01-25:
Yes, just use `~local_position/pose`.

closed 2016-01-25 by vooon