mavlink/mavros #481: FCU: REJECT AUTO_RTL error continously.

rs1990, 2016-01-26:
hello,

i have mavros connected and working but everytime i launch it on my system it works but i keep getting this error.

Wasnt getting this till yesterdat but it just started. and i cant arm my system via the RC controller and. when i check on qgroundcontrol, it always goes to altitude control mode and doesnot allow me to shift to manual or offboard mode.  id appreciate any help.

'''
[ WARN] [1453843636.093903309]: GP: No GPS fix
[ERROR] [1453843641.840625256]: FCU: REJECT AUTO_RTL
[ERROR] [1453843651.833700623]: FCU: REJECT AUTO_RTL
[ERROR] [1453843661.993413307]: FCU: REJECT AUTO_RTL
[ WARN] [1453843666.164390281]: GP: No GPS fix
[ERROR] [1453843673.102744423]: FCU: REJECT AUTO_RTL
[ERROR] [1453843683.103070840]: FCU: REJECT AUTO_RTL
[ WARN] [1453843691.826531036]: CMD: Unexpected command 245, result 0
[ WARN] [1453843696.313504225]: GP: No GPS fix
[ERROR] [1453843696.924775543]: FCU: REJECT AUTO_RTL
[ WARN] [1453843701.972477033]: CMD: Unexpected command 176, result 0
[ERROR] [1453843707.090209327]: FCU: REJECT AUTO_RTL
[ERROR] [1453843718.973833929]: FCU: REJECT AUTO_RTL
[ WARN] [1453843726.372690082]: GP: No GPS fix
[ERROR] [1453843731.421973023]: FCU: REJECT AUTO_RTL
[ERROR] [1453843742.263992617]: FCU: REJECT AUTO_RTL
[ERROR] [1453843752.625998081]: FCU: REJECT AUTO_RTL
[ WARN] [1453843756.454665710]: GP: No GPS fix
[ERROR] [1453843763.073263487]: FCU: REJECT AUTO_RTL
[ERROR] [1453843774.566435424]: FCU: REJECT AUTO_RTL
[ERROR] [1453843785.163735391]: FCU: REJECT AUTO_RTL
[ WARN] [1453843786.482735168]: GP: No GPS fix
[ERROR] [1453843797.632289641]: FCU: REJECT AUTO_RTL
[ERROR] [1453843808.337995944]: FCU: REJECT AUTO_RTL
[ WARN] [1453843816.792700082]: GP: No GPS fix
[ERROR] [1453843818.479951126]: FCU: REJECT AUTO_RTL
[ERROR] [1453843828.903100726]: FCU: REJECT AUTO_RTL
[ERROR] [1453843841.311650002]: FCU: REJECT AUTO_RTL
[ WARN] [1453843846.860430358]: GP: No GPS fix
[ERROR] [1453843853.508945206]: FCU: REJECT AUTO_RTL
[ERROR] [1453843863.653112420]: FCU: REJECT AUTO_RTL
[ERROR] [1453843873.861688922]: FCU: REJECT AUTO_RTL
[ WARN] [1453843876.950545865]: GP: No GPS fix
[ERROR] [1453843884.565274485]: FCU: REJECT AUTO_RTL
[ERROR] [1453843895.300719317]: FCU: REJECT AUTO_RTL
[ERROR] [1453843905.463554321]: FCU: REJECT AUTO_RTL
[ WARN] [1453843907.038554048]: GP: No GPS fix
[ERROR] [1453843915.690748904]: FCU: REJECT AUTO_RTL
[ERROR] [1453843926.165875641]: FCU: REJECT AUTO_RTL
[ERROR] [1453843936.485730465]: FCU: REJECT AUTO_RTL
[ WARN] [1453843937.130275339]: GP: No GPS fix
[ERROR] [1453843949.483347577]: FCU: REJECT AUTO_RTL
[ERROR] [1453843959.595335940]: FCU: REJECT AUTO_RTL
[ WARN] [1453843967.245692021]: GP: No GPS fix
[ERROR] [1453843970.672248538]: FCU: REJECT AUTO_RTL
[ERROR] [1453843981.049435807]: FCU: REJECT AUTO_RTL
[ERROR] [1453843991.113163474]: FCU: REJECT OFFBOARD
[ WARN] [1453843997.383994681]: GP: No GPS fix
[ERROR] [1453844002.336098902]: FCU: REJECT OFFBOARD
[ERROR] [1453844012.790423026]: FCU: REJECT OFFBOARD
[ERROR] [1453844023.470758195]: FCU: REJECT OFFBOARD
[ WARN] [1453844027.440617384]: GP: No GPS fix
[ERROR] [1453844033.632333954]: FCU: REJECT OFFBOARD
[ERROR] [1453844043.834285528]: FCU: REJECT OFFBOARD
[ERROR] [1453844053.905118329]: FCU: REJECT OFFBOARD
[ WARN] [1453844057.514848172]: GP: No GPS fix
[ERROR] [1453844064.257942460]: FCU: REJECT OFFBOARD
[ERROR] [1453844075.473932128]: FCU: REJECT AUTO_RTL
[ERROR] [1453844085.890345551]: FCU: REJECT OFFBOARD
[ WARN] [1453844087.570834293]: GP: No GPS fix
[ERROR] [1453844096.826419254]: FCU: NOT ARMING: Switch to MANUAL mode first.
[ERROR] [1453844107.492249782]: FCU: REJECT OFFBOARD
[ERROR] [1453844117.529087197]: FCU: REJECT OFFBOARD
[ WARN] [1453844117.616947502]: GP: No GPS fix
[ERROR] [1453844129.920919591]: FCU: REJECT OFFBOARD
[ERROR] [1453844140.337098057]: FCU: REJECT OFFBOARD
[ WARN] [1453844147.708327170]: GP: No GPS fix
[ERROR] [1453844150.728682385]: FCU: REJECT OFFBOARD
[ERROR] [1453844161.925149351]: FCU: REJECT OFFBOARD
[ERROR] [1453844174.149191140]: FCU: REJECT OFFBOARD
[ WARN] [1453844177.795598898]: GP: No GPS fix
[ERROR] [1453844184.270518501]: FCU: REJECT OFFBOARD
'''

julianoes, 2016-01-27:
How exactly are you starting the simulation? It is obviously complaining about "no GPS fix" so it doesn't allow RTL.

mhkabir, 2016-01-27:
@rs1990 You have an incorrect radio calibration or mode switch configuration. 
Please follow http://px4.io/docs/flight-mode-configuration/

Not a mavros issue. Closing.

closed 2016-01-27 by mhkabir