mavlink/mavros #482: Publish low level rotor speed message

tuofeichen, 2016-01-29:
Hi folks,
I'm new to mavros and I'm using the px4 flight stack. Right now I've been using the mavros_plugins like the setpoint_velocity to control the dynamics of the flight. What I've been trying to find is since I'm sending a higher level velocity command, if I can subscribe to a topic or call a service by mavros that echoes the low level speed settings of the rotors (either the throttle PWM, or whatever). I haven't run into any specific problems that I need to know that yet, but I'd appreciate if I can access that level of detail. 

Now I'm aware I can use the actuator_control to directly send out low level control, but that's not quite what I'm looking for and I suppose I can look into what's being transmitted via Mavlink and if that information is even available. But anyway, just want to throw it out here and see if I missed anything, or if you guys can add that to the package later. 

Really appreciate the work on mavros by the way, I'm a complete noob and this package catch me up to speed quite nicely. 

Thank you.

Best Regards,
Francis

vooon, 2016-01-29:
[rc_io](http://wiki.ros.org/mavros#mavros.2BAC8-Plugins.rc_io).

closed 2016-01-29 by vooon

pmdoll, 2019-09-06:
What are the pre-conditions for `mavros_msgs/OverrideRCIn.msg` to take effect? I tried `rostopic pub -r 5 /mavros/rc/override mavros_msgs/OverrideRCIn '[1600, 1600, 1600, 1600, 1500, 1500, 0, 0]'` in manual mode but no joy.

thx