mavlink/mavros #486: setpoint_velocity plugin, how it works?

tuofeichen, 2016-02-01:
Hey guys,
Just wondering how the setpoint_velocity works. Does it have a one to one mapping between speed input and rotor speed (this would be odd with different frame size and motors and propellers)? or does it adjust the rotor speed based on IMU feedback until it reaches the desired speed?

Also, I wonder what the setpoint_velocity = 0 0 0 0 0 0 will do , if there's no GPS or px4flow or anything that can help it loiter? 

Thank you so much!

Best Regards
Francis

andre-nguyen, 2016-02-01:
> Does it have a one to one mapping between speed input and rotor speed (this would be odd with different frame size and motors and propellers)?

No. It's velocity with respect to the world.

>  if there's no GPS or px4flow or anything that can help it loiter

No. You can't expect it to calculate it's velocity with no sensors capable of measuring it. (Integrating the accelerometer over time does **not** work).

tuofeichen, 2016-02-01:
Hey thanks for the response, 
I guess just to clarify, I'm trying to use the setpoint_velocity plugin to **set** velocity instead of **read** it. But I think you get what I mean. 

1. Ok I quick follow up on that. So if I take off the propeller, the drone will keep cranking up the velocity to its maximum because it's IMU is detecting its not moving? 

2. Yeah, that's what I thought, so what will the system do if I send out a stream of velocity command like 0 0 0 0 0 0?  because I don't see it spitting out an error either?

andre-nguyen, 2016-02-01:
> Ok I quick follow up on that. So if I take off the propeller, the drone will keep cranking up the velocity to its maximum because it's IMU is detecting its not moving?

That's a question you will have to ask to the people developing whatever firmware you are using on your flight controller. Do remember that mavros is just a communication layer.

> Yeah, that's what I thought, so what will the system do if I send out a stream of velocity command like 0 0 0 0 0 0? because I don't see it spitting out an error either?

Again depends on your firmware. Assuming you're on px4 it will do nothing and eventually reject the transition to offboard mode when you attempt to do so.

tuofeichen, 2016-02-01:
I see, thanks a lot. Yeah I'm using px4 firmware, I don't know why I didn't mention that. I see, so on the other hand, if I don't have sensors for measuring the speed, this setpoint_velocity plugin is gonna be rather unreliable? 

And do I need to turn to lower level control like the actuator_control to make sure things don't go wrong? I guess then how people control the drone in offboard mode if there's no sensors available? 

Thank you so much.

closed 2016-04-29 by vooon