mavlink/mavros #489: Attitude Setpoint with APM Firmware

damanfb, 2016-02-01:
Is it possible to set a desired roll, pitch, yaw rate, and thrust through the attitude setpoint using a Pixhawk with the APM firmware on it?

vooon, 2016-02-02:
[setpoint_raw](http://wiki.ros.org/mavros#mavros.2BAC8-Plugins.setpoint_raw).

APM require specific set of ignore mask, currently not supported by other setpoint plugins.
So read firmware sources and send what you want.

closed 2016-02-02 by vooon

damanfb, 2016-02-02:
Thanks @vooon, do you have any suggestions on where to look to find what flight mode the APM needs to be in to read the setpoint_raw topic? I know with the PX4 stack that it has the OFFBOARD flight mode, but I'm hoping to stick with APM if possible for their LAND mode as I'm not using GPS and PX4 only has the RTL for some failsafes.

vooon, 2016-02-02:
APM: `GUIDED`. `GCS_Mavlink.cpp`