mavlink/mavros #490: ImportError:cannot import name setpoint

Phoenix5862, 2016-02-02:
Hey,
I want to launch setpoint_position_demo.py which is settled here: https://github.com/mavlink/mavros/blob/master/test_mavros/scripts/setpoint_position_demo
But when I launch it i get this:

Traceback (most recent call last):
  File "setpoint_position_demo.py", line 31, in <module>
    from mavros import setpoint as SP
ImportError: cannot import name setpoint

What could be the reason? Any ideas? 
Well, this code was written for PX4 but I'm using APM2. As far as I know this shouldn't be a problem to launch it since you switch to offboard(guided) mode after you launched it, or am I going the wrong way right now?
And is there any code for apm2, or any ideas what to change in this code...?
Thanks

vooon, 2016-02-02:
How do you install mavros? Did you update environment using `source <ros>/setup.bash`?

Phoenix5862, 2016-02-02:
Actually I just followed the steps from here(the part after Installation): https://github.com/mavlink/mavros/blob/master/mavros/README.md
So I don't really now if there's still something missing.

vooon, 2016-02-03:
But what type os installation: *binary* or *from source*?
If it's soutce build then you should add something like this in your `.bashrc`:

```sh
if [[ -f "$HOME/ros/devel/setup.bash" ]]; then
    source "$HOME/ros/devel/setup.bash"
elif [[ -f "/opt/ros/jade/setup.bash" ]]; then
    source "/opt/ros/jade/setup.bash"
fi
```

Where "$HOME/ros" is your catkin workspace (installation uses "~/catkin_ws").

closed 2016-04-29 by vooon