mavlink/mavros #492: launching Gazebo with ROS wrappers -> No startup of px4 app

gokoreas, 2016-02-03:
Following the [tutorials](http://dev.px4.io/simulation-ros-interface.html),

After launching gazebo using

'''
roslaunch gazebo_ros empty_world.launch
'''

I have added the IRIS model. However, the PX4 sitl app doesn't seem to startup as MAVROS cannot connect to the fcu

Am I doing something wrong?

tuofeichen, 2016-02-04:
Assuming you're using px4, you can use 
`
make posix_sitl_default gazebo
`
in the Firmware directory to start the simulation 

Also try this, instead of 14540, I ran into this problem recently as well, the tutorial's a little outdated.
`
 <arg name="fcu_url" default="udp://:14550@127.0.0.1:14557"/>
`

LorenzMeier, 2016-02-04:
@tuofeichen This is not correct for master. The MAVROS link will start on 14540.

tuofeichen, 2016-02-04:
Hey @LorenzMeier,
That's what I thought when I read the documentation, so what I did is 
`
make posix_sitl default gazebo
`
and 
`
roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"
`
however I was never able to get the heartbeat. So I asked here at
https://github.com/PX4/Firmware/issues/3368#issuecomment-175465604

and people suggest I changed that to 15550 ,which works perfectly now. I was confused at that point as well, but as long as it starts working I wasn't really digging into what's the real problem and moved on with my work. Maybe you can help clarify this? 

Thank you so much and again, sorry for the confusion here.

Best
Francis

LorenzMeier, 2016-02-04:
Francis, you must be using old code because the port has been set to 15540. What you're doing is hijacking the link to the ground control station, which is not sustainable and incorrect. Let me know if the proper approach doesn't work with master.

tuofeichen, 2016-02-04:
I see what you mean. That's my impression as well when I saw that picture of the UDP port between ground control and mavlink and stuff, but oh well. I recloned the Firmware directory and I think it works now with 15540. You're right, it must be because I have an older version of the Firmware. 

@gokoreas, my bad, the 15540 should work just fine. 

I'm still a little worried about the mavlink sim timeout warning. Maybe you can help provide some insight here as well (or should I open another issue?). The SITL is constantly spitting out this warning, however, it doesn't seem to affect the simulation. Is this common/critical and what might be the cause? 

'
WARN  mavlink sim timeout for 10 ms
WARN  mavlink sim timeout for 10 ms
WARN  mavlink sim timeout for 10 ms
WARN  mavlink sim timeout for 10 ms
WARN  mavlink sim timeout for 10 ms
WARN  mavlink sim timeout for 10 ms
'

Sorry again for the confusion. 

Best
Francis

mtsakaguchi, 2016-02-04:
Hey @LorenzMeier 

I'm running into a similar issue. I run the SITL

make posix_sitl_default gazebo

Which brings up Gazebo and the Iris+. I verify that I'm able to command the vehicle as

commander takeoff

works. However when I try to interface with MAVROS by running:

roslaunch mavros px4.launch fcu_url:"udp://:14540@127.0.0.1:14557"

the MAVROS node launches correctly with the last info line listing:

MAVROS started. My ID [1, 240], TARGET ID [1, 1]

but no HEARTBEAT is detected. Checking the /diagnostics topic also shows that the fcu connection is not connected. I have tried multiple ports as people have said above:

fcu_url:="udp://:15540@127.0.0.1:14557"
fcu_url:="udp://:15550@127.0.0.1:14557"

but still no heartbeat is detected. Any clue as to why MAVROS cannot communicate?

Thanks,

Mark

gokoreas, 2016-02-05:
@mtsakaguchi  I think I am having a different issue with yours, (Think you need a seperate issue for that!)

I CAN communicate with the quadrotor using
' make posix_sitl_default gazebo '

However, what I CANNOT make to do is using ros to launch gazebo
' roslaunch gazebo_ros empty_world.launch '
and add the IRIS+ using the insert menu in gazebo

The PX4 app doesn't seem to automatically start(There is no fcu on 14540 port)
@LorenzMeier  Any thoughts to trouble shoot this problem?

Best,
Jaeyoung Lim

andre-nguyen, 2016-02-05:
I looked into this (im the one who wrote the instructions) and I definitely made a mistake with an underscore in:

```
nosim_=1 make posix_sitl_default gazebo
```

should be 

```
nosim=1 make posix_sitl_default gazebo
```

apart from that, everything is working on my side (on a computer different from the one I wrote the instructions). My guess here is that you may have messed up setting your environment variables.

How did you do the setting the environment variables part? did you put it in a bash script and source it or did you copy paste into the terminal?

My guess is you copy pasted it in the terminal and you didn't actually add your plugin folder and somehow through some other means you were still able to add the models folder.

I'll try updating the docs to make it clearer.

gokoreas, 2016-02-05:
Thanks @andre-nguyen 
I have included below in the ~/.bashrc
However, the gazebo does not seem to include the IRIS model

```
# Set the plugin path so Gazebo finds our model and sim
export GAZEBO_PLUGIN_PATH=${GAZEBO_PLUGIN_PATH}:~/src/Firmware/Tools/sitl_gazebo/Build
# Set the model path so Gazebo finds the airframes
export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:~/src/Firmware/Tools/sitl_gazebo/models

```
I am new in using Linux so I might have made a mistake

Best,
Jaeyoung Lim

gokoreas, 2016-02-05:
And...the rotors on IRIS are not spinning in gazebo :)
could it be related to the problem?

Best,
Jaeyoung Lim

andre-nguyen, 2016-02-05:
Try replacing 
```
export GAZEBO_PLUGIN_PATH=${GAZEBO_PLUGIN_PATH}:~/src/Firmware/Tools/sitl_gazebo/Build
```
with 
```
export GAZEBO_PLUGIN_PATH=${GAZEBO_PLUGIN_PATH}:${HOME}/src/Firmware/Tools/sitl_gazebo/Build
```

same thing with the models variable

andre-nguyen, 2016-02-05:
Also, can you do:

```
end | grep gazebo
```
and paste the output here.

gokoreas, 2016-02-05:
@andre-nguyen  Thank you for your advice,

I have tried changing 
```
export GAZEBO_PLUGIN_PATH=${GAZEBO_PLUGIN_PATH}:~/src/Firmware/Tools/sitl_gazebo/Build
```
with
```
export GAZEBO_PLUGIN_PATH=${GAZEBO_PLUGIN_PATH}:${HOME}/src/Firmware/Tools/sitl_gazebo/Build
```
However, mavros still cannot access the fcu at 
'''
fcu_url:="udp://:14540@127.0.0.1:14557"
'''
The result from 
'''
end | grep gazebo
'''
is as below
[image]

andre-nguyen, 2016-02-05:
Wow, i dont know why I keep making typos.

I meant

```
env | grep gazebo
```

tuofeichen, 2016-02-05:
Hi again, @LorenzMeier ,
While I tried out two days ago the 15540 works, now I'm transferring files to a new workbench and had to reclone the master, and it stops working again. I had to go back to 15550, where I can still get the heartbeat. And just as you said, if I use 15550, I can't use qgroundcontrol. Any idea why 15540 is not working again? 

Thanks

Best
Francis

gokoreas, 2016-02-06:
@andre-nguyen 
I really appreciate your help!

It seems the path is on the right place...
[image]

gokoreas, 2016-02-06:
@andre-nguyen  I have solved the problem

I have found out that it should not be 
```
nosim=1 make posix_sitl_default gazebo
```
but

```
no_sim=1 make posix_sitl_default gazebo
``` 

This will be great help for me in my research! Thank you!

closed 2016-02-06 by gokoreas

amsandbox, 2018-07-16:
@Jaeyoung-Lim 

Hi, I am trying to make a virtual quadrotor hover using gazebo, but when I run the command
`roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"`

I receive this error: `ERROR: cannot launch node of type [mavros/mavros_node]: can't locate node [mavros_node] in package [mavros].`

After I run the simulation, I want to make my real quadrotor hover. I followed the instructions on this website [https://dev.px4.io/en/ros/mavros_offboard.html](https://dev.px4.io/en/ros/mavros_offboard.html)
but it did not work.

I am new to quadrotors. Please let me know how to do this step by step. I saw your website as well 404warehouse.

Thank you,

-Alex

andre-nguyen, 2018-07-17:
That means you haven't sourced your environment. I strongly suggest you complete the beginner and intermediate tutorials before doing anything else http://wiki.ros.org/ROS/Tutorials