mavlink/mavros #497: Arming Error caused by GPS.

Phoenix5862, 2016-02-04:
This issue is actually divided in two parts. I wasn't sure if this is the right section, but i will try.
As many other people I want control my drone offboard with mavros. My main project is the connection between TurtleBot(based on ROS) with a drone(that you can imagine what I'm aiming for). This is part of my thesis, so I'm a bit desperate, because somehow it HAS to work.

Well first of all I start my drone and so on, connect with Mission Planner via 3DR Telemetry Radio and arm using RC Transmitter(just to see that it arms). ARMED. Ok. DISARMED next step:
So now I launch the apm2_radio.launch in one terminal.  And than I use `rosrun mavros mavsys -c guided` in a second terminal. Mode changed. When I want to arm now with `rosrun mavros mavsafety arm` I get an error message in the first terminal: FCU: FCU: PreArm: GPS numsats 5 (needs 6). What exactly does this mean? Do I have to place the drone into a better position?

So now the part why I actually wrote into this section. I don't want to work with GPS. I am working indoor. So is there a way to arm within the GUIDED-mode without GPS? I read some stuff about fake-gps. But I couldn't find a description how to do this. Should mention that I'm a real beginner in this topic. The best thing would be to just skip this GPS-stuff. 
 I should also mention that can't use PX4 because than I can't use the 3dr radio telemetry(neither with MP nor with QGC) that's why I'm using APM.

I hope someone can help. 
Thanks.

Phoenix5862, 2016-02-04:
An additional information: I have GPS: 3D Fix shown on the Mission Planner, but in the terminal I get the constant warning GPS: no fix.

andre-nguyen, 2016-02-04:
Sounds like a question for the ardupilot forum

https://groups.google.com/forum/#!forum/drones-discuss

vooon, 2016-02-04:
@Phoenix5862 note that Hudro version is too outdated, try to build from master or Indigo release.

What you need is to disable PreArm cheks and provide external position info. But that all is firmware side, so check APM wiki http://copter.ardupilot.com/wiki/prearm_safety_check/

Phoenix5862, 2016-02-05:
@vooon Well, arming is not a problem anymore. At least not in the stabilize mode. Somehow i just get 5 satellites and need 6. So I cant arm in the guided mode. But shouldn't I be able to takeoff in the stabilize mode? With `rosrun mavros mavcmd takeoffcur 0.0 0 0` for example? Because when I try the request fails. It seems to be trying it a few times until the topic times out.

Phoenix5862, 2016-02-05:
Additional info: When I start the motors with the RC transmitter and use `rosrun mavros mavcmd landcur 0 0` it seems to work. Well at least the motors turn off. I don't know if this is the way i should work..maybe it just "jumps out".

Phoenix5862, 2016-02-19:
This topic can be closed.

closed 2016-02-19 by mhkabir