mavlink/mavros #498: mavros_node symbol lookup error

mtsakaguchi, 2016-02-04:
Hello,

I am trying to get the PX4 SITL up and running with MAVROS communication to the simulated autopilot. Currently the steps that I am taking are:

1. make posix_sitl_default gazebo

2. roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"

which results in the output:

[ INFO] [1454629479.826786554]: FCU URL: udp://:14540@127.0.0.1:14557
[ INFO] [1454629479.827226227]: udp0: Bind address: 0.0.0.0:14540
[ INFO] [1454629479.827271931]: udp0: Remote address: 127.0.0.1:14557
[ INFO] [1454629479.827427775]: GCS bridge disabled
[ INFO] [1454629479.830288556]: udp0: Remote address: 127.0.0.1:14557
[ INFO] [1454629479.858536232]: Plugin 3dr_radio loaded and initialized
[ INFO] [1454629479.860360673]: Plugin actuator_control loaded and initialized
[ INFO] [1454629479.861033113]: Plugin altitude loaded and initialized
[ INFO] [1454629479.896832312]: Plugin cam_imu_sync loaded and initialized
[ INFO] [1454629479.899723241]: Plugin command loaded and initialized
[ INFO] [1454629479.899753583]: Plugin distance_sensor blacklisted
[ INFO] [1454629479.903583774]: Plugin ftp loaded and initialized
[ INFO] [1454629479.908219885]: Plugin global_position loaded and initialized
[ INFO] [1454629479.908260722]: Plugin image_pub blacklisted
[ INFO] [1454629479.912673303]: Plugin imu_pub loaded and initialized
/opt/ros/indigo/lib/mavros/mavros_node: symbol lookup error: /opt/ros/indigo/lib//libmavros_plugins.so: undefined symbol: _ZN6mavros3UAS19update_attitude_imuERN5boost10shared_ptrIN11sensor_msgs4Imu_ISaIvEEEEE
================================================================================REQUIRED process [mavros-2] has died!
process has died [pid 9514, exit code 127, cmd /opt/ros/indigo/lib/mavros/mavros_node __name:=mavros __log:=/home/user/.ros/log/4196e2a8-cb99-11e5-b597-40167eabb2a6/mavros-2.log].
log file: /home/user/.ros/log/4196e2a8-cb99-11e5-b597-40167eabb2a6/mavros-2*.log
Initiating shutdown!
================================================================================
[mavros-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Any idea what the undefined symbol "ZN6mavros3UAS19update_attitude_imuERN5boost10shared_ptrIN11sensor_msgs4Imu_ISaIvEEEEE" is and how I can correct this issue?

Also, after looking through the documentation at http://dev.px4.io/simulation-ros-interface.html and previous github issues, I am still unclear on how the setup process should go. In order to run the PX4 SITL with Gazebo and MAVROS communication, does the the firmware have to be built with no_sim=1 i.e.:

1. no_sim=1 make posix_sitl_default gazebo
2. launch gazebo and add iris model
3. roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"

The above steps correctly runs the mavros_node and MAVROS indicates that it has started however no heartbeat is detected and the /diagnostics topic indicates that FCU is not connected.

Can someone point me in the right direction for a step-by-step process?

Thanks

vooon, 2016-02-05:
Little strange:
```
$ echo _ZN6mavros3UAS19update_attitude_imuERN5boost10shared_ptrIN11sensor_msgs4Imu_ISaIvEEEEE | c++filt 
mavros::UAS::update_attitude_imu(boost::shared_ptr<sensor_msgs::Imu_<std::allocator<void> > >&)
```

What version used? What ldd say:
```
ldd /opt/ros/indigo/lib//libmavros_plugins.so
ldd /opt/ros/indigo/lib//libmavros.so
```

mtsakaguchi, 2016-02-05:
@vooon 
Running the commands:

`ldd /opt/ros/indigo/lib/libmavros_plugins.so` outputs:
[cmd1.txt](https://github.com/mavlink/mavros/files/118498/cmd1.txt)

`ldd /opt/ros/indigo/lib//libmavros.so` outputs:
[cmd2.txt](https://github.com/mavlink/mavros/files/118499/cmd2.txt)

Thanks

vooon, 2016-02-05:
Hmm, looks normal. This:
```
objdump ~/ros/devel/lib/libmavros_plugins.so -t | c++filt | grep attitude
```

(you may post here as code block)

mtsakaguchi, 2016-02-05:
@vooon 
Running the commands:

`objdump /opt/ros/indigo/lib/libmavros_plugins.so -t | c++filt | grep attitude` does not output anything

`objdump /opt/ros/indigo/lib/libmavros.so -t | c++filt | grep attitude` does not output anything

vooon, 2016-02-05:
Ahh, it's stripped. @mtsakaguchi i recommend build from source.

mtsakaguchi, 2016-02-09:
@vooon 
I'm having issues building mavros from source. Everything goes well until I execute `catkin build`. `catkin build` then executes and it says "mavros_msgs" "mavlink" and "libmavconn" Finished building but when it gets to "mavros" it gets to [build -34.0] [mavros -3.8] and stops and the computer freezes and restarts. I've tried starting from a new workspace and this happens everytime. Any ideas what could be causing it?

vooon, 2016-02-10:
Looks like your computer overheats or your out of memory.
Compilling process takes both CPU and RAM. So check cooling and check swap usage.
If you do not has a swap partition/file make it first. Also you can limit concurent process count with `-j 2` flag. It also reduces RAM usage, but build are slower.

On ARM machines usually good idea is to limit to 2 processes (mainly because slowish eMMC/SD and low RAM: 1 GiB enough only for two g++ on some projects).

mtsakaguchi, 2016-02-19:
@vooon 
Thanks for the help!
I was able to build from source with the -j 2 flag and now I'm able to connect to the SITL using mavros.

closed 2016-02-19 by mtsakaguchi