mavlink/mavros #502: Convert mavros to mavlink - error: ‘move’ is not a member of ‘std’

hsilva7, 2016-02-08:
I'm getting this error while trying to convert mavros to mavlink:

"error: ‘move’ is not a member of ‘std’"

I'm using this instruction:
`mavros_msgs::mavlink::convert(msg,rmsg);`

I also get this warning:

"/opt/ros/indigo/include/mavconn/mavlink_dialect.h:22:3: warning: #warning "No MAVLINK_DIALECT specified. fallback to " MAVLINK_DIALECT [-Wcpp]
 # warning "No MAVLINK_DIALECT specified. fallback to " MAVLINK_DIALECT
   ^"

vooon, 2016-02-08:
Please post source code link / part of file. Did you enable C++11 (`-std=c++11`)?
What is `msg`, `rmsg`?

hsilva7, 2016-02-08:
Should I enable `-std=c++11` like: `catkin_make -std=c++11`?

msg and rmsg are the input and output message respectively.

```
mavlink_message_t rmsg;
mavros_msgs::Mavlink msg;

mavros_msgs::mavlink::convert(msg,rmsg);
pub.publish(rmsg);
```

The purpose is to convert the message (msg) to mavlink, which with the `mavros_msgs::mavlink::convert` to be converted to rmsg.

vooon, 2016-02-08:
You should edit CMakeLists.txt to add c++11 compiler flag. Code incorrect: you need convert **m**msg (mavlink_message_t) to **r**msg (::Mavlink) and then publish rmsg.

hsilva7, 2016-02-09:
It solved the previous error, but now I get an error when passing values to mmsg.payload64.

```
mavlink_message_t mmsg;
mavros_msgs::Mavlink rmsg;

std::vector<long unsigned int > payload64(6, 0);
(...)
mmsg.payload64 = payload64;
mavros_msgs::mavlink::convert(rmsg,mmsg);
pub.publish(rmsg);
```
The error I get:
```
error: incompatible types in assignment of ‘std::vector<long unsigned int, std::allocator<long unsigned int> >’ to ‘uint64_t [33] {aka long unsigned int [33]}’
         mmsg.payload64 = payload64;
```

Thank you

vooon, 2016-02-09:
What did you trying to do? Payload quads are used by `mavlink_*_decode()`, `mavlink_*_pack_chan()`, not by hand.

hsilva7, 2016-02-09:
I'm trying to send a message to a pixhawk, this is a custom message that has been created on the pixhawk side, and now I'm trying to send from a ROS application information to the pixhawk.

vooon, 2016-02-10:
Current implementation of mavlink (v1.0) drops all unknown packets. So be sure that yout mavros build with same headers.

I done plugin system exactly for do all converting in mavros_node and use clear ROS messages in your nodes.
Yes you can produce Mavlink.msg in your node, but in plugin you has all needed environment, and in your code you have to do this yourself.

Pluginlib can load mavros plugins from any package with proper definition, example is mavros_extras.

So i recommend first to make define message and write a plugin.

closed 2016-04-29 by vooon