mavlink/mavros #503: Can't change yaw_rate with setpoint_raw/attitude

damanfb, 2016-02-11:
I'm trying to control the roll and pitch angles, yaw rate, and altitude and running into trouble. I am using the setpoint_raw/attitude command. I can control my throttle through it, but I don't seem to get any response on the yaw_rate command. 

I have tried setting the type_mask to 128 to completely ignore attitude while I'm debugging but I still don't get any yaw_rate command. I have also tried setting the type_mask to 3 to ignore roll and pitch _rates_ but get the same response. Does anyone know the best way to control roll and pitch _angles_ while still controlling yaw _rate_?

closed 2016-06-23 by vooon

JJJJJJJack, 2016-12-05:
Hey, same issues here. The yaw_rate won't affect anything on the quad and when I set the yaw angle to 0, it will always try to align itself to the north.

JJJJJJJack, 2016-12-05:
Oops, Hi Daman.

damanfb, 2016-12-05:
Hahaha hey.

I got around it by making me own yawn rate controller and updating yaw.
read in the current yaw to "zero" it out and then add yaw_rate  times delta
the each time step.

JJJJJJJack, 2016-12-05:
Always good to have help from the expert.