mavlink/mavros #504: iris_empty_world.launch not part of rotors_simulator master branch

JMLambert, 2016-02-11:
This launch file is required by `iris_empty_world_offboard_ctl.launch` which is part of test_mavros. It can be found, however in the PX4-rebase_v2 branch of rotors_simulator.

vooon, 2016-02-11:
test_mavros little outdated. Read px4 docs: http://dev.px4.io/simulation-gazebo.html

JMLambert, 2016-02-11:
Is there a model that can be loaded in rviz or do they all run in gazebo exclusively?

TSC21, 2016-02-11:
> Is there a model that can be loaded in rviz or do they all run in gazebo exclusively?

Rviz is not a simulation environment, while Gazebo is. And you need Gazebo to test the simulation. Also you can try launching a new model instead. Will update the test environment with new models soon.

JMLambert, 2016-02-12:
Actually you can use RViz to perform path planning and kinematic simulation(not dynamic like Gazebo does) which is what I would like to use it for. You can also simulate sensor output. Combining ROS with Gazebo gives you a GCS/Real World paradigm. I'll see if I can load the Gazebo models in ROS, maybe with small modifications. Is there a mailing list I can subscribe to to get info related to rotors_simulator, mavros and such?

Thanks very much for getting back to me.

crzx5, 2016-03-16:
RViz is a visualization tool. It does not simulate sensor output; rather, it allows you to visualize the sensors that are simulated in Gazebo. Whether or not RViz runs will not affect path-planning algorithms...

closed 2016-06-23 by vooon