mavlink/mavros #505: setpoint_accel plugin

tuofeichen, 2016-02-12:
Hey guys,
I was wondering if the setpoint_accel plugin is supported by px4 firmware now. I was a little confused because:
1. According to this https://github.com/mavlink/mavros/blob/master/test_mavros/README.md, 
2, I followed the exact same procedure for sending out setpoint_accel as setpoint_velocity, but the SITL and Pixhawk both didn't seem to respond to it. Whereas setpoint_velocity works perfectly fine for me. 

Thank you!

Best Regards,
Francis

TSC21, 2016-02-12:
I don't think it is. But you may confirm this on the mc_pos_control code of PX4 Firmware.

tuofeichen, 2016-02-12:
Hi @TSC21 
From the first glance, doesn't seem like it, but extensive documentation on px4 website referenced it should be implemented, I asked the same question on px4 github and I will dig into the code a little deeper and follow up on this thread later

TSC21, 2016-02-12:
Great. We will be happy if you are able to get it to work. Let us know.

Stifael, 2016-10-22:
I think my question is related to this issue: I am able to send setpoint_accel (at least messages are sent), but the vehicle is not really doing what I would expect. For instance, sending zero acceleration, I would expect to hover, but it turns out that it just drops (and with 0.6 already accelerates a lot in the z-direction). What do the mavros acceleration command represent?

TSC21, 2017-04-04:
@Stifael the mavros accel command should represent the accel in a certain axis. So if you send a 0 value means that there is no accel on that axis, meaning there is no throughput to the motors I suppose. Right now I don't now how's the current implementation on the FW side. @tuofeichen have you got any answer regarding this? @mhkabir do you know how is this implemented on `mc_pos_controler`? (if it is)

TSC21, 2017-06-29:
PX4 handles force setpoint - https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L886-L902. But after searching a bit, I don't see this msg being used anywhere else on code, so I believe it's just a uORB topic being published in the void, without being subscribed.
For APMCopter: https://github.com/ArduPilot/ardupilot/blob/afc382c342e09a139d9b2ac1e589f07a42fea8c9/ArduCopter/defines.h#L495-L496, https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L1752-L1762, https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L1672-L1683, so seems that the definition is there but it's not considered the case of accel. Besides, https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L1635, which means there's the possibility of using forces, but no implementation yet.
So this is something related and to be covered in the Firmware side. Closing.

closed 2017-06-29 by TSC21