mavlink/mavros #508: Fail to build mavros on Raspberry pi from source

sputnick1124, 2016-02-15:
I am trying to install mavros from source as described in the README. I added mavlink and mavros to my rosinstall file and tried to proceed from there, but it complained about some undefined packages (control_toolbox and some others) for  OS[debian]. I added mavros_extras to the rosinstall file and `rosdep install` ...blah... did not complain anymore.

Running catkin build, however, fails after building almost all of the packages when it gets to mavros.

```
 ./build_env.sh /usr/bin/make -j1
[  0%] Built target mavros_msgs_generate_messages_cpp
[ 27%] Built target mavros
[ 30%] Built target gcs_bridge
[ 33%] Built target mavros_node
[ 36%] Building CXX object CMakeFiles/mavros_plugins.dir/src/plugins/sys_status.cpp.o
/home/pixhawk/mr_catkin_ws/src/mavros/mavros/src/plugins/sys_status.cpp: In member function ‘void mavplugin::SystemStatusPlugin::autopilot_version_cb(const ros::TimerEvent&)’:
/home/pixhawk/mr_catkin_ws/src/mavros/mavros/src/plugins/sys_status.cpp:698:26: error: ‘MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES’ was not declared in this scope
    cmd.request.command = MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES;
                          ^
CMakeFiles/mavros_plugins.dir/build.make:77: recipe for target 'CMakeFiles/mavros_plugins.dir/src/plugins/sys_status.cpp.o' failed
make[2]: *** [CMakeFiles/mavros_plugins.dir/src/plugins/sys_status.cpp.o] Error 1
CMakeFiles/Makefile2:1016: recipe for target 'CMakeFiles/mavros_plugins.dir/all' failed
make[1]: *** [CMakeFiles/mavros_plugins.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

I see that this issue has come up before (#240) but I am not running ubuntu, so the package referenced in that solution (even the armhf build) does not install on the Pi.

If it helps, the version number recorded in `mavlink.pc` is 2016.2.5.

I don't think I'm forging new roads by installing mavros on a RPi. Is there some known resolution to this hurdle? It appears that `MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES` is supposed to be an `enum`. Could I just stick its definition/declaration in a header file somewhere to fix the issue?

TSC21, 2016-02-15:
mavros version and ros version?

sputnick1124, 2016-02-15:
mavros: 0.17.0
ros: 1.11.16

Sorry. Should have included that from the get-go.

TSC21, 2016-02-15:
Are you on Jade or Indigo?

TSC21, 2016-02-15:
Trying using `gdebi` and install one of the packages I suggested above, depending on the ROS edition (Jade or Indigo).

vooon, 2016-02-15:
@sputnick1124 what instructions are you used? Current one describe usage catkin tool (**not** catkin_make) and pulls latest mavlink headers.

Problem definitely in mavlink package version, if you installed it via apt remove first.

TSC21, 2016-02-15:
> Problem definitely in mavlink package version, if you installed it via apt remove first.
:+1:

sputnick1124, 2016-02-15:
I've used th instructions at [https://github.com/mavlink/mavros/tree/master/mavros](url) with the exception that I needed to add `--os=debian:wheezy` to the `rosdep` command and I am installing indigo instead of jade. Also I added `--cmake-args -DMAVLINK_DIALECT=common` to the `catkin build` command.

I didn't install anything via apt. Everything has been done from source.

@TSC21 gdebi complains about unmet dependencies (ros-indigo-catkin) when trying to install [.](http://packages.ros.org/ros-shadow-fixed/ubuntu/pool/main/r/ros-indigo-mavlink/ros-indigo-mavlink_2016.2.5-0trusty-20160204-174534-0800_armhf.deb) This is the same error as my initial attempt to follow the advice in issue #240.

@vooon Do I need to pull the source for a different mavlink version from somewhere?

vooon, 2016-02-15:
`rosinstall_generator` should select mavlink from mavlink/mavlink-gpb-release.git check that you have mavlink folder in `src`.

vooon, 2016-02-15:
Then when you do `catkin build` you should see `mavlink` package, and it should be build before `libmavconn`.

sputnick1124, 2016-02-15:
<my_catkin_ws>/src/mavlink exists.

vooon, 2016-02-15:
Also try this:
```
catkin build --verbose -i -j1
```

sputnick1124, 2016-02-15:
Okay. I think I'm starting to understand. I'll try the rebuild with more output and update here in a little bit. Thank you both for all of your help.

TSC21, 2016-02-15:
be very precise with this:
```bash
$ rosinstall_generator --upstream mavros | tee /tmp/mavros.rosinstall
$ rosinstall_generator --upstream-development mavros | tee /tmp/mavros.rosinstall
$ rosinstall_generator mavlink | tee -a /tmp/mavros.rosinstall
$ wstool merge -t src /tmp/mavros.rosinstall
$ wstool update -t src
$ rosdep install --from-paths src --ignore-src --rosdistro indigo --os=debian:wheezy -y
$ catkin build --cmake-args -DMAVLINK_DIALECT=common -j1 -l1
```

vooon, 2016-02-15:
@TSC21 second call to `rosinstall_generator` is alternative to first!

With first you get latest released version from this repo (`--upstream`, without that flag you will use `mavros-release.git`).
With second `--upstream-development` is to pull master.

TSC21, 2016-02-15:
@vooon yeah you're right. Forgot that detail.

TSC21, 2016-02-15:
Well for better or worst, maybe we can keep up to `master` then? So maybe it's better to use `--upstream-development`?

sputnick1124, 2016-02-15:
Should I use `--upstream-development` or just the `--upstream` branch? (my current situation has arisen using the release branch).

vooon, 2016-02-15:
Currently master has one difference from 0.17.0 - APM:Sub support.

To disable control_toolbox you may blacklist test_mavros.

TSC21, 2016-02-15:
@sputnick1124 it doesn't make a difference as this is `mavros` related, not `mavlink`. Try `--upstream-development` if you want to have current `master` of `mavros` built.

TSC21, 2016-02-15:
> To disable control_toolbox you may blacklist test_mavros.

Or, you can added it to your `catkin_ws` and build it from source: https://github.com/ros-controls/control_toolbox

vooon, 2016-02-15:
Better via `rosinstall_generator control_toolbox` (same tee trick).

rosinstall_generator also may pull all dependencies (`--deps`), but it then take most roscore packages.
Or it is only on my source installation? Because on 15.10 i build all from source...

TSC21, 2016-02-15:
> Because on 15.10 i build all from source...

So probably we should stick to build it from source so it doesn't mess with the rospkgs.

sputnick1124, 2016-02-15:
Update:
I attempted the build once more with `--verbose`. Mavlink finishes first, then libmavconn builds to get the same error as above. 

Digging through the output log of the build attempt, though, I found the following in the verbose output of the mavlink build:
```
Checking .pth file support in /home/pixhawk/mr_catkin_ws/devel/mavlink/lib/python2.7/dist-packages/
/usr/bin/python -E -c pass
TEST FAILED: /home/pixhawk/mr_catkin_ws/devel/mavlink/lib/python2.7/dist-packages/ does NOT support .pth files
error: bad install directory or PYTHONPATH

You are attempting to install a package to a directory that is not
on PYTHONPATH and which Python does not read ".pth" files from.  The
installation directory you specified (via --install-dir, --prefix, or
the distutils default setting) was:

    /home/pixhawk/mr_catkin_ws/devel/mavlink/lib/python2.7/dist-packages/

and your PYTHONPATH environment variable currently contains:

    '/opt/ros/indigo/lib/python2.7/dist-packages'

Here are some of your options for correcting the problem:

* You can choose a different installation directory, i.e., one that is
  on PYTHONPATH or supports .pth files

* You can add the installation directory to the PYTHONPATH environment
  variable.  (It must then also be on PYTHONPATH whenever you run
  Python and want to use the package(s) you are installing.)

* You can set up the installation directory to support ".pth" files by
  using one of the approaches described here:

  https://pythonhosted.org/setuptools/easy_install.html#custom-installation-locations

Please make the appropriate changes for your system and try again.
[mavlink] <== '/home/pixhawk/mr_catkin_ws/build/mavlink/build_env.sh /usr/bin/make install' finished with return code '0'
```
I just added the local install directory to my PYTHONPATH and started it going again. I'll post here with results.

TSC21, 2016-02-15:
```bash
export PYTHONPATH="/home/pixhawk/mr_catkin_ws/devel/mavlink/lib/python2.7/dist-packages/:$PYTHONPATH"
```

vooon, 2016-02-15:
Little strange. Do you `source /opt/ros/indigo/setup.bash` before build?

By default you should source setup.bash when following ros installation instructions (`.bashrc`).
Note that after build you need to source `devel/setup.bash`.

sputnick1124, 2016-02-15:
I `source /opt/ros/indigo/setup.bash` by default (in my `.bashrc`). After prepending the `.../devel/mavlink/.../dist-packages` to `PYTHONPATH`, it seems that mavlink builds correctly now with `catkin build --cmake-args -DMAVLINK_DIALECT=common -j1`, but mavros still failed. I realized that `libmavconn` built rather quickly, so i entered `.../build/libmavconn` and did `./build_env.sh make clean && ./build_env.sh make`. Then I entered `.../build/mavros` and did the same `clean && make` cycle. `mavros` still fails.

Would it be worth deleting `/opt/ros/indigo` and all `catkin` workspaces and starting over from scratch?

vooon, 2016-02-16:
Try to clean workspace first. `catkin clean --all`.
Also you may try to install headers by hand:
```
cd src/mavlink
catkin build --this -j1 --verbose
```

After that you should see headers in `devel/include/mavlink/v1.0/`

sputnick1124, 2016-02-16:
Interesting. `catkin clean` complains about `unknown verb 'clean'`. Is my `catkin` broken?

vooon, 2016-02-16:
Nope, misprint `catkin clear --all`

sputnick1124, 2016-02-16:
Success!! A summary of the issues and what worked in the end to fix them:
1) My `PYTHONPATH` variable did not have the local (catkin_ws) library path in it causing `mavlink` to fail while building. Prepending the path with 
`export PYTHONPATH=$HOME/mr_catkin_ws/devel/lib/python2.7/dist_packages:$PYTHONPATH`
fixed the `mavlink` build.
2) From the previously failed build, the `build/` directories were polluted with possible other malformed packages causing `mavros` to fail with a complaint of an undeclared variable. Cleaning the entire workspace with
`catkin clean --all`
and rebuilding with 
`catkin build --cmake-args -DMAVLINK_DIALECT=common -j1 --verbose` 
succeeded in building the entire setup.
3) My `catkin` tool was acting up with complaints about `unknown verb`s. Updating `catkin` with
`sudo pip install -U catkin-tools`
and re`source`ing `/opt/ros/indigo/setup.bash` gave `catkin` the `clean`, `config`, `create`, and other missing verbs.

The final build took over 2.5 hours on my Raspberry Pi 2 running the latest Raspbian based on Debian Jessie (the lite version). I don't exactly recall what I did to put together my `.rosinstall` file, but it contained 69 packages. With just `mavros` and `mavlink`, there wee complaints from `rosdep` about `undefined package for OS [debian]`. I think I kept `rosinstall_generator`ing the `undefined package`s into my `.rosinstall` file until `rosdep` didn't complain anymore. There is probably a better way to go about it, but I think I have a working set up now, so I'm not worried.

Many thanks to @vooon and @TSC21 for your help!

sputnick1124, 2016-02-16:
For the sake of documentation, my `mavros.rosinstall` file.
[mavros.rosinstall.txt](https://github.com/mavlink/mavros/files/132743/mavros.rosinstall.txt)

TSC21, 2016-02-16:
Great! I ask you to please close the issue if you think it is solved and also do a PR to the README.md file with a description of the aditional steps required to setup mavros and mavlink on a RPi. Thanks

closed 2016-02-16 by sputnick1124

vooon, 2016-02-16:
@TSC21 i'm not sure that on ubuntu 14.04 (official packages) there were similar problems. Maybe on raspbian? But according to http://dev.px4.io/ros-raspberrypi-installation.html maybe not...

@sputnick1124 seems that you build all ros code packages from source. I recommend to setup distcc, cross-compilation on x86_64 give huge boost.

TSC21, 2016-02-16:
@vooon yeah this is RPi OS release related, not really Ubuntu related. But we can add a guide on the README file so to setup it correctly on RPi systems. That was what I was suggesting. Also @mhkabir has some experience with RPi-based autopilots (with Navio), so maybe he can also give a hand here.