mavlink/mavros #510: mocap/pose only updating position in local_position/pose  .. orientation data wrong.

shiohigari, 2016-02-19:
I am using an odroid with ubuntu 14.04 server running mavros to talk to my pixhawk running the px4 firmware. I can update the the pixhawk's local position correctly using the mocap topic however the orientation data is not updating and I find my drone cannot hold its yaw orientation with a slow drift over time. 

The mocap and local pos topics are shown below. 
I have the ATT_EXT_HDG_M parameter set to 2 and everything else going into the ekf set to 0 other than mocap. I tried setting ATT_EXT_HDG_M to 1 and using the vision topic instead but had the same result. 

Is this normal?

[image]

vooon, 2016-02-19:
Please ask at PX4 support forum. Mavros only convert frames while data interpretation is on firmware side.

walkeryu123, 2016-05-27:
hi ,I want to use vision to estimate the position,can you tell me where can i set ATT_EXT_HDG_M? i tried massion plannner and QGroungcontrol,but I can not find the parameter

quentin1002, 2016-06-06:
@walkeryu123  in my case it is available in QGC... 

@shiohigari if you have any answer to your question on px4 side, could you please paste the link here? Tks!

closed 2016-06-23 by vooon