mavlink/mavros #511: Optical Flow Message

tuofeichen, 2016-02-19:
Hey guys, 
I was a little confused by this message:
`~px4flow/raw/optical_flow_rad (mavros_msgs/OpticalFlowRad)`

Specifically, is the integratedx the position estimate? or is it the still the velocity estimate? I ran the px4flow_node here http://wiki.ros.org/px4flow_node and it seems to yield reasonable result. However, when I tried to capture and plot the  `~px4flow/raw/optical_flow_rad', I can't quite make sense out of the integreatedx and integratedy measurement either way (as position or as velocity).

Also, even though the topic name is quite  self-explanatory, I just want to make sure this is the raw pose estimate from the flow sensor ONLY right? not fused with any IMU data or any other velocity/position sources. 

Thanks for the clarification!

Best
Francis

mhkabir, 2016-02-20:
@tuofeichen IntegratedX is integrated (accumulated) optical flow rate around body X axis (similar to a gyroscope). And yes, this is raw data.

tuofeichen, 2016-02-20:
Hi @mhkabir,
Sorry so what's the unit of it? Is that a m/s or anything like that ? cuz my understanding is flow rate raw data is something like a pixel/s or something?

Thank you so much!

Best
Francis

tuofeichen, 2016-02-20:
oh wait I think I got it, this is the radian .... duhh... no wonder I'm confused.

Thanks!

mhkabir, 2016-02-21:
Radians yes. See the i2c integral frame on the  PX4Flow page. 

pixhawk.org/modules/px4flow

closed 2016-02-21 by mhkabir