mavlink/mavros #512: PX4 SITL/mavros error "unexpected command 176" using setpoint_* topics

mtsakaguchi, 2016-02-22:
Hello,

I am running the PX4 SITL with Gazebo and am trying to command the attitude using the setpoint_* topics. I am able to successfully connect over UDP with mavros however when I publish to the /mavros/setpoint_raw/attitude topic, the quadcopter in the Gazebo environment immediately falls to the ground after commanding to OFFBOARD mode. The "make posix_sitl default gazebo" terminal output then prints out the warning "uORB::DeviceNode::publish called with invalid handle" and the terminal output from the mavros node prints out the warning "CMD: Unexpected command 176, result 0". 

This happens when I publish to both the /mavros/setpoint_attitude/attitude and /mavros/setpoint_raw/attitude topics however I am able to command position using the /mavros/setpoint_position/local topic.

Any help would be appreciated.

vooon, 2016-02-23:
Command [176 - MAV_CMD_DO_SET_MODE](http://mavlink.org/messages/common#MAV_CMD_DO_SET_MODE) not a problem. PX4 resend `SET_MODE` message as command on internal bus.

Seems bug somewhere in mavlink module of firmware. Check [SET_ATTITUDE_TARGET](http://mavlink.org/messages/common#SET_ATTITUDE_TARGET) handler. Also ask at PX4 forums for support.

closed 2016-04-29 by vooon

csorice147, 2016-10-25:
@vooon I'm trying to use gazebo + offboard mode as well. 

I'm able to connect, and I publish to mavros/setpoint_attitude/attitude topic successfully, but the copter keeps on arming/disarming itself for some reason. It never takes off. Just sits on the ground.
I am using this code ([this example](http://dev.px4.io/ros-mavros-offboard.html)) and it works perfectly when I publish /mavros/setpoint_position/local. But when I change the topic to be attitude, it doesn't work.

Sometimes I get this warning: CMD: Unexpected command 176, result 0. Sometimes I get this warning: CMD: Unexpected command 400, result 0. When I look at the /mavros/state, it is connected and in offboard mode, but it keeps flipping between arm/disarm. 

Any advice?

physics-lzs, 2019-04-07:
have you solve this problem？i meet the same problem, can you give me some help? thanks

thealchemist89, 2020-09-30:
Hi @physics-lzs, in case you are still having this issue, you may need to publish setpoint_attitude/thrust and give it a value that's > 0.5 to see some effects. If you continue to see the same observation as earlier posts, publish attitude commands to setpoint_raw/attitude instead. Again, remember to give a value to your thrust :) Hope this helps!

Iko-Git, 2020-12-03:
Could you please provide more clarification @thealchemist89 , I am facing this issue while running the example [here](https://dev.px4.io/master/en/ros/mavros_offboard.html) , sometimes it works and some other times running the ROS node keeps giving me 
[ INFO] [1606989220.873892028]: Offboard enabled
[ INFO] [1606989223.420495370]: AUTO.LAND
[ INFO] [1606989223.424325004]: Offboard enabled
[ INFO] [1606989225.970496764]: AUTO.LAND
[ INFO] [1606989225.973307611]: Offboard enabled
Somtimes, I can work the ROS node successfully again by restarting
roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"
and some other times it requires me to restart the whole computer

yagomrs, 2022-01-30:
Which version of PX4-Autopilot are you using?

I have tested a lot of possibilities here and with the last release (version 12) I have this problem, the drone stays in fail safe mode and it doesn't take off. A friend of mine tested the same commands on his PC and works, so we tried to step back in the version, now  I'm using version PX4-Autopilot 11.3 (the last version that worked for me) and it is working fine now!

To flight using commander directly the PX4 it is needed to define the takeoff point to avoid error. To flight using script you just keep the position and the drone stays stable! 

About the PX4-Avoindace I haven't success to run on my actual PC.

OS: Ubuntu 18
ROS: Melocid
Gazebo: 9
PX-Autopilot: 11.3.0

KBhanup, 2022-05-10:
Thank You!   "unexpected command 176" , this was resolved for me when I started sending position to /mavros/vision_pose/pose... Earlier my Vrpn client was turned off.