mavlink/mavros #516: local_position twist and pose as a ***WithCovarianceStamped msg

quentin1002, 2016-03-02:
Hi! 

I want to interface mavros with some SLAM algorithms. To perfom an efficient sensor fusion, I need msgs of the type ***WithCovarianceStamped. Unfortunatly, the local_position messages don't include the covariance. Is there a simple way to get the messages with the covariance? I am still new to ROS programming... 

For you info, I am using both px4 pixhawk and px4flow. I guess that we could use the covariance derived from the IMU to add it the local_position msgs... But I really don't know how to do that.

Thanks for you help

scott-eddy, 2016-03-07:
While mavlink supports a [LOCAL_POSITION_NED_COV](https://pixhawk.ethz.ch/mavlink/#LOCAL_POSITION_NED_COV) message that passes the estimate and the covariance, and the PX4 can stream that message, there is no mavros plugin to parse it yet.  You would have to alter local_position plugin to get that data.  

If you do you should submit a PR with the changes

quentin1002, 2016-03-08:
In the mean time, I developed a solution that looks better to me. Inspired by the [hector_quadrotor](https://github.com/tu-darmstadt-ros-pkg), I broke down the process like that: 

**The TF tree: **
base_link -> base_stabilized: I use the IMU information returned by mavros in the [hector_imu_attitude_to_tf](https://github.com/tu-darmstadt-ros-pkg/hector_slam/tree/catkin/hector_imu_attitude_to_tf) node

base_stabilized -> base_footprint: I simply subscribe to the altitude topic of mavros and publish the z translation. I still have some reliability problems as the barometer is not really accurate. I might try to use the sonar height combined with the IMU information. To be continued

base_footprint -> odom: I use [robot_localization](http://wiki.ros.org/robot_localization). It is working in 2d mode. It combines the IMU and the optical flow. To use the [mavros_msgs/OpticalFlowRad](http://docs.ros.org/jade/api/mavros_msgs/html/msg/OpticalFlowRad.html) in robot_localization, I use make use of [optflow_odometry](https://github.com/lalten/tas/blob/master/tas_odometry/src/optflow_odometry.cpp). At the end of the day, robot_localization is also publishing my 2D odom.

odom -> map: I use another EKF, again [robot_localization](http://wiki.ros.org/robot_localization), to fuse my 2D odom with my SLAM position. 

For the moment, everything looks good, except the base_stabilized -> base_footprint. This problem does not affect the system. Do you think my solution is a good one, compared to the use of a single mavros usage?  Am I creating to much computing and processing requirements or maybe inducing errors?

closed 2016-06-23 by vooon