mavlink/mavros #520: POSCTL fly mode with mavros

Agachon, 2016-03-11:
Hi guys,
I need help ! My setup is the following : 
Quadrotor with pixhawk;
PX4 with updated firmware;
Companion computer with updated mavros. 

I use mocap, I remapped /mocap/pose to /mavros/mocap/pose. When I checked in QCS, everything seems good for the position x,y and z and for the attitude.

When I'm flying in manual mode, all is good.
When I switch my remote to ALTCTL, altitude is controlled automatically as expected. Roll and pitch are in manual mode.
When I switch my remote to POSCTL, the quad immediately oscillates and I have to switch to manual mode quickly to avoid big crash! My gains (attitude and position controller seem good). 

Any ideas ?
Thx in advance

andre-nguyen, 2016-03-11:
Why would you think your gains are good when your quad oscillates?

Agachon, 2016-03-11:
Thx Andre for your comment.
I think it is not the gains since I already fly inside and outside with GPS and the same gains without any issues.

mhkabir, 2016-03-12:
Something is obviously wrong on the estimator side. Please refer to Firmware mailing lists or Disqus, with full logs. 

This is not a Mavros issue at first glance.

Agachon, 2016-03-14:
Ok thx @mhkabir . I'm going to check with Firmware group.
I'll come back here to close when I'll find issue

closed 2016-06-23 by vooon