mavlink/mavros #521: Offboard Reject (was previously working)

MarcGyongyosi, 2016-03-13:
Hi,

Did something major change around the offboard setpoint plugin? I had offboard mode working using the setpoint_position plugin until I recently decided to updated to the newest version of mavros (alongside a firmware upgrade on the pixhawk to the latest px4 flightstack). 

I am feeding my vision estimates to the Pixhawk through the vision estimate topic. I verified that the position is estimated properly on both sides (pixhwak and offboard vision). I am feeding setpoints to the setpoint_position plugin which get registered by it (I verified by adding a printout every time it receives a setpoint in the setpoint_position.cpp). 

Once I verify proper initialization of all systems and make sure the setpoints are published, I expect to be able to switch into offboard mode via a switch on my RC after arming. I can switch into altitude control, but not into position control. 

When flipping the offboard control switch, I keep getting an offboard rejected 176 error. 

Maybe someone can help me out tracking down this error...

Thanks!

Marc

vooon, 2016-03-13:
In current release (0.17) only one major change - changed frame transformations.
And no changes in plugins logic, so i think newer firmware require some configuration.

MarcGyongyosi, 2016-03-14:
hmm this is very weird... @LorenzMeier do you happen to have any insights into what could have changed? 

I have CBRK_NO_VISION set to 0 and the companion link parameter set to 921600. I am getting proper data from the pixhawk (via UART to FTDI). I am able to switch into position control and altitude control. I also increased the setpoint publisher rate to 20Hz. As verified through printouts, the mavros setpoint plugin is receiving my setpoints. Any ideas as to what could be going on here? 

Thanks!
Marc

MarcGyongyosi, 2016-03-20:
wrong component id..... fixed

closed 2016-03-20 by MarcGyongyosi

vooon, 2016-03-20:
FYI: `rosrun mavros checkid` program checks this type of errors.

MarcGyongyosi, 2016-03-20:
Yep that's how I figured it out eventually. Thanks a lot!

Btw: has anyone played with the idea of building a stripped down version of
mavros for situations where only a very limited number of
functionalities/features are needed? Might be interesting if one doesn't
want to write it from scratch a la c_uart example and only needs core
things like IMU, vision position, setpoint control