mavlink/mavros #524: Best practice for creating mavlink dialect that will play nicely with mavros/px4

locustcox, 2016-03-23:
(Not sure how to add the "**question**" label, but feel free to add it.)

I've created a custom mavlink message dialect by creating a custom xml that includes common.xml as well as some custom message. I ran the message generation script and installed my dialect in opt/ros/indigo/include/mavlink/v1.0 as well as the px4/Firmware/mavlink/.../v1.0 folder. I then modified the px4 to include <v1.0/custom/mavlink.h> instead of <v1.0/common/mavlink.h>.

However, getting mavros to play nicely with my dialect is another matter. it seems like there would be a lot more modification to mavros' source files than I would like. Is there a better way to get my dialect working with mavros?

vooon, 2016-03-24:
1. Follow [source installation instructions][1] to get prepared environment and sources of latest released mavros and mavlink packages.
2. Put your xml at `src/mavlink/message_definitions/v1.0/`
3. Edit [src/mavros/libmavconn/include/mavconn/mavlink_dialect.h][2] to add section for your dialect
4. Run `catkin build --cmake-args -DMAVLINK_DIALECT=<your dialect>`

After 2 + `catkin build` you should see dialect headers directory in `devel/include/mavlink/v1.0/`.
Also try to use verbose output if you're get in trouble.

Note: If #522 done, then step 3 unneeded. 

[1]: https://github.com/mavlink/mavros/tree/master/mavros#source-installation
[2]: https://github.com/mavlink/mavros/blob/master/libmavconn/include/mavconn/mavlink_dialect.h#L79

locustcox, 2016-03-25:
Thanks, @vooon . I guess I should switch over to catkin tools from catkin_make. I assume there shouldn't be any compatibility problems. Thanks again for the help!

closed 2016-03-25 by locustcox

samacumen, 2016-12-21:
Where do we find mavlink_dialect.h file? And what dialect must be added? Is it "Common", "ArduPilot", or a custom definition XML file that has been added (that includes other dialects in it). 

I searched and there is no such file: "src/mavros/libmavconn/include/mavconn/mavlink_dialect.h", there is only mavlink_dialect.h.em file.  In my opinion, a well written code without complete documentation is not too encouraging!

vooon, 2016-12-21:
That guide applicable for versions before 0.18. For 0.18 you only need step 1 and 2.

samacumen, 2016-12-22:
Thanks @vooon . Also, could you please tell me where I can find the file "mavlink_dialect.h" that is included at different files in libMavConn library?

samacumen, 2016-12-22:
[image]

vooon, 2016-12-22:
Did you see cmake scripts? [em_expand() for mavlink_dialect.h generation][1].

[1]: https://github.com/mavlink/mavros/blob/bf941be34478dce7a855edef56318519fec2dbf0/libmavconn/CMakeLists.txt#L57

samacumen, 2016-12-22:
Yes I did see that. But it did not compile, complaining the "missing files". Here is what I did (off the shelf code). I am looking for ROS-independence, i.e. compile this library without ROS. I did google this, but all I could find it some random forum with random suggestion. Any lead it appreciated!

1) Download libMavConn library, install necessary dependencies as given on libMavConn github README.
2) cd libMavConn
3) mkdir build
4) cd build
5) cmake ..

> The following is what I get (attached log):

CMake Error at CMakeLists.txt:7 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    **catkinConfig.cmake
    catkin-config.cmake**

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "~/test_mavros/libmavconn/build/CMakeFiles/CMakeOutput.log

[CMakeOutput.txt](https://github.com/mavlink/mavros/files/669206/CMakeOutput.txt)

vooon, 2016-12-22:
Default cmakelists.txt are written for usage with ROS, so it depends on catkin package (ROS build system).

So to use libmavconn independent you should write your own build script, which:

1) will find "Boost", "console_bridge"
2) precess EmPy templates to make generates sources, see https://github.com/jack-oquin/catkin/blob/indigo-devel/cmake/em_expand.cmake
3) Build library from source files.
4) opt. Install

samacumen, 2016-12-23:
Alright, I have been able to write my build script/ Make file and it compiles all the way through (your steps 1, 3) except for generating the "mavlink_dialect.h" file. I tried to check the meaning of EmPy templates, but I am afraid that is ROS dependent and this is the closest I could get (http://docs.ros.org/jade/api/catkin/html/dev_guide/generated_cmake_api.html#em-expand) and could not make much sense of it.

I am surprised that libMavConn is said to be ROS independent making many users use it, but there are no clear instructions to make it build ROS independent. Any clear instructions on generating this part (mavlink_direct.h) would be much appreciated!

vooon, 2016-12-23:
```cmake
set(PythonIterp_FIND_VERSION "2")
find_package(PythonInterp REQUIRED)

macro(em_expand context_in context_out em_file_in file_out)
  configure_file(${context_in} ${context_out} @ONLY)
  #stamp(${em_file_in})

  # create directory if necessary
  get_filename_component(_folder_out ${file_out} PATH)
  if(NOT IS_DIRECTORY ${_folder_out})
    file(MAKE_DIRECTORY ${_folder_out})
  endif()

  set(command "${PYTHON_EXECUTABLE};${EMPY_SCRIPT}")
  safe_execute_process(COMMAND
    ${command}
    --raw-errors
    -F ${context_out}
    -o ${file_out}
    ${em_file_in})
endmacro()
```

samacumen, 2016-12-24:
Thanks @vooon. I will try the above, is that part of CMake to be executed separately? Also, I am curious as to why you chose Boost over standard C++ libraries, as we can do what is done in libMavConn with it as well? I have been looking at your implementation. Is it for the ease of threading and using function pointers across threads, or was it only a choice?

vooon, 2016-12-24:
If you carefully view, i only use ASIO. Treading - all standard C++11. Standard does not have networking and serial.
There are pure c++11 asio variant, but i don't switch to it, because ROS v1 already use Boost.