mavlink/mavros #525: Standard ROS frame is NWU, but mavros assumes it is ENU

madsciencetist, 2016-03-24:
In 89fa9f03242f8600257cb91f8464925907cf1577, frame conversions was made more pedantic (nothing wrong with that), but the ROS-to-FCU frame conversion was defined as ENU-to-NED. The ROS convention is NWU, so the conversion should instead have been defined as NWU-to-NED. The effect of this bug is that x and y are swapped, which by itself is not harmful because the same incorrect conversion is performed on FCU->ROS local_position and ROS->FCU position_setpoint, so for the most part everything is copacetic. The big problem is in transform_frame_yaw_enu_ned() and transform_frame_yaw_ned_enu(), which each call transform_frame_yaw(), which simply negates yaw, which implies NWU, not NED. The transform_frame_yaw* functions are [only] called in setpoint_raw.cpp, which will transform position and velocity by a 180-degree roll and 90-degree yaw, but only transform yaw by a 180-degree roll, leaving it still 90 degrees off.

An easy fix would be to just change the definition of transform_frame_yaw() from `return -yaw` to `return M_PI/2 - yaw`, but the right thing to do is change all ENU references to NWU to remove the 90-degree rotations entirely, as both ROS and flight controllers both assume x is forward.

mhkabir, 2016-03-24:
> ROS convention is NWU

Where are you getting this from?

madsciencetist, 2016-03-24:
Upon reviewing [REP 103](http://www.ros.org/reps/rep-0103.html), I guess the convention is supposed to be ENU, not NWU, though that surprises me. I suppose the right thing to do then is the easy fix: keep the x/y flip and correct transform_frame_yaw().

closed 2016-06-23 by vooon