mavlink/mavros #532: Multiple Iris on Gazebo using PX4 Firmware

ItzMeJP, 2016-04-09:
Hello guys,

I am working with some SITL simulation on Gazebo v6.0, using PX4 firmware.

I have satisfactory results using MAVROS and only one model of the UAV Iris. So, I tried to insert another model of the quadrotor, but the mavros can't recognize it, even changing the MAVLINK's field "system_id" to zero (broadcast).

Anyone know how can i resolve this problem?

vooon, 2016-04-10:
Each drone require it's own mavros node instance.

ItzMeJP, 2016-04-11:
Yes,
even running two mavros instances, they both don't differentiate the quadrotos on Gazebo. 

I am looking for some help in this link:
https://github.com/PX4/sitl_gazebo/pull/20

Jaeyoung-Lim, 2016-04-22:
How does the UDP port for the IRIS change when you have multiple quadrotors?

closed 2016-06-23 by vooon

cynthiamoon, 2016-11-15:
how to insert another PX4 model on gazebo?